intel/media-driver #1617: [Bug]: av1_qsv - Segmentation fault (core dumped) with "free kernel" build (-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF)

gnat, 2023-02-23:
### Which component impacted?

Decode, Encode

### Is it regression? Good in old configuration?

None

### What happened?

### :arrow_right: Discovery + tons of background information: https://github.com/intel/cartwheel-ffmpeg/issues/235

Previously discussed with: @dvrogozh @nyanmisaka @xhaihao @wenbinc-Bin over at `inte/cartwheel-ffmpeg`

### Short description

#### `intel-media-va-driver` master, default.

* :green_circle:  `av1_qsv` works
  * `./ffmpeg -loglevel verbose -hwaccel qsv -i ./input.mp4 -c:v av1_qsv -y output.mp4`
* :green_circle: `h264_qsv` works 
  * `./ffmpeg -loglevel verbose -hwaccel qsv -i ./input.mp4 -c:v h264_qsv -y output.mp4`

#### `intel-media-va-driver` master, with `-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF`

* :red_circle:  `av1_qsv` `Segmentation fault (core dumped)`
  * `./ffmpeg -loglevel verbose -hwaccel qsv -i ./input.mp4 -c:v av1_qsv -y output.mp4`
* :green_circle: `h264_qsv` works.
  * `./ffmpeg -loglevel verbose -hwaccel qsv -i ./input.mp4 -c:v h264_qsv -y output.mp4`

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Content Creation

### What impacted?

### Timeframe / Urgency (Ubuntu 23.04)

The `intel-media-va-driver` being shipped with Ubuntu suffers from this crash- usage of `av1_qsv` is currently broken. It'll be paramount to let the Ubuntu `intel-media-va-driver` package maintainers know when the fix is out for this before the development window for Ubuntu 23.04 Lunar closes- Latest end of March to get the fix in time for release of Ubuntu 23.04.

* It would be unfortunate to have this needlessly broken until October (next Ubuntu release, 23.10).
* Release schedule is here: https://discourse.ubuntu.com/t/lunar-lobster-release-schedule/27284

### Debug Information

* Backtrace (gdb): https://github.com/intel/cartwheel-ffmpeg/issues/235#issuecomment-1440536899
* `vainfo -a`: https://github.com/intel/cartwheel-ffmpeg/issues/235#issuecomment-1440627311
* `apt list --installed | grep intel-media-va-driver`: https://github.com/intel/cartwheel-ffmpeg/issues/235#issuecomment-1440669367

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2023-03-01:
When enabling these two build options '-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF', kernel shader is required, if media driver doesn't provide the shader, it may need to disable related features.

xhaihao, 2023-03-02:
av1_vaapi encoder has the same issue with -DENABLE_NONFREE_KERNELS=OFF
$ ffmpeg -loglevel verbose -hwaccel vaapi -hwaccel_output_format vaapi  -i input.mp4 -c:v av1_vaapi -y output.mp4

intel-mediadev, 2023-03-02:
Auto Created VSMGWL-62074 for further analysis.

eero-t, 2023-03-02:
> https://github.com/intel/cartwheel-ffmpeg/issues/235

I guess the just released upstream FFmpeg 6.0 version can also be used to trigger this: https://ffmpeg.org/index.html#news ?

Sherry-Lin, 2023-03-06:
@gnat Thanks for reporting this issue. Like you mentioned, intel-media-va-driver-non-free could resolve this issue that build is our default build to support more features. We are debugging why intel-media-va-driver build has this issue. May I know any preference for driver build selection? Prefer to use intel-media-va-driver-non-free default build or intel-media-va-driver build?

xhaihao, 2023-03-06:
@eero-t  Yes, you may use av1_qsv encoder in FFmpeg 6.0 to trigger this issue.

closed 2023-03-13 by intel-mediadev

xhaihao, 2023-03-16:
This issue still exists. 
```
$ ffmpeg -f lavfi -i yuvtestsrc -vf "format=nv12" -c:v av1_qsv -f null -
...
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf60.4.100
  Stream #0:0: Video: av1, nv12(tv, progressive), 320x240 [SAR 1:1 DAR 4:3], q=2-31, 1000 kb/s, 25 fps, 25 tbn
    Metadata:
      encoder         : Lavc60.6.101 av1_qsv
Segmentation fault (core dumped)      0kB time=-577014:32:22.77 bitrate=  -0.0kbits/s speed=N/A
```

xhaihao, 2023-03-20:
backtrace info for the segfault issue when running `ffmpeg -f lavfi -i yuvtestsrc -vf "format=nv12" -c:v av1_qsv -f null -`

#0  0x0000000000000000 in ?? ()
#1  0x00007fffc4f6e56b in SwizzleSurfaceByHW (surface=0x5555564eb650, isDeSwizzle=false) at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:1147
#2  0x00007fffc4f6e9fb in DdiMediaUtil_LockSurfaceInternal (surface=0x5555564eb650, flag=3)
    at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:1225
#3  0x00007fffc4f6e709 in DdiMediaUtil_LockSurface (surface=0x5555564eb650, flag=3) at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:1169
#4  0x00007fffc4f6f1b1 in DdiMediaUtil_LockBuffer (buf=0x7fffc0001130, flag=3) at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:1385
#5  0x00007fffc4f449e4 in DdiMedia_MapBufferInternal (ctx=0x5555565a01e0, buf_id=13, pbuf=0x7fffc77fd5c8, flag=3)
    at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3624
#6  0x00007fffc4f44ab6 in DdiMedia_MapBuffer (ctx=0x5555565a01e0, buf_id=13, pbuf=0x7fffc77fd5c8)
    at /home/gta/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3652
#7  0x00007ffff6fd6fa5 in vaMapBuffer (dpy=0x555556592000, buf_id=13, pbuf=0x7fffc77fd5c8) at ../va/va.c:1445
#8  0x00007fffc870bf2d in VAAPIVideoCORE_VPL::DoFastCopyExtended(mfxFrameSurface1*, mfxFrameSurface1*, unsigned int) () from /usr/lib/x86_64-linux-gnu/libmfx-gen.so.1.2.8
#9  0x00007fffc870b111 in VAAPIVideoCORE_VPL::DoFastCopyWrapper(mfxFrameSurface1*, unsigned short, mfxFrameSurface1*, unsigned short, unsigned int) ()
   from /usr/lib/x86_64-linux-gnu/libmfx-gen.so.1.2.8
#10 0x00007fffc8929e4c in AV1EHW::Base::General::SubmitTask(AV1EHW::FeatureBlocks const&, std::function<void (unsigned int, std::function<mfxStatus (MfxFeatureBlocks::StorageW&, MfxFeatureBlocks::StorageW&)>&&)>)::{lambda(MfxFeatureBlocks::StorageW&, MfxFeatureBlocks::StorageW&)#2}::operator()(MfxFeatureBlocks::StorageW&, MfxFeatureBlocks::StorageW&) const [clone .constprop.0]
    () from /usr/lib/x86_64-linux-gnu/libmfx-gen.so.1.2.8

xhaihao, 2023-03-20:
mediaDrvCtx->pfnMediaMemoryTileConvert is NULL for `-DENABLE_NONFREE_KERNELS=OFF`

Sherry-Lin, 2023-03-22:
> mediaDrvCtx->pfnMediaMemoryTileConvert is NULL for `-DENABLE_NONFREE_KERNELS=OFF`

Issue should be fixed by [e68feab](https://github.com/intel/media-driver/commit/e68feab7d)

commit 93cb7417 (2023-03-14): [Decode] Add strict check for mmc using

Fixes: #1617
MMC build option is default off when enabling free build