intel/media-driver #1618: [Bug]: MADI/MCDI is still broken, proposed fix is ignored

smp79, 2023-02-25:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

ADI is non-functional, newer platforms (TGL+) that are not supported by intel-vaapi-driver can not use high quality advanced deinterlacing. 
#1162 fixes the issue but that PR seem to be forgotten/ignored for almost 2 years. Please fix this.

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

LhGu, 2023-03-01:
Fix in this PR: https://github.com/intel/media-driver/pull/1162.

intel-mediadev, 2023-03-01:
Auto Created VSMGWL-62032 for further analysis.

LhGu, 2023-05-06:
Close this issue, pls chat in this PR: https://github.com/intel/media-driver/pull/1162

closed 2023-05-06 by LhGu