intel/media-driver #1619: [Bug]: The vaGetImage()/vaPutImage() pair performance is lower than then vaDeriveImage()

HeJunyan, 2023-03-01:
### Is it regression? Good in old configuration?

None

### What happened?

When we want to dump a surface from GPU to CPU side,  the vaGetImage()/vaPutImage() pair performance is lower than then the manner using vaDeriveImage().

So far as we know, the vaGetImage()/vaPutImage() pair should be the first choice for downloading  and uploading the surface from CPU to GPU side, especially on the DGPU mode. And when we use vaGetImage()/vaPutImage() to implement surface mapping,  we can provide the direction and read only, write only hint to optimize the mapping performance.

But currently, the vaGetImage()/vaPutImage() performance is about 20% lower than the vaDeriveImage() when mapping the same surface.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2023-03-01:
Do you try the latest media driver to check if it still has perf drop issue?
Add similar issue for refering: https://github.com/intel/media-driver/issues/1537

intel-mediadev, 2023-03-01:
Auto Created VSMGWL-62044 for further analysis.

MicroYY, 2023-03-07:
There is CPU mem copy in vaGetImage/vaPutImage which I guess may be the culprit of perf drop.
@XinfengZhang Is the CPU copy always necessary? Can we assume the perf drop is expected?
For me, it doesn't make sense to always do CPU copy on a locked surface which has already been unswizzled.

XinfengZhang, 2023-03-16:
certainly we should use GPU copy, and not use CPU copy

mengker33, 2023-04-21:
Hi, any update on this issue?

Sherry-Lin, 2023-04-21:
Is this issue resolved by https://github.com/intel/media-driver/pull/1569?

MicroYY, 2023-04-26:
The perf drop is expected as CPU copy introduced by vagGetImage/vaPutImage. To avoid the perf drop, new code design is needed which isn't a short term solution and may need much effort.