intel/media-driver #1620: [Bug]: AV1 report wrong max reference list number.

HeJunyan, 2023-03-01:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

The AV1 encoder reports wrong VAConfigAttribEncMaxRefFrames attruibute query.

It reports 3 reference list number, such as:
https://github.com/intel/media-driver/blob/master/media_driver/linux/Xe_M/ddi/media_libva_caps_dg2.cpp#L52
which violate the vaapi defination:
/*  the value represents the maximum number
     * of reference frames for both the reference picture list 0 (bottom
     * 16 bits) and the reference picture list 1 (top 16 bits). */

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-03-29:
Auto Created VSMGWL-63005 for further analysis.

leyu-yao, 2023-07-20:
Thanks for reporting this issue. We fixed this in https://github.com/intel/media-driver/commit/bce5a8a36f7e46f8bbe377fb307f88c0fe7e23ee.

Sherry-Lin, 2023-08-02:
Issue fixed by https://github.com/intel/media-driver/commit/bce5a8a36f7e46f8bbe377fb307f88c0fe7e23ee.

closed 2023-08-02 by Sherry-Lin