intel/media-driver #1623: [Bug]: AV1 10-bit hw decoding artifacts

smp79, 2023-03-03:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

I see this issue on TGL when playing some 10-bit AV1 videos in Kodi - pixelation when using VAAPI hw acceleration. The same video plays fine when using libdav1d sw decode.
Sample of problematic av1 video - https://www.dropbox.com/s/pn723cfgpenlpvj/av1-10bit-1080p-sample.mkv?dl=1

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

libva info: VA-API version 1.17.0
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.2 (b82a6833bb)

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2023-03-06:
Did you try other player with vaapi hw like sample_decoder or ffmpeg to check issue still existing?
You could also try to change 'Kodi mode' mentioned in https://github.com/intel/media-driver/issues/1471

intel-mediadev, 2023-03-06:
Auto Created VSMGWL-62138 for further analysis.

protomucca, 2023-04-08:
same issue here with libreELEC, the problem appears randomly and not related to specific file nor specific position, indeed the same file sometime play well and going back for a few seconds artifacts disappears.  Hope this helps

Jexu, 2023-07-12:
Could you help try your video content with ffmpeg or sample_decoder firstly,  to check if having same issue. If not, you could take this issue to Kodi and ask to triage firstly

closed 2023-11-01 by smp79