intel/media-driver #1624: [Bug]: doc is out of date

xhaihao, 2023-03-04:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

Seems both https://github.com/intel/media-driver#readme and https://github.com/intel/media-driver/blob/master/docs/media_features.md are out of date. 

E.g. according to the output of vainfo on TGL,  VAProfileHEVCMain12 and VAProfileHEVCMain422_12 are supported for encoding, however I can't find HEVC 422 12bit in https://github.com/intel/media-driver#readme  and HEVC 420/422 12bit in https://github.com/intel/media-driver/blob/master/docs/media_features.md for encoding.

$ vainfo | grep VAEntrypointEnc | grep 12
error: can't connect to X server!
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-03-06:
Auto Created VSMGWL-62137 for further analysis.

OttoYang, 2023-07-28:
Hi @xhaihao 
Since VDEnc doesn’t support 12bit, we don’t recommend customer to use 12bit encoding on VME to avoid regression after migrating to VDEnc.  
So there is no plan to update the docs on 12bit encoding support on VME.  
And appreciate your findings and asking.

OttoYang, 2023-08-01:
> Hi @xhaihao Since VDEnc doesn’t support 12bit, we don’t recommend customer to use 12bit encoding on VME to avoid regression after migrating to VDEnc. So there is no plan to update the docs on 12bit encoding support on VME. And appreciate your findings and asking.

Hi @xhaihao, Can I close this issue if there are no more questions about it?

OttoYang, 2023-08-04:
I'll close this issue and feel free to reopen it if you have any concerns about this.

closed 2023-08-04 by OttoYang