intel/media-driver #1625: [Bug]: Build issue - Custom installation path not handled for iHD_drv_video.so

th0ma7, 2023-03-05:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

When cross-compiling using a non-standard installation path, the `libigfxcmrt.so` gets installed at the right destination.  Altough the driver `iHD_drv_video.so` add the full installation path in front of the destination directory.

The end-result `tree` output looks like this where the `iHD_drv_video.so` directory tree is bieng behaving badly.
```
work-x64-7.1/install/home/
└── spksrc
    └── ffmpeg5
        └── spksrc
            └── cross
                └── intel-media-driver-latest
                    └── work-x64-7.1
                        └── install
                            └── usr
                                └── local
                                    └── intel-media
                                        └── lib
                                            └── iHD_drv_video.so
work-x64-7.1/install/usr/
└── local
    └── intel-media
        ├── lib
        │   ├── libigfxcmrt.so -> libigfxcmrt.so.7
        │   ├── libigfxcmrt.so.7 -> libigfxcmrt.so.7.2.0
        │   ├── libigfxcmrt.so.7.2.0
        │   └── pkgconfig
        │       └── igfxcmrt.pc
        └── ...
```

This issue as been there since years but I was able to work-around it by enforcing `CMAKE_ARGS += -DCMAKE_INSTALL_PREFIX=$(WORK_DIR)/install/$(INSTALL_PREFIX)` and not using `DESTDIR` variable.  But that no longer works since version 22.6.6.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

To work-around this I'll have to manually mv the  `iHD_drv_video.so` and change it's built-in path for it to work.  Hopefully the CMAKE install process can be fixed to work properly OR another variable can be set at configure time so it gets installed properly.

Full build log output:
[build-x64-7.1.log](https://github.com/intel/media-driver/files/10892714/build-x64-7.1.log)

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-03-06:
Auto Created VSMGWL-62136 for further analysis.

MicroYY, 2023-03-07:
Could you provide the exact cmake && make cmd line?

There may be alternative ways to workaround this by set env variables if you just want to let driver work.
```
$ export LIBVA_DRIVERS_PATH=path_to_iHD_drv_video.so
$ export LIBVA_DRIVER_NAME=iHD
```
If you must install iHD to the customized path, I'm not sure whether the install path is hardcoded to /usr/lib/x86_64-linux-gnu/dri. 
@XinfengZhang @Sherry-Lin  Could you help comment?

MicroYY, 2023-03-07:
From driver cmake, iHD_drv_video.so is always installed on path_to_libva.so/**dri** if libva is found. `-DCMAKE_INSTALL_PREFIX` won't work in this case. No support for custom installation path of iHD yet.

haichund1, 2023-03-30:
@th0ma7 any update the WA we provided?

th0ma7, 2023-03-30:
I have not retried it as  there isn't any support for what i require yet.  Did anything change since?

MicroYY, 2023-04-13:
I double checked with @XinfengZhang, the driver installation path depends on the libva installation path, which means the iHD_drv_video.so will locate in path_to_va/dri/iHD_drv_video.so
Currently we don't support customized installation path.
@XinfengZhang Is it a new feature request?

th0ma7, 2023-05-06:
Following-up on this, it seems the CMAKE rules where fixed since last I looked at this issue.  Version 23.1.6 behave as expected and installs the `iHD_drv_video.so` at the proper location.

Thnx for following up.

closed 2023-05-06 by th0ma7