intel/media-driver #1626: [Bug]: AV1 ENCODE QSV/ARC - 2 encodes at same time results in errors

vid-bin, 2023-03-06:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

Using ffmpeg on linux with the following packages running two QSV_AV1 encodes at the same time will result in errors. The video does finish encoding but there are corrupt frames that cause hls.js to black screen throughout the video.

Using only one encode and the problem goes away.

```
error[3/6/2023, 6:42:50 AM] Error in ffmpeg.

{
  "stdout": "",
  "stderr": "frame=17658 fps=129 q=-0.0 size=  163840kB time=00:04:54.66 bitrate=4555.0kbits/s speed=2.16x    \nframe=17722 fps=129 q=-0.0 size=  164608kB time=00:04:55.76 bitrate=4559.3kbits/s speed=2.16x    \nframe=17785 fps=129 q=-0.0 size=  165120kB time=00:04:56.80 bitrate=4557.5kbits/s speed=2.16x    \nframe=17852 fps=129 q=-0.0 size=  165376kB time=00:04:57.92 bitrate=4547.4kbits/s speed=2.16x    \nframe=17918 fps=129 q=-0.0 size=  165632kB time=00:04:59.02 bitrate=4537.7kbits/s speed=2.16x    \nframe=17984 fps=129 q=-0.0 size=  166400kB time=00:05:00.12 bitrate=4542.0kbits/s speed=2.16x    \nframe=18050 fps=129 q=-0.0 size=  167168kB time=00:05:01.22 bitrate=4546.3kbits/s speed=2.16x    \nframe=18115 fps=129 q=-0.0 size=  167936kB time=00:05:02.30 bitrate=4550.9kbits/s speed=2.16x    \nframe=18182 fps=129 q=-0.0 size=  168448kB time=00:05:03.42 bitrate=4547.9kbits/s speed=2.16x    \nframe=18248 fps=129 q=-0.0 size=  168960kB time=00:05:04.52 bitrate=4545.2kbits/s speed=2.16x    \nframe=18314 fps=129 q=-0.0 size=  169984kB time=00:05:05.60 bitrate=4556.6kbits/s speed=2.16x    \nframe=18380 fps=129 q=-0.0 size=  170496kB time=00:05:06.72 bitrate=4553.7kbits/s speed=2.16x    \nframe=18446 fps=129 q=-0.0 size=  171264kB time=00:05:07.80 bitrate=4558.1kbits/s speed=2.16x    \nframe=18513 fps=129 q=-0.0 size=  171776kB time=00:05:08.94 bitrate=4554.9kbits/s speed=2.16x    \nframe=18580 fps=129 q=-0.0 size=  172288kB time=00:05:10.06 bitrate=4552.0kbits/s speed=2.16x    \nframe=18646 fps=129 q=-0.0 size=  172544kB time=00:05:11.16 bitrate=4542.6kbits/s speed=2.16x    \nframe=18711 fps=129 q=-0.0 size=  172800kB time=00:05:12.24 bitrate=4533.6kbits/s speed=2.16x    \nframe=18776 fps=129 q=-0.0 size=  173056kB time=00:05:13.32 bitrate=4524.7kbits/s speed=2.16x    \nframe=18840 fps=129 q=-0.0 size=  173568kB time=00:05:14.38 bitrate=4522.8kbits/s speed=2.16x    \nframe=18905 fps=129 q=-0.0 size=  174848kB time=00:05:15.46 bitrate=4540.5kbits/s speed=2.16x    \nframe=18972 fps=129 q=-0.0 size=  175616kB time=00:05:16.58 bitrate=4544.3kbits/s speed=2.16x    \nframe=19035 fps=129 q=-0.0 size=  175616kB time=00:05:17.64 bitrate=4529.2kbits/s speed=2.16x    \nframe=19102 fps=129 q=-0.0 size=  176128kB time=00:05:18.74 bitrate=4526.7kbits/s speed=2.16x    \nframe=19169 fps=129 q=-0.0 size=  176384kB time=00:05:19.86 bitrate=4517.4kbits/s speed=2.16x    \nframe=19234 fps=129 q=-0.0 size=  177664kB time=00:05:20.96 bitrate=4534.6kbits/s speed=2.16x    \nframe=19301 fps=129 q=-0.0 size=  178432kB time=00:05:22.06 bitrate=4538.6kbits/s speed=2.16x    \nframe=19368 fps=129 q=-0.0 size=  178944kB time=00:05:23.16 bitrate=4536.2kbits/s speed=2.16x    \nframe=19433 fps=129 q=-0.0 size=  179712kB time=00:05:24.26 bitrate=4540.2kbits/s speed=2.16x    \nframe=19500 fps=129 q=-0.0 size=  180224kB time=00:05:25.38 bitrate=4537.4kbits/s speed=2.16x    \nframe=19566 fps=129 q=-0.0 size=  180736kB time=00:05:26.46 bitrate=4535.3kbits/s speed=2.16x    \nframe=19633 fps=129 q=-0.0 size=  181504kB time=00:05:27.60 bitrate=4538.7kbits/s speed=2.16x    \nframe=19699 fps=129 q=-0.0 size=  182016kB time=00:05:28.70 bitrate=4536.3kbits/s speed=2.16x    \nframe=19765 fps=129 q=-0.0 size=  182784kB time=00:05:29.80 bitrate=4540.2kbits/s speed=2.16x    \nframe=19832 fps=129 q=-0.0 size=  183296kB time=00:05:30.92 bitrate=4537.5kbits/s speed=2.16x    \nframe=19897 fps=129 q=-0.0 size=  184064kB time=00:05:32.00 bitrate=4541.7kbits/s speed=2.16x    \nframe=19962 fps=129 q=-0.0 size=  184832kB time=00:05:33.08 bitrate=4545.9kbits/s speed=2.16x    \nframe=20028 fps=129 q=-0.0 size=  185344kB time=00:05:34.18 bitrate=4543.5kbits/s speed=2.16x    \nframe=20093 fps=129 q=-0.0 size=  185856kB time=00:05:35.26 bitrate=4541.3kbits/s speed=2.16x    \nframe=20157 fps=129 q=-0.0 size=  186368kB time=00:05:36.34 bitrate=4539.2kbits/s speed=2.16x    \nframe=20222 fps=129 q=-0.0 size=  187136kB time=00:05:37.40 bitrate=4543.6kbits/s speed=2.16x    \nframe=20288 fps=129 q=-0.0 size=  187904kB time=00:05:38.52 bitrate=4547.2kbits/s speed=2.16x    \nframe=20352 fps=129 q=-0.0 size=  188416kB time=00:05:39.58 bitrate=4545.3kbits/s speed=2.16x    \nframe=20418 fps=129 q=-0.0 size=  189184kB time=00:05:40.66 bitrate=4549.4kbits/s speed=2.16x    \nframe=20480 fps=129 q=-0.0 size=  189696kB time=00:05:41.72 bitrate=4547.5kbits/s speed=2.16x    \nframe=20545 fps=129 q=-0.0 size=  190208kB time=00:05:42.80 bitrate=4545.4kbits/s speed=2.16x    \nframe=20611 fps=129 q=-0.0 size=  190976kB time=00:05:43.90 bitrate=4549.2kbits/s speed=2.16x    \nframe=20676 fps=129 q=-0.0 size=  191488kB time=00:05:44.98 bitrate=4547.1kbits/s speed=2.16x    \nframe=20742 fps=129 q=-0.0 size=  192000kB time=00:05:46.08 bitrate=4544.8kbits/s speed=2.16x    \nframe=20808 fps=129 q=-0.0 size=  192768kB time=00:05:47.16 bitrate=4548.8kbits/s speed=2.16x    \nframe=20871 fps=129 q=-0.0 size=  193536kB time=00:05:48.24 bitrate=4552.7kbits/s speed=2.16x    \nframe=20935 fps=129 q=-0.0 size=  194048kB time=00:05:49.30 bitrate=4550.9kbits/s speed=2.16x    \nframe=21002 fps=129 q=-0.0 size=  194816kB time=00:05:50.40 bitrate=4554.6kbits/s speed=2.16x    \nframe=21066 fps=129 q=-0.0 size=  195328kB time=00:05:51.48 bitrate=4552.5kbits/s speed=2.16x    \nframe=21131 fps=129 q=-0.0 size=  195840kB time=00:05:52.56 bitrate=4550.5kbits/s speed=2.16x    \nframe=21196 fps=129 q=-0.0 size=  196608kB time=00:05:53.66 bitrate=4554.1kbits/s speed=2.16x    \nframe=21259 fps=129 q=-0.0 size=  197376kB time=00:05:54.70 bitrate=4558.5kbits/s speed=2.16x    \nframe=21325 fps=129 q=-0.0 size=  197888kB time=00:05:55.80 bitrate=4556.2kbits/s speed=2.16x    \nframe=21388 fps=129 q=-0.0 size=  198400kB time=00:05:56.86 bitrate=4554.4kbits/s speed=2.16x    \nframe=21451 fps=129 q=-0.0 size=  198912kB time=00:05:57.90 bitrate=4552.9kbits/s speed=2.16x    \nframe=21516 fps=129 q=-0.0 size=  199680kB time=00:05:58.96 bitrate=4557.0kbits/s speed=2.16x    \nframe=21579 fps=129 q=-0.0 size=  200192kB time=00:06:00.06 bitrate=4554.7kbits/s speed=2.16x    \nframe=21643 fps=129 q=-0.0 size=  200704kB time=00:06:01.10 bitrate=4553.2kbits/s speed=2.16x    \nframe=21707 fps=129 q=-0.0 size=  201472kB time=00:06:02.16 bitrate=4557.3kbits/s speed=2.16x    \nframe=21771 fps=129 q=-0.0 size=  201984kB time=00:06:03.24 bitrate=4555.2kbits/s speed=2.16x    \nframe=21834 fps=129 q=-0.0 size=  202496kB time=00:06:04.28 bitrate=4553.8kbits/s speed=2.16x    \nframe=21900 fps=129 q=-0.0 size=  203264kB time=00:06:05.38 bitrate=4557.3kbits/s speed=2.16x    \nframe=21965 fps=129 q=-0.0 size=  204032kB time=00:06:06.46 bitrate=4561.0kbits/s speed=2.16x    \nframe=22031 fps=129 q=-0.0 size=  204544kB time=00:06:07.56 bitrate=4558.8kbits/s speed=2.16x    \nframe=22098 fps=129 q=-0.0 size=  205312kB time=00:06:08.68 bitrate=4562.0kbits/s speed=2.16x    \nframe=22162 fps=129 q=-0.0 size=  205824kB time=00:06:09.76 bitrate=4560.0kbits/s speed=2.16x    \nframe=22226 fps=129 q=-0.0 size=  206592kB time=00:06:10.82 bitrate=4563.9kbits/s speed=2.16x    \nframe=22290 fps=129 q=-0.0 size=  207104kB time=00:06:11.86 bitrate=4562.4kbits/s speed=2.16x    \nframe=22355 fps=129 q=-0.0 size=  207616kB time=00:06:12.96 bitrate=4560.2kbits/s speed=2.16x    \nframe=22422 fps=129 q=-0.0 size=  208384kB time=00:06:14.06 bitrate=4563.6kbits/s speed=2.16x    \nframe=22488 fps=129 q=-0.0 size=  208896kB time=00:06:15.18 bitrate=4561.2kbits/s speed=2.16x    \nframe=22552 fps=129 q=-0.0 size=  209664kB time=00:06:16.26 bitrate=4564.8kbits/s speed=2.16x    \nframe=22619 fps=129 q=-0.0 size=  210176kB time=00:06:17.36 bitrate=4562.6kbits/s speed=2.16x    \nframe=22683 fps=129 q=-0.0 size=  210944kB time=00:06:18.44 bitrate=4566.2kbits/s speed=2.16x    \nframe=22748 fps=129 q=-0.0 size=  211456kB time=00:06:19.52 bitrate=4564.3kbits/s speed=2.16x    \nframe=22814 fps=129 q=-0.0 size=  212224kB time=00:06:20.60 bitrate=4567.9kbits/s speed=2.16x    \nframe=22878 fps=129 q=-0.0 size=  212736kB time=00:06:21.68 bitrate=4565.9kbits/s speed=2.16x    \n [... cut ...]
### What's the usage scenario when you are seeing the problem?

Content Creation

### What impacted?

Content creation via peertube

### Debug Information

ffmpeg git
libva-2.17.0
libva-utils-2.17.1
gmmlib-22.3.4
libva-intelmedia-driver-23.1.2

```
ls -l /dev/dri
total 0
drwxr-xr-x  2 root root         80 Mar  6 13:35 by-path
crw-rw----+ 1 root video  226,   0 Mar  6 13:35 card0
crw-rw-rw-  1 root render 226, 128 Mar  6 13:35 renderD128
```

[image]

### Do you want to contribute a patch to fix the issue?

None

vid-bin, 2023-03-15:
I'm actually having the same error when transcoding a 1440p video to a 360p video, but the error doesn't appear to happen on 480p.

Not sure if it's the low resolution or the low bitrate though.

Sherry-Lin, 2023-04-18:
@xhaihao could you help check it firstly? Thanks.

xhaihao, 2023-04-18:
> I'm actually having the same error when transcoding a 1440p video to a 360p video, but the error doesn't appear to happen on 480p.
> 
Could you provide the command line ?

xhaihao, 2023-04-18:
I can't produce the original issue, could you provide your clip ?

wenbinc-Bin, 2023-04-19:
@vid-bin Where do you get the FFmpeg? The command in "shellCommand" doesn't work on my environment. I need to remove options to make it work. Also I cannot reproduce this issue neither.

vid-bin, 2023-04-21:
The shell command did change over the last couple of weeks (the original wasn't the best). I'll test this again and see if I can replicate it with the updated software and kernel and then update this ticket.

I'm using the latest git ffmpeg with 30-60 minute 1080p60 videos.

vid-bin, 2023-04-21:
I'm not seeing the issue anymore but haven't had time to test this extensively.

Here is the new shell command:

`/usr/bin/ffmpeg -i /var/www/peertube/storage/videos/private/4bae91e4-8110-4ae4-9f97-51896b6f1b47-1440.mp4 -y -acodec libopus -vcodec av1_qsv -threads 12 -f mp4 -movflags faststart -max_muxing_queue_size 1024 -map_metadata -1 -b:a 128k -strict -2 -vbr on -vf scale=w=-2:h=1080 -pix_fmt yuv420p10le -extbrc 1 -b_strategy 1 -look_ahead 1 -g 600 -preset veryslow -tile_cols 2 -tile_rows 2 -minrate 10k -b:v 4000k -maxrate 6000k /var/www/peertube/storage/tmp/transcoded.mp4`

vid-bin, 2023-04-22:
Yes, I can replicate this issue again. It seems to only happen on very long 40+ minute videos though.

`n[av1_qsv @ 0x55b4ded57b00] extract_extradata filter failed to receive output packet\n[vost#0:0/av1_qsv @ 0x55b4ded57700] Error submitting video frame to the encoder\n[libopus @ 0x55b4df2108c0] 1 frames left in the queue on closing\nConversion failed!\n",`

wenbinc-Bin, 2023-04-24:
Hi @vid-bin, thanks for your experiment. Could you please cut the input video and just encode the broken part? This help us to figure out if it is content specific issue or it is caused by long time running. I tried to encode a video that is one hour long, but I still cannot reproduce it.

vid-bin, 2023-04-26:
It's not a specific video. I have been able to reproduce this on high bitrate 1440p60 youtube videos.

It only happens when you have two simultaneous encodes running at the same time @ 40-60 minutes long. It's possible this can be triggered in other ways, but that's how I've been able to trigger it every time.

The source videos are 1440p60 and the encoded resolution is 1080p60.

xhaihao, 2023-05-08:
@vid-bin did you see the warning message below in your FFmpeg output ?
```
[av1_qsv @ 000001FE632189C0] Unknown FrameType, set pict_type to AV_PICTURE_TYPE_NONE.speed=N/A
```

leyu-yao, 2023-07-21:
Hi @vid-bin . I have reproduced this on my side and we may need more discussion on the fixing plan. Temporarily, my suggestion is to use single tile for this resolution, that is, using "-tile_cols 1 -tile_rows 1" in ffmpeg option. Please feel free to let me know if you have any concern about this temporal solution.

leyu-yao, 2023-09-19:
Starting from this commit https://github.com/intel/media-driver/commit/20bca244131064d7a84ec484906438f7e04ae7c8
This failure is expected to be fixed.

vid-bin, 2023-09-19:
Great! I will report back if it's not fixed. May be a few weeks due to my hectic schedule though.

Feierkuula, 2023-09-24:
I tested it, it works perfectly.

Feierkuula, 2023-09-25:
I have been batch-encoding many videos since 2 days with the very recent libs like here, using 8 ffmpeg in total, 4x ffmpeg per GPU. I did not notice an glitch in the produced videos until now so far. I am testing more to be 100% sure.

leyu-yao, 2023-09-25:
> I have been batch-encoding many videos since 2 days with the very recent libs like here, using 8 ffmpeg in total, 4x ffmpeg per GPU. I did not notice an glitch in the produced videos until now so far. I am testing more to be 100% sure.

@Feierkuula 
Nice to hear that! Thanks for your experiments and info.

Feierkuula, 2023-10-02:
Still testing since 1 week, atm.

Yesterday, my script with another program QSVEnc misselected the GPU and gave 100% full load on it, on one of the videos, it was glitchy a bit, the keyframes jumped back and forths, etc

Then i thought what can be wrong... it supposed on the same driver.

Then i remember, that i have 2 GPUs of ASRock A380 OC - which is factory overclocked, it may be this issue, because overclocked chips may produce glitchy clips.

So i lowered the frequency from 2450 MHz to orginially designed 2000 MHz and thus causing a perfomance drop of about ~1-2% but may improve stability and fix the glitch.

So, i am still testing. Other customers of ASRock A380 OC should consider having their GPUs downclocked. 
The command i used under Linux to downclock the GPU was:

```
echo 2000 | sudo tee /sys/class/drm/card0/gt_boost_freq_mhz 
echo 2000 | sudo tee /sys/class/drm/card1/gt_boost_freq_mhz 
echo 2000 | sudo tee /sys/class/drm/card0/gt_max_freq_mhz
echo 2000 | sudo tee /sys/class/drm/card1/gt_max_freq_mhz
```

Feierkuula, 2023-10-19:
This message comes sometimes, but when tried at later time again, the encoder works again

```
[av1_qsv @ 0x5614df9d5800] Invalid FrameType:0.
[vost#0:0/av1_qsv @ 0x5614dfa48480] Error submitting video frame to the encoder
Error while filtering: Invalid data found when processing input
[h264_qsv @ 0x5614df9a7600] A decode call did not consume any data: expect more data at input (-10)

```

Feierkuula, 2023-10-23:
there is new log ouput in kernel, when doing 8 encodes at same time with just 2 gpus:

`
[94569.689649] i915 0000:2f:00.0: [drm] GPU HANG: ecode 12:4:a8ffff7d, in ffmpeg [374811]
[94569.689658] i915 0000:2f:00.0: [drm] ffmpeg[374811] context reset due to GPU hang
`

Feierkuula, 2023-10-23:
I attached the file with dump from /sys/kernel/debug/dri/0/i915_error_state

[i915_error_state.txt](https://github.com/intel/media-driver/files/13065043/i915_error_state.txt)

Galvo87, 2023-11-07:
Hi, I am having a similar issue when encoding long videos with FFmpeg.

[av1_qsv @ 0x556fbd2c0880] Invalid FrameType:0.=00:17:05.17 bitrate= 509.4kbits/s speed=6.78x
[vost#0:0/av1_qsv @ 0x556fbd45ac40] Error submitting video frame to the encoder
Error while filtering: Invalid data found when processing input

Any hints?

leyu-yao, 2023-12-19:
Hi @Feierkuula 
I would like to know whether your case is a multi-tile case? I believe the original failure is this issue only reproduceable in multi-tile  case.
From your dmesg, that should be a different issue. We will check the i915_error_state.