intel/media-driver #1627: [Bug]:

Lu-Anxiao, 2023-03-09:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

HEVC 4K 16 channels video wall composition FPS drops to 15 while it could achieve 40 FPS for 15 channels.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Video composition

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

FurongZhang, 2023-03-09:
Thanks @Lu-Anxiao . I saw you filed some patch for this issue. Will review your patch and merge that once it is ready.

FurongZhang, 2023-03-09:
https://github.com/intel-innersource/drivers.gpu.unified/pull/111159

intel-mediadev, 2023-04-18:
Auto Created VSMGWL-63676 for further analysis.

closed 2023-04-19 by Lu-Anxiao

xiaoton1, 2023-04-19:
duplicated to https://jira.devtools.intel.com/browse/VSMGWL-62274