intel/media-driver #1628: [Bug][DG2][VPP]: tonemap_vaapi extremely slow on certain input size 1280*(722 vs 720)

nyanmisaka, 2023-03-13:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. Linux 6.2.5, FFmpeg 5/6/cartwheel, Arc A380, ReBAR On
2. libva 2.17, gmmlib 22.3.4, media-driver 23.1.3
3. Run FFmpeg cli and watch the fps:

Two bad resolutions: 1280x**722** and 1280x**692** - Results in very slow FPS - < 5 FPS.

```
frame=   22 fps=3.8 q=-0.0 Lsize=N/A time=00:00:00.35 bitrate=N/A speed=0.0606x
```

But 1280x**720** and 1280x**694** works just fine - hundreds FPS.

Other resolutions may also affected but I haven't found the pattern yet.

```
ffmpeg -hwaccel vaapi -hwaccel_output_format vaapi -i ANY_10Bit_HDR_CLIP.mp4 -vf scale_vaapi=1280:722,tonemap_vaapi=format=nv12:t=bt709:m=bt709:p=bt709 -f null -
```

```
ffmpeg -hwaccel vaapi -hwaccel_output_format vaapi -i ANY_10Bit_HDR_CLIP.mp4 -vf scale_vaapi=1280:692,tonemap_vaapi=format=nv12:t=bt709:m=bt709:p=bt709 -f null -
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

HDR10 to SDR tone-mapping in media delivery.

### Debug Information

1. libva 2.17, gmmlib 22.3.4, media-driver 23.1.3
2. yes renderD128
3. [vainfo.log](https://github.com/intel/media-driver/files/10958206/vainfo.log)
4. [libva_trace.log.141251.zip](https://github.com/intel/media-driver/files/10958240/libva_trace.log.141251.zip)
5. no i915 hang

Guess it's due to the incorrect image size alignment on DG2 platform?
Older platforms are not affected such as KBLx and TGLx.

Other common VPP filters are not affected too.
```
 ... deinterlace_vaapi V->V       Deinterlacing of VAAPI surfaces
 ... denoise_vaapi     V->V       VAAPI VPP for de-noise
 ... overlay_vaapi     VV->V      Overlay one video on top of another
 ... procamp_vaapi     V->V       ProcAmp (color balance) adjustments for hue, saturation, brightness, contrast
 ... scale_vaapi       V->V       Scale to/from VAAPI surfaces.
 ... sharpness_vaapi   V->V       VAAPI VPP for sharpness
 ... transpose_vaapi   V->V       VAAPI VPP for transpose
```

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2023-03-13:
@FurongZhang May need to trouble you to look into it. Thanks in advance.

intel-mediadev, 2023-03-21:
Auto Created VSMGWL-62709 for further analysis.

eero-t, 2023-03-22:
Does not look like an alignment issue, at least not a straight-forward one.  While fast 720 height is divisible by 16, 694 is only by 2, whereas slow 692 height is divisible by 4...

nyanmisaka, 2023-03-22:
Indeed. So I haven't found the pattern of this issue.
Common resolutions works just fine such as 1280x720, 1920x1080 and 3840x2160.

FurongZhang, 2023-03-24:
@nyanmisaka , well received the information, will update the status ww13.

nyanmisaka, 2023-04-07:
> @nyanmisaka , well received the information, will update the status ww13.

Any news on this? I see that the LUT media kernel has been updated recently but it doesn't seem to help this issue.

closed 2023-04-20 by intel-mediadev

FurongZhang, 2023-04-20:
The fix code https://github.com/intel/media-driver/pull/1646 has been merged and upstreamed.

nyanmisaka, 2023-04-20:
@FurongZhang Tested and it works just fine now. Thanks for your help!

nyanmisaka, 2023-05-23:
> If there's a better place to discuss this, apologies - couldn't find a related issue in the jellyfin-ffmpeg or jellyfin github projects.

We can discuss here in jellyfin repo https://github.com/jellyfin/jellyfin-ffmpeg/pull/245

almightiest, 2023-05-25:
> @FurongZhang It's a different issue on ADL-S / 13th Gen. Using `tonemap_vaapi` hangs and resets the i915 kernel driver.
> 
> ```
> [    2.700921] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
> [    2.704972] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.6.4.bin version 70.6.4
> [    2.704974] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
> [    2.719629] i915 0000:00:02.0: [drm] HuC authenticated
> [    2.719633] i915 0000:00:02.0: [drm] GuC submission disabled
> [    2.719636] i915 0000:00:02.0: [drm] GuC SLPC disabled
> [    2.722123] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
> [    2.722747] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
> [    2.780112] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> [    2.844668] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> [   16.962506] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> [   16.967171] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
> [   16.969761] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
> [65426.599038] i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
> [65426.599107] i915 0000:00:02.0: [drm] ffmpeg[1963716] context reset due to GPU hang
> [65426.751730] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:8:2cffbffd, in ffmpeg [1963716]
> [65433.158859] i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
> [65433.158928] i915 0000:00:02.0: [drm] ffmpeg[1963716] context reset due to GPU hang
> [65433.311510] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:8:2cffbffd, in ffmpeg [1963716]
> ```

and in case it helps, my monitor output is via VGA on the ASPEED AST2600, while the Raptor Lake-S GT1 [UHD Graphics 770] doesn't have a monitor attached and has been used as "headless" igpu transcoding for months now.  this is on a Supermicro X13SAE-F with BIOS 2.1, max TOLUD set to 2gb to allow for both graphics to co-exist without a monitor always on (server mode).  the GPU HANG occurs during any `tonemap_vaapi` operations, even with the latest 'master' iHD driver.

nyanmisaka, 2023-05-25:
@almightiest Can you apply this patch to master branch and try again?

https://github.com/intel/media-driver/commit/d00ada51.patch

almightiest, 2023-05-25:
> @almightiest Can you apply this patch to master branch and try again?
> 
> https://github.com/intel/media-driver/commit/d00ada51.patch

Sorry I thought that was applied to master already.  Same issue, though.  Input/Output error.

Sherry-Lin, 2023-05-26:
> @FurongZhang It's a different issue on RPL-S / 13th Gen. ~Using `tonemap_vaapi` hangs and resets the i915 kernel driver.~
> 
> ```
> [    2.700921] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
> [    2.704972] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.6.4.bin version 70.6.4
> [    2.704974] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
> [    2.719629] i915 0000:00:02.0: [drm] HuC authenticated
> [    2.719633] i915 0000:00:02.0: [drm] GuC submission disabled
> [    2.719636] i915 0000:00:02.0: [drm] GuC SLPC disabled
> [    2.722123] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
> [    2.722747] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
> [    2.780112] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> [    2.844668] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> [   16.962506] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> [   16.967171] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
> [   16.969761] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
> [65426.599038] i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
> [65426.599107] i915 0000:00:02.0: [drm] ffmpeg[1963716] context reset due to GPU hang
> [65426.751730] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:8:2cffbffd, in ffmpeg [1963716]
> [65433.158859] i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
> [65433.158928] i915 0000:00:02.0: [drm] ffmpeg[1963716] context reset due to GPU hang
> [65433.311510] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:8:2cffbffd, in ffmpeg [1963716]
> ```
> 
> Edit: `tonemap_vaapi` won't hang the GPU driver but it will fail shortly.
> 
> ```
> frame= 0 fps=0.0 q=0.0 size=N/A time=00:00:00.00 bitrate=N/A speed= 0x
> frame= 0 fps=0.0 q=0.0 size=N/A time=00:00:00.00 bitrate=N/A speed= 0x
> frame= 0 fps=0.0 q=11.0 size=N/A time=00:00:00.00 bitrate=N/A speed= 0x
> frame= 2 fps=0.6 q=25.0 size=N/A time=00:00:00.00 bitrate=N/A speed= 0x
> frame= 4 fps=1.0 q=25.0 size=N/A time=00:00:00.00 bitrate=N/A speed= 0x
> frame= 6 fps=1.2 q=25.0 size=N/A time=00:00:00.25 bitrate=N/A speed=0.0504x
> frame= 8 fps=1.3 q=25.0 size=N/A time=00:00:00.25 bitrate=N/A speed=0.0423x
> frame= 10 fps=1.4 q=25.0 size=N/A time=00:00:00.25 bitrate=N/A speed=0.0365x
> frame= 11 fps=1.5 q=25.0 size=N/A time=00:00:00.64 bitrate=N/A speed=0.085x
> [Parsed_tonemap_vaapi_6 @ 0x557580bb0440] Failed to start picture processing: 1 (operation failed).
> Error while filtering: Input/output error
> Failed to inject frame into filter network: Input/output error
> ```

@nyanmisaka Considering it's different issue, could you create a new one? Thanks.

almightiest, 2023-05-26:
@Sherry-Lin logged #1672