intel/media-driver #1632: [Bug]: media-driver-23.1.5 regresses (compared to 23.1.4) build with undefined reference in the final linking phase

frantisekz, 2023-03-29:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Build media-driver with GCC and : 

```
  -DENABLE_NONFREE_KERNELS:BOOL=OFF \
  -DENABLE_KERNELS:BOOL=OFF \
  -DBUILD_KERNELS:BOOL=OFF \
  -DMEDIA_BUILD_FATAL_WARNINGS=OFF \
  -DVC1_Decode_Supported="no" \
  -DAVC_Encode_VME_Supported="no" \
  -DAVC_Encode_VDEnc_Supported="no" \
  -DHEVC_Encode_VME_Supported="no" \
  -DHEVC_Encode_VDEnc_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -DHEVC_Decode_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -G Ninja
```

23.1.4: build finishes and results in a viable driver
23.1.5: build fails with media-driver-intel-media-23.1.5/media_softlet/agnostic/common/os/mos_utilities.h:2920: undefined reference to `decode::Vp9DecodeMemCompM12::Vp9DecodeMemCompM12(CodechalHwInterface*)'
collect2: error: ld returned 1 exit status

Entire buil log can be found here: https://download.copr.fedorainfracloud.org/results/frantisekz/intel-media-driver-free/fedora-rawhide-x86_64/05719197-intel-media-driver-free/builder-live.log.gz

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Build

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-04-03:
Auto Created VSMGWL-63139 for further analysis.

Jexu, 2023-04-03:
Could you help take a check with above fix?

frantisekz, 2023-04-03:
@Jexu Yes, applying both # https://github.com/intel/media-driver/pull/1635 and # https://github.com/intel/media-driver/pull/1636 solves the issue. Thanks for the quick fix!

PR #1635 (2023-04-04): [Decode] Fix compiling issue for vp9
Fixes: #1632
commit 6573106a (2023-04-04): [Decode] Fix compiling issue for vp9

Fixes: #1632
closed 2023-04-04 by intel-mediadev