intel/media-driver #1638: [Feature]: Encode VP8 on Skylake

AnterCreeper, 2023-04-05:
### What Feature?

Debian 5.18.16-1~bpo11+1 (2022-08-12) x86_64 GNU/Linux
Version: 21.1.1+ds1-1

diff between i965 and intel-media-driver, missing VAProfileVP8Version0_3 : VAEntrypointEncSlice

--- out	2023-04-05 15:46:52.681791117 +0800
+++ out.1	2023-04-05 15:47:09.453879183 +0800
@@ -1,30 +1,36 @@
 vainfo: VA-API version: 1.10 (libva 2.10.0)
-vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.1 ()
+vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.4.1
 vainfo: Supported profile and entrypoints
-      VAProfileNone                   :	VAEntrypointVideoProc
-      VAProfileNone                   :	VAEntrypointStats
       VAProfileMPEG2Simple            :	VAEntrypointVLD
       VAProfileMPEG2Simple            :	VAEntrypointEncSlice
       VAProfileMPEG2Main              :	VAEntrypointVLD
       VAProfileMPEG2Main              :	VAEntrypointEncSlice
+      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
+      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
+      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
+      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
+      VAProfileH264ConstrainedBaseline:	VAEntrypointStats
       VAProfileH264Main               :	VAEntrypointVLD
       VAProfileH264Main               :	VAEntrypointEncSlice
-      VAProfileH264Main               :	VAEntrypointFEI
       VAProfileH264Main               :	VAEntrypointEncSliceLP
+      VAProfileH264Main               :	VAEntrypointFEI
+      VAProfileH264Main               :	VAEntrypointStats
       VAProfileH264High               :	VAEntrypointVLD
       VAProfileH264High               :	VAEntrypointEncSlice
-      VAProfileH264High               :	VAEntrypointFEI
       VAProfileH264High               :	VAEntrypointEncSliceLP
+      VAProfileH264High               :	VAEntrypointFEI
+      VAProfileH264High               :	VAEntrypointStats
+      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
+      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
+      VAProfileH264StereoHigh         :	VAEntrypointVLD
+      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
       VAProfileVC1Simple              :	VAEntrypointVLD
       VAProfileVC1Main                :	VAEntrypointVLD
       VAProfileVC1Advanced            :	VAEntrypointVLD
+      VAProfileNone                   :	VAEntrypointVideoProc
       VAProfileJPEGBaseline           :	VAEntrypointVLD
       VAProfileJPEGBaseline           :	VAEntrypointEncPicture
-      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
-      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
-      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
-      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
       VAProfileVP8Version0_3          :	VAEntrypointVLD
+      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
       VAProfileHEVCMain               :	VAEntrypointVLD
       VAProfileHEVCMain               :	VAEntrypointEncSlice
-      VAProfileHEVCMain               :	VAEntrypointFEI

### What's the usage scenario would be benifited?

Immersive Media

### Do you want to contribute a patch to develop this feature?

None

Sherry-Lin, 2023-04-06:
I'd recommend you to choose more newer platform like KBL serial as below or ICL which supports VP8 encode.
KBLx (KBL: Kaby Lake, CFL: Coffe Lake, WHL: Whiskey Lake, CML: Comet Lake, AML: Amber Lake)

Sherry-Lin, 2023-04-27:
Is it ok to close this issue? I will close it in 2 weeks if no more update. Please feel free to re-open it if any concerns.

closed 2023-07-13 by DenWolf