intel/media-driver #1640: [Bug]: build of 23.1.5 with gcc 13 fails with dangling-pointer

heitbaum, 2023-04-09:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

None

### What happened?

Build media-driver with gcc-13-20230402.
builds fine with gcc-12.2

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Build fails using gcc-13 20230402
```
In member function 'void std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::swap(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, CMRTKernelBase*>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, CMRTKernelBase*> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, CMRTKernelBase*> >]',
    inlined from 'void std::map<_Key, _Tp, _Compare, _Alloc>::swap(std::map<_Key, _Tp, _Compare, _Alloc>&) [with _Key = std::__cxx11::basic_string<char>; _Tp = CMRTKernelBase*; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, CMRTKernelBase*> >]' at /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-12.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/13.0.1/bits/stl_map.h:1172:18,
    inlined from 'virtual MOS_STATUS CodechalFeiHevcStateG9Skl::FreeEncResources()' at ../media_driver/agnostic/gen9_skl/codec/hal/codechal_fei_hevc_g9_skl.cpp:6414:24:
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-12.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/13.0.1/bits/stl_tree.h:2092:36: error: storing the address of local variable 'deallocator' in '*MEM[(struct _Rb_tree_node_base * &)&deallocator + 16].std::_Rb_tree_node_base::_M_parent' [-Werror=dangling-pointer=]
 2092 |           __t._M_root()->_M_parent = __t._M_end();
      |           ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
../media_driver/agnostic/gen9_skl/codec/hal/codechal_fei_hevc_g9_skl.cpp: In member function 'virtual MOS_STATUS CodechalFeiHevcStateG9Skl::FreeEncResources()':
../media_driver/agnostic/gen9_skl/codec/hal/codechal_fei_hevc_g9_skl.cpp:6413:22: note: 'deallocator' declared here
 6413 |      CmKernelMapType deallocator;
      |                      ^~~~~~~~~~~
../media_driver/agnostic/gen9_skl/codec/hal/codechal_fei_hevc_g9_skl.cpp:6413:22: note: 'deallocator.std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*> > >::_M_t.std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<ch# Linker hash-style is set to gnu via gcc default
ar> >, std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*> > >::_M_impl.std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, CMRTKernelBase*> > >::_Rb_tree_impl<std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, true>::<unnamed>.std::_Rb_tree_header::_M_header.std::_Rb_tree_node_base::_M_parent' declared here
cc1plus: all warnings being treated as errors
[537/1453] Building CXX object media_driver/CMakeFiles/iHD_drv_v...CODEC.dir/agnostic/gen11/codec/hal/codechal_decode_vc1_g11.cpp.o
ninja: build stopped: subcommand failed.
```

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-04-10:
Auto Created VSMGWL-63404 for further analysis.

MicroYY, 2023-04-19:
Media driver support latest compiler, but gcc 13 is not formally released. The latest GCC release is 12.2. https://gcc.gnu.org/releases.html
Prereleases could be unstable so we may not support it.
But considering the build error may be real issues as newer compiler may be stricter, we can try to fix it if you can provided GCC13 binary package. Otherwise we will reject this issue.
So do https://github.com/intel/media-driver/issues/1639 and https://github.com/intel/media-driver/issues/1641

frantisekz, 2023-04-19:
> Media driver support latest compiler, but gcc 13 is not formally released. The latest GCC release is 12.2. https://gcc.gnu.org/releases.html Prereleases could be unstable so we may not support it. 

GCC 13.1 will most likely be released in a few days as a stable release, with not much changes from its current state, just for the context.

heitbaum, 2023-04-19:
> Media driver support latest compiler, but gcc 13 is not formally released. The latest GCC release is 12.2. https://gcc.gnu.org/releases.html
> Prereleases could be unstable so we may not support it. [...]

Hi @MicroYY - The GCC 13.1 release candidate announcement can be read on the [GCC mailing list](https://gcc.gnu.org/pipermail/gcc/2023-April/241162.html). The GCC 13.1 release date is penciled in for Wednesday, 26 April. For our package builds, we build the gcc compiler as part of the distribution process.

I would be happy to test patches / run do any other build tests. I have just tested with 20230416 (which was yesterday) but I’ll compile with the rc1 and confirm for us. Snapshots were downloaded from https://gcc.gnu.org/pub/gcc/snapshots

heitbaum, 2023-04-19:
Just tested build of media-driver 23.2.0 with gcc 13.1.0-RC2-20230419 - build errors are still there (as expected)

heitbaum, 2023-04-24:
With #1648 and with gcc-13.1.0-RC2-20230419

Tested-by: Rudi Heitbaum <rudi@heitbaum.com>

MicroYY, 2023-04-24:
> With #1648 and with gcc-13.1.0-RC2-20230419
> 
> Tested-by: Rudi Heitbaum [rudi@heitbaum.com](mailto:rudi@heitbaum.com)

May I know if the compilation errors are fixed?

heitbaum, 2023-04-24:
> > With #1648 and with gcc-13.1.0-RC2-20230419
> > Tested-by: Rudi Heitbaum [rudi@heitbaum.com](mailto:rudi@heitbaum.com)
> 
> May I know if the compilation errors are fixed?

Perfect 👍

PR #1648 (2023-05-29): Fix GCC13 build errors
Fix #1639
Fix #1640
Fix #1641
commit 069c2690 (2023-05-29): Fix GCC13 build errors

Fix #1639
Fix #1640
Fix #1641
closed 2023-05-29 by intel-mediadev