intel/media-driver #1649: [Bug][regression][DG2] commit 2e6afc5dd cause regression encode/avc for ffmpeg-vaapi

Bin-CI, 2023-04-24:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

good commit: 7f07fc849
effect: ffmpeg-vaapi encode/avc part
example error command: 
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt yuv420p -s:v 720x480  -r:v 30 -i input.yuv -vf 'format=nv12,hwupload' -an -c:v h264_vaapi -rc_mode CBR -profile:v main -g 30 -slices 1 -b:v 2000k -maxrate 2000k -low_power 1 -vframes 150 -y output.h264

without IPB keyword catch on application message

Just effect on DG2

### What's the usage scenario when you are seeing the problem?

Video Conference

### Do you want to contribute a patch to fix the issue?

None

Sherry-Lin, 2023-04-24:
@dmitrybrazhkin to take a look.

Bin-CI, 2023-04-25:
issue confirm it not commit cause, it is new feature for the DG2:
add the B- frame as default option which is older process not expect
close issue

closed 2023-04-25 by Bin-CI