intel/media-driver #1650: [Bug]: Media sdk sample(sample_encode) encoding problem with bit rate control doesn't work on the specific yuv file.

HeXianrui, 2023-04-26:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

1. Run Media SDK sample, sample_decode, on Linux platform, to generate **target.yuv file** by decoding the given hevc source. 
Running Command line: ./sample_decode h265 -gpucopy::on -device /dev/dri/card0 -i ./program_2.hevc -o ./**target.yuv** -n 300
You can access the input source file program_2.hevc here:
_**program_2.hevc  ：   https://pan.baidu.com/s/1fjsNCfDrsBoTos-_qm6POQ         Extraction code：7lym**_  
2. Run Media SDK sample , sample_encode , on Linux platform,  bit rate control doesn't work well on the **target.yuv file**. The output bit rate is much higher than the set value.
3. Running command: ./sample_encode  h265  -gpucopy::on  -i ./**target.yuv** -o ./output.h265 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -f 50 -hw -async 1  -u speed -r 1 -g 100 .
4. The input yuv file only contains  300 frames, but with bit rate set value being 10000, framerate being 50, the output.h265 files is about 28MB which is much higher than the expected. 
5. We have tried with cbr , vbr, la and other methods, all failed but except method : **"extbrc:on"** .The output file with  extbrc method on  is about 7.4MB, but the output file has a few totally black frames. We have no idea about it.
6. Please help us find a valid bit rate control method without any problem on the given source . 
 
Thank you very much!

### What's the usage scenario when you are seeing the problem?

Playback, Video Analytics

### What impacted?

intel media sdk version

# /opt/intel/mediasdk/bin/vainfo

libva info: VA-API version 1.12.0
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.12 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.2.3 (008fb3b4)
vainfo: Supported profile and entrypoints

os version :5.10.0-1044-oem     ubuntu1~20.04

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-04-26:
Auto Created VSMGWL-64056 for further analysis.

bai-isaac, 2023-04-27:
hi @HeXianrui , we analyzed the bit stream and encoding commands you provided.
We found that the complexity of this content is very high, and it is a scene similar to Gaussian white noise.
When -b is set to 1m, the maximum quantization parameter 51 has been used during brc encoding.
Even if -b is set smaller, the size of the bitstream will not decrease much, which is expected. This is due to the complexity of the content.

HeXianrui, 2023-04-27:
Hi， @bai-isaac , I got it, Thank you!  But I still confused about one question.
We tryed brc method with extbrc:on in that encode sample, it successfully controls the output bit rate, but with a few totally black frames in the output file. Could you help analyze it? 
Thank you very much!

HeXianrui, 2023-04-27:
> hi @HeXianrui , we analyzed the bit stream and encoding commands you provided. We found that the complexity of this content is very high, and it is a scene similar to Gaussian white noise. When -b [...]

Hi， @bai-isaac , I got it, Thank you! But I still confused about one question.
We tryed brc method with extbrc:on in that encode sample, it successfully controls the output bit rate, but with a few totally black frames in the output file. Could you help analyze it?
Thank you very much!

HeXianrui, 2023-04-27:
> hi @HeXianrui , we analyzed the bit stream and encoding commands you provided. We found that the complexity of this content is very high, and it is a scene similar to Gaussian white noise. When -b [...]

Hi，@bai-issac, one more important thing is that sample works fine with h264 encoder, no bit rate out of control issue.

bai-isaac, 2023-04-27:
hi @HeXianrui  could you please share h264 encoder command here? Thanks!

HeXianrui, 2023-04-27:
> hi @HeXianrui could you please share h264 encoder command here? Thanks!

./sample_encode  h264  -gpucopy::on  -i ./target.yuv -o ./output.h264 -w 1920 -h 1080 -b 10000 -f 50 -hw -async 1 -u speed -r 1 -g 100