intel/media-driver #1651: [Bug]: [regression][DG2][encode/hevc]:  regression catch of 92c00a8

Bin-CI, 2023-04-26:
### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

ffmpeg-vaapi, ffmpeg-libvpl, gst-libvpl, gst-va, gst-vaapi part refer encode/hevc part verify failed, the command without response or command failed

with revert 92c00a8 verify passed

some example command line：
```
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt yuv420p -s:v 3840x2160  -r:v 30 -i input.yuv -vf 'format=nv12,hwupload' -an -c:v hevc_vaapi -rc_mode CBR -low_power 1 -vframes 75 -y output.h265
```
with 92c00a8
```
[in#0/rawvideo @ 0x555d8966d840] Input file #0 (input.yuv):
[in#0/rawvideo @ 0x555d8966d840]   Input stream #0:0 (video): 2 packets read (24883200 bytes); 2 frames decoded; 
[in#0/rawvideo @ 0x555d8966d840]   Total: 2 packets (24883200 bytes) demuxed
[AVIOContext @ 0x555d8966f200] Statistics: 49766400 bytes read, 0 seeks
Conversion failed!
```
revert 92c00a8
```
[in#0/rawvideo @ 0x5648b8c91840] Input file #0 (input.yuv):
[in#0/rawvideo @ 0x5648b8c91840]   Input stream #0:0 (video): 75 packets read (933120000 bytes); 75 frames decoded;
[in#0/rawvideo @ 0x5648b8c91840]   Total: 75 packets (933120000 bytes) demuxed
[AVIOContext @ 0x5648b8c93200] Statistics: 933120000 bytes read, 0 seeks
```
```
ffmpeg -v verbose -hwaccel qsv -init_hw_device qsv=qsv,child_device=/dev/dri/renderD128 -hwaccel_output_format qsv -f rawvideo -pix_fmt yuv420p -s:v 3840x2160  -r:v 30 -i input.yuv -vf 'format=nv12,hwupload=extra_hw_frames=120' -an -c:v hevc_qsv  -low_power 1 -vframes 75 -y output.h265
```
with 92c00a8
```
Output #0, hevc, to 'output.h265':
  Metadata:
    encoder         : Lavf60.5.100
  Stream #0:0: Video: hevc, 1 reference frame, qsv(tv, progressive), 3840x2160 (0x0), q=2-31, 15000 kb/s, 30 fps, 30 tbn
    Metadata:
      encoder         : Lavc60.10.100 hevc_qsv
    Side data:
      cpb: bitrate max/min/avg: 15000000/0/15000000 buffer size: 0 vbv_delay: N/A
frame=    0 fps=0.0 q=0.0 size=       0kB time=-577014:32:22.77 bitrate=  -0.0kbits/s speed=N/A    
[hevc_qsv @ 0x55c75999a700] Unknown FrameType, set pict_type to AV_PICTURE_TYPE_NONE.
```
command line Out of responce
revert 92c00a8
```
Output #0, hevc, to 'output.h265':
  Metadata:
    encoder         : Lavf60.5.100
  Stream #0:0: Video: hevc, 1 reference frame, qsv(tv, progressive), 3840x2160 (0x0), q=2-31, 15000 kb/s, 30 fps, 30 tbn
    Metadata:
      encoder         : Lavc60.10.100 hevc_qsv
    Side data:
      cpb: bitrate max/min/avg: 15000000/0/15000000 buffer size: 0 vbv_delay: N/A
frame=    0 fps=0.0 q=0.0 size=       0kB time=-577014:32:22.77 bitrate=  -0.0kbits/s speed=N/A    
frame=   20 fps=0.0 q=-0.0 size=    1024kB time=00:00:00.56 bitrate=14803.4kbits/s speed=1.12x    
frame=   53 fps= 52 q=-0.0 size=    3328kB time=00:00:01.66 bitrate=16357.8kbits/s speed=1.64x    
[in#0/rawvideo @ 0x560a26e1a540] EOF while reading input
[in#0/rawvideo @ 0x560a26e1a540] Terminating demuxer thread
No more output streams to write to, finishing.
[out#0/hevc @ 0x560a26d63d00] All streams finished
[out#0/hevc @ 0x560a26d63d00] Terminating muxer thread
[AVIOContext @ 0x560a26cac040] Statistics: 4584738 bytes written, 0 seeks, 18 writeouts
```

### What's the usage scenario when you are seeing the problem?

Video Conference

### What impacted?

encode/hevc part of vaapi & libvpl on DG2

### Debug Information

HW: DG2 A380
Kernel: https://launchpad.net/ubuntu/+source/linux/6.2.0-18.18
firmware: https://launchpad.net/ubuntu/+source/linux-firmware/20230323.gitbcdcfbcf-0ubuntu1

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-04-27:
Auto Created VSMGWL-64073 for further analysis.

uartie, 2023-04-27:
This also broke AVC encode on DG2.  Please fix asap.

LhGu, 2023-05-03:
Hi， @Bin-CI  @uartie  Do you use  **ENABLE_PRODUCTION_KMD=ON** config to build our driver for DG2?  I think my code only modify non-production build.

LhGu, 2023-05-04:
@uartie @Bin-CI We could not reproduce this issue in DG2. Could you try https://github.com/intel/media-driver/pull/1655 fix, Maybe it could fix our issue. 
BTW, Is your dgpu DG2 on platform mtl?

Bin-CI, 2023-05-04:
Just verify on the DG2 the issue already fixed on #1655 for the commit: 337647563b4e5045600eb40f8c4c990f0e3e96b9
Current we disable the iGPU for the verify part
the build option, we use the default option expect add `-DINSTALL_DRIVER_SYSCONF=OFF`

LhGu, 2023-05-04:
Thanks, @Bin-CI  Could you try the new code in https://github.com/intel/media-driver/pull/1655, I changed my fix code because some perf impact.

Bin-CI, 2023-05-04:
verify for c05b6348370bc7f4bce2258f268eb0bc263e37aa issue fixed

LhGu, 2023-05-04:
@Bin-CI Thanks for your great support!  We will merge it asap.

PR #1655 (2023-05-05): Add code for old kernel compatibility
Fixed #1651
commit fd375cf2 (2023-05-05): Add code for old kernel compatibility

Fixed #1651
closed 2023-05-05 by intel-mediadev