intel/media-driver #1652: [Bug]: Gentoo Linux(musl + llvm no-gcc) Build Failed >23.1.5

meleehao, 2023-04-29:
### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

ia_softlet/linux/common/os/mos_os_specific_next.cpp.o
/home/sv/source/source-code/media-driver-intel-media-23.1.6/media_softlet/linux/common/os/osservice/mos_utilities_specific.cpp:2552:23: error: use of undeclared identifier 'backtrace'
            int num = backtrace(stack, ((sizeof(traceBuf)-nLen)/sizeof(void *)));
                      ^
[ 92%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/__/media_softlet/linux/common/os/mos_os_mock_adaptor_specific.cpp.o
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/__/media_softlet/linux/common/os/mos_oca_rtlog_mgr_specific.cpp.o
1 error generated.
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/__/media_softlet/linux/common/os/mos_vma.c.o
gmake[2]: *** [media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/build.make:398: media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/__/media_softlet/linux/common/os/osservice/mos_utilities_specific.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/mos_decompression_base.cpp.o
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/mos_mediacopy_base.cpp.o
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/mos_user_setting_specific.cpp.o
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/mos_os_mock_adaptor_specific.cpp.o
/home/sv/source/source-code/media-driver-intel-media-23.1.6/media_softlet/linux/common/os/osservice/mos_utilities_specific.cpp:2552:23: error: use of undeclared identifier 'backtrace'
            int num = backtrace(stack, ((sizeof(traceBuf)-nLen)/sizeof(void *)));
                      ^
[ 93%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/mos_oca_rtlog_mgr_specific.cpp.o
1 error generated.
gmake[2]: *** [media_driver/CMakeFiles/iHD_drv_video_mos.dir/build.make:748: media_driver/CMakeFiles/iHD_drv_video_mos.dir/__/media_softlet/linux/common/os/osservice/mos_utilities_specific.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:443: media_driver/CMakeFiles/iHD_drv_video_mos.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:469: media_driver/CMakeFiles/iHD_drv_video_mos_softlet.dir/all] Error 2
​gmake: *** [Makefile:156: all] Error 2

### What's the usage scenario when you are seeing the problem?

Web browser, Others

### Do you want to contribute a patch to fix the issue?

None

meleehao, 2023-04-29:
For libbacktrace.so It belong package that sys-libs/libbacktrace and Installed .

intel-mediadev, 2023-05-08:
Auto Created VSMGWL-64295 for further analysis.

wennibao, 2023-07-04:
Hi @meleehao could you pls try this PR https://github.com/intel/media-driver/pull/1690

wennibao, 2023-07-20:
Hi @meleehao, this bug cannot be reproduced locally for environment issue. You can verify this PR https://github.com/intel/media-driver/pull/1690. I will close this ticket first, and if you still have concern after trying fix PR, feel free to reopen this bug.

Sherry-Lin, 2023-07-20:
Thanks @MicroYY @wennibao provided the PR. 
@meleehao I'm going to close this issue cause no feedback in recent 2 weeks. Could you try PR https://github.com/intel/media-driver/pull/1690 if it's convenience for you? Please feel free to re-open it if any issues found.

closed 2023-07-20 by Sherry-Lin