intel/media-driver #1653: [Bug]: HEVC/H.265 not showing up under vainfo

dontcrash, 2023-05-01:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

According to this release note from Intel, it should work:
 [https://community.intel.com/t5/Graphics/Intel-Iris-and-HD-Graphics-Driver-update-posted-for-Haswell-and/td-p/456967](https://community.intel.com/t5/Graphics/Intel-Iris-and-HD-Graphics-Driver-update-posted-for-Haswell-and/td-p/456967)

CPU: i5-4278U

From what I understand, it should be able to do 8bit HEVC (not 10bit pipeline)

sudo vainfo:
```
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2023-05-01:
i5-4278U [Haswell](https://ark.intel.com/content/www/us/en/ark/products/codename/42174/haswell.html) is not supported by this driver.

https://github.com/intel/media-driver#supported-platforms

bkuhls, 2023-05-01:
According to https://www.cpu-monkey.com/en/igpu-intel_iris_graphics_5100-74 your CPU does not support hardware-accelerated HEVC de-/encode.

closed 2023-05-06 by XinfengZhang