intel/media-driver #1654: [Bug]: every app  using the intel-media-driver tries to create /etc/igfx_user_feature{,_next}.txt

darix, 2023-05-03:
### Is it regression? Good in old configuration?

None

### What happened?

Found via running the apps in apparmor. it even tries to open the file in write mode. which makes me wonder ... should distributions install those 2 files? if not ... can the code that tries to write those files properly guarded that it is not built during normal building?

```
type=AVC msg=audit(1682983399.735:1932): apparmor="DENIED" operation="mknod" class="file" profile="vlc" name="/etc/igfx_user_feature.txt" pid=18661 comm="vlc" requested_mask="c" denied_mask="c" fsuid=1000 ouid=1000
type=AVC msg=audit(1682983399.735:1933): apparmor="DENIED" operation="mknod" class="file" profile="vlc" name="/etc/igfx_user_feature_next.txt" pid=18661 comm="vlc" requested_mask="c" denied_mask="c" fsuid=1000 ouid=1000
```

If those files should be user writeable ... /etc/ seems to be a weird path for them.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

-

### Debug Information

intel-media-23.1.3

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-05-04:
Auto Created VSMGWL-64146 for further analysis.

VickyZengg, 2023-05-18:
Hi @darix， May I know what version of driver you are using? Release, debug or release internal?

VickyZengg, 2023-05-18:
Hi @darix ，I checked the issue in code, we will create "/etc/igfx_user_feature.txt" and "/etc/igfx_user_feature_next.txt" in release/debug/release internal version. So you find you use release version but it still open the two files. Because those two files is for regkey write and read. We have regkey used in release version not only in debug and release internal. So, the two files need create in release version. May I know if you still have concern about it?

VickyZengg, 2023-05-19:
Thank you for raising the issue, we will consider it and reply to you as soon as possible.

VickyZengg, 2023-06-09:
Hello, we already add code to guard igfx_user_feature.txt and igfx_user_feature_next.txt auto create in release version, if you want to use the files, you can create it manually. But it still under review and test. After merge will let you know. :) Thank you!

closed 2023-06-13 by intel-mediadev