intel/media-driver #1658: [Feature]: VC-1 decoding support for DG2

mitzsch, 2023-05-07:
### What Feature?

Currently, VC-1 decoding support is not available on Linux on DG2 devices even though they support it and decoding of vc1 works with Windows. Intel reference document also shows VC-1 support for DG2 cards. => https://www.intel.com/content/www/us/en/docs/onevpl/developer-reference-media-intel-hardware/1-1/overview.html#DECODE-OVERVIEW-DISCRETE

### What's the usage scenario would be benifited?

Transcode for media delivery, Playback

### What impacted?

Having VC-1 decoding support is necessary for playback and also transcoding scenarios. 

Accelerating playback is nice, but more importantly for transcoding purposes. The sw decoder for vc-1 is limited to a single thread and therefore limiting transcodes quite a bit - having a GPU that could accelerate this is a must. As it already works on Windows I don´t really understand why it's not supported on Linux (yet).

Would be awesome if it could be added in a future release. Thanks.

### Do you want to contribute a patch to develop this feature?

None

intel-mediadev, 2023-07-13:
Auto Created VSMGWL-66369 for further analysis.

Jexu, 2023-07-14:
suppose that is issue in document. On all xe platforms, media driver has no support for vc1 decode anymore which is not our POR feature. If you anyone still want to use it, you may need to enable it by self, but it is not commitment without any potential issues.

mitzsch, 2023-07-23:
@Jexu 
>  you may need to enable it by self

How do I do this? Is this some compiler flag that needs to be added?

Jexu, 2023-07-26:
The feature is disabled through ddi report.
You could try to enable it in https://github.com/intel/media-driver/blob/master/media_driver/linux/Xe_M/ddi/media_sku_wa_xe.cpp#L83C13-L83C13

nyanmisaka, 2023-07-26:
It fails when calling `vaCreateContext()` and the commit 295b5b3 says DG2's VC1 kernel requires cm rt.

> vc1 decode needs to call cm interface to init olp kernel.

But from the header file, it seems the cm rt currently only supports up to TGLx/DG1.
https://github.com/intel/media-driver/tree/master/cmrtlib/agnostic/share

@Jexu Is it possible to fix this by adding headers just like 2bb733b? Or is it no-trivial and requires more changes in cmrtlib?

Jexu, 2023-07-28:
cm rt is also abandoned from dg2 which is used to init olp kernel. 
you could try to add it back, or refer to cm rt to init olp kernel where it is required.

mitzsch, 2023-09-24:
May I ask if there is anything new on this one? @Jexu

bobby3605, 2023-12-22:
@Jexu If you can point me in the right direction for fixing vc1 decode, I can try to fix it and submit a PR. I'm not sure what the exact issue is, so any background information you can provide me will be helpful.

Jexu, 2023-12-27:
@bobby3605 Currently, vc1 caps is disabled in media sku and cm rt is abandoned as well. Both are what you need to fix.

First one is easy to enable in media sku mentioned in previous comment, but cm rt is not available on DG2. It is only an interface to load binary kernel, you probobaly could refer to the code logic for tgl e.g to add it back in anywhere you want and send pr out for a review.

But for a note, even you could submit a PR finally, I think it is not easy to be accepted and merged into media driver again since it is an abandoned feature. You may need to maintain it as a private change/ feature etc.

nyanmisaka, 2024-08-23:
I submitted a simple fix for this. https://github.com/intel/media-driver/pull/1846

If any of you are interested you can give it a try. @mitzsch @bavdevc @bobby3605 @Jexu