intel/media-driver #1662: [Bug]: vaCreateContext fails for VAEntrypointVideoProc in intel-media-va-driver

tykel, 2023-05-12:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. Using Ubuntu 22.04. This is a native app.
2. This can be reproduced with libva-utils VPP samples. But originally a custom application is where this bug was found.
3. vaCreateContext fails with error -1 (OPERATION_FAILED) with intel-media-va-driver 22.3.1.
4. vaCreateContext succeeds and returns a valid context with intel-media-va-driver-non-free 22.3.1.

I have attached a sample app to showcase it. Can be compiled using `gcc test.c -o test -lX11`.
[test.c.txt](https://github.com/intel/media-driver/files/11467320/test.c.txt)

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

This necessitates users to install intel-media-va-driver-non-free, which I am not sure is required -- surely intel-media-va-driver can do some sort of VPP?

### Debug Information

1. intel-media-va-driver 22.3.1, as packaged by Ubuntu.
2. `ls /dev/dri` shows `by-path`,`card0`,`renderD128`
3. Graphics are Ice Lake Iris Plus G7 rev 07. (CPU is Ice Lake i5-1065G7)
4. Attached.
5. Attached.
6. N/A
[libva_trace.log.204706.thd-0x00005e8d.log](https://github.com/intel/media-driver/files/11467310/libva_trace.log.204706.thd-0x00005e8d.log)
[vainfo.log](https://github.com/intel/media-driver/files/11467311/vainfo.log)

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

tykel, 2023-05-12:
Is this related to https://github.com/intel/media-driver/issues/1132 and its fix in https://github.com/intel/media-driver/commit/66fa14da59ba6a81f1d604d1c71fa93502884599 ?

LhGu, 2023-05-15:
@tykel  Could you try latest media driver . Maybe [intel-media-23.2.1](https://github.com/intel/media-driver/releases/tag/intel-media-23.2.1) is ok.   
I will also try latest driver.

intel-mediadev, 2023-05-15:
Auto Created VSMGWL-64589 for further analysis.

LhGu, 2023-05-15:
Hi, latest driver is ok when using your sample cpp. I will try 22.3.1.

tykel, 2023-05-15:
@LhGu I tried 23.2.1 built from source with -DENABLE_NONFREE_SHADERS=OFF and it also works for me.
I will do the same and try a 22.3.1 build as above.

tykel, 2023-05-15:
And back on the original Ice Lake machine, the 22.3.1 build fails in the vaCreateContext call. I'll try successive media-driver releases to bisect to the first good one.

LhGu, 2023-05-15:
Hi, @tykel  I have root caused. Our free shaders files are not updated when we changed our non-free shaders files . It will cause free shaders with wrong vpkrnheader.h which is shared by  free and non-free.   

The index of IDR_VP_LinkFile kernel in vpkrnheader.h file is always changing. If we want to use full free shader, we must rebuild all free shader file and vpkrnheader.h instead of using the exist file(vpkrnheader.h and free shader file).

tykel, 2023-05-15:
Hi @LhGu , thank you. Is this something we can get addressed in the Ubuntu/Debian package builds?

LhGu, 2023-05-15:
We updated non-free shaders a lot of times. However, only one person has updated full free shaders file. I will sync with him tomorrow.
His commit id: 469f8c9f0d497320772629c912464018da653876. Tag: intel-media-22.6.5  I think we could use full-free driver after  intel-media-22.6.5

LhGu, 2023-05-16:
I have synced with some related people. After this https://github.com/intel/media-driver/commit/469f8c9f0d497320772629c912464018da653876 patch, this issue has been fixed. We have also enabled some internal cases to test full-free kernel.
Tag after intel-media-22.6.5, full-free shaders work.
@tykel  Could you help to close this issue? We have fixed this issue in this https://github.com/intel/media-driver/commit/469f8c9f0d497320772629c912464018da653876 patch

tykel, 2023-05-16:
I will verify on my end @LhGu, thanks.

In the meantime I have filed https://bugs.launchpad.net/ubuntu/+source/intel-media-driver/+bug/2019831

tykel, 2023-05-16:
Confirming it works for me with 22.6.5, so closing. Thanks