intel/media-driver #1663: [Bug]: confusion between intel-media-va-driver and intel-media-va-driver-non-free

XinfengZhang, 2023-05-15:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. media driver is a open source project, it is "free" to use, copy, modify , publish ...  following MIT license. 
2. Debian package the driver with intel-media-va-driver and intel-media-va-driver-non-free which is generated by build options.  the differences is about EU kernel source code. 

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

I suppose media driver should clarify that people should not pay anything to using intel-media-va-driver.  word "free" has different meanings to different understanding,  should not use standard open source community understanding to represent all.  so , suggest some clarification in README.
and from personal perspective, I would like people always use non-free, because it is "free to use..." and with more features.  

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2023-05-15:
In Debian, "non-free" usually means that the source code contains binary blobs that cannot be read directly by humans. Similar to EU shaders, most linux-firmware are also known as "non-free".

IMHO for hardware features, it's pointless to sacrifice functionality for the title "free". One have paid for graphics card, he should be eligible for full drivers.

So users should only choose the "free" version if they are aware of the risks that some hardware features are not available in this driver. Otherwise the "non-free" one is recommended and you don't have to pay anything.

For some other distros, there is another matter. Some of them deliberately disable H.264 and HEVC codecs in the build stage to avoid patent issues.

tykel, 2023-05-16:
Maybe the intel-media-va-driver-non-free package would have been better named intel-media-va-driver-proprietary, to indicate there is some closed-source code involved? I think everyone knows the packages are free as in "free beer", but in the Linux/OSS circles it usually indicates free as in "libre".

XinfengZhang, 2023-05-16:
> In Debian, "non-free" usually means that the source code contains binary blobs that cannot be read directly by humans. Similar to EU shaders, most linux-firmware are also known as "non-free".
>
> IMHO [...]
yes,  fully agree

> For some other distros, there is another matter. Some of them deliberately disable H.264 and HEVC codecs in the build stage to avoid patent issues.

yes, we enabled such build option , if they want to release a product without such codec support , such as #1643

bavdevc, 2023-06-07:
which software version is included in the windows driver package? I think that's even worse than the difference between intel-media-va-driver and intel-media-va-driver-non-free.

Intel windows driver get new media-driver and onevpl features without a version bump in the readme files (txt/pdf) - open source media driver for linux and the other platforms has a clear changelog in github and you know what is included in every release/tag - but they are lagging a bit behind compared to the windows version

nyanmisaka, 2023-06-08:
There's no such issue in Windows world. You are unlikely to get a detailed changelog for Windows driver as it is closed source. Only the MFX/VPL API version is documented.

As far as I know VPL and OpenCL runtime share code between Windows and Linux. UMD on Windows depends on DirectX/DXVA, so it should have a different version than the Linux media-driver.

In addition, WHQL and internal testing process also slow down the release speed of Windows driver.

Sherry-Lin, 2024-01-08:
We do receive some feedback for the confusion of `free`, so the readme file https://github.com/intel/media-driver?tab=readme-ov-file#components-and-features has been updated as below. This issue will be closed. Please feel free to re-open it if any concerns.

`About Ubuntu/Debian OSV, they provide [intel-media-va-driver-non-free](https://packages.ubuntu.com/disco/intel-media-va-driver-non-free) (Full feature build) and [intel-media-va-driver](https://packages.ubuntu.com/disco/intel-media-va-driver) (Free kernel build) two packages. Free here means open source kernel but not related to fee need to pay. `

closed 2024-01-08 by Sherry-Lin