intel/media-driver #1666: [Bug]: AVC: Low power: Pipeline stack smashing(terminated) if number of ROI's are greater than 4

VunnySodhi, 2023-05-18:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1) In Linux pull DL Streamer docker image from https://dlstreamer.github.io/get_started/install/install_guide_ubuntu.html#option-2-pull-and-run-intel-dl-streamer-pipeline-framework-docker-image 
2) Run following pipeline 
gst-launch-1.0 filesrc location=test.mp4 ! qtdemux ! h264parse ! vaapih264dec ! video/x-raw\(memory:VASurface\) !  gvadetect device=CPU pre-process-backend=vaapi model=FP32/face-detection-retail-0004.xml model-proc=face-detection-retail-0004.json ! vaapih264enc tune=low-power ! h264parse ! mp4mux ! fakesink
3) Issue is not coming if we modify above command with vaapih264enc tune=none

**Resources:**
**Video file:** https://cdn.videvo.net/videvo_files/video/free/video0488/large_watermarked/_import_629da40d4fc106.16643143_preview.mp4
**Face-detection-retail-004 model** files can be downloaded using: https://www.intel.com/content/www/us/en/developer/articles/technical/video-analytics-service.html#inpage-nav-3-1 

### What's the usage scenario when you are seeing the problem?

Video Analytics

### What impacted?

Not impacting any milestone as of now, only performance impact. 

### Debug Information

1) Libva: 2.17.0.2-44 / gmmlib: 22.3.3+i550 / media-driver: 23.1.0+i553
2) ls /dev/dri
card0  renderD128
3) lspci -nn |grep  -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
4) Attached vinfo.log
5) Attached libva_trace.log files
6) No dmesg seen (not attached)

### Do you want to contribute a patch to fix the issue?

None

VunnySodhi, 2023-05-18:
Attached vainfo and libvatrace files.
[libva_trace.log.083318.thd-0x00000055.txt](https://github.com/intel/media-driver/files/11506250/libva_trace.log.083318.thd-0x00000055.txt)
[vainfo.log](https://github.com/intel/media-driver/files/11506251/vainfo.log)
[libva_trace.log.083317.thd-0x00000053.txt](https://github.com/intel/media-driver/files/11506253/libva_trace.log.083317.thd-0x00000053.txt)
[libva_trace.log.083318.thd-0x0000005b.txt](https://github.com/intel/media-driver/files/11506254/libva_trace.log.083318.thd-0x0000005b.txt)

VunnySodhi, 2023-05-18:
Found out issue is coming from https://github.com/intel/media-driver/blob/intel-media-23.1.0/media_driver/agnostic/gen12/hw/vdbox/mhw_vdbox_vdenc_g12_X.h#L1768
When I run pipeline: on #line 1768 avcPicParams->NumROI is coming as 5 whereas on #line 1772 we are allocating array of 3(ENCODE_VDENC_AVC_MAX_ROI_NUMBER_G9), that why stack is corrupted and pipeline terminates after that. 

Not sure we can we increase range of array to ENCODE_VDENC_AVC_MAX_ROI_NUMBER_ADV as its G12 and I am also running on Tigerlake machine.

VunnySodhi, 2023-05-23:
Hi @XinfengZhang and @Jexu any update on this issue?

VunnySodhi, 2023-06-01:
Hi @Jexu / @XinfengZhang any update on raised issue?

DenWolf, 2023-06-06:
Hi @VunnySodhi ,
We will take a look at this issue, most likely on the next week (based in engineering Team bandwidth).

VunnySodhi, 2023-06-07:
Thanks for the update @DenWolf

intel-mediadev, 2023-06-07:
Auto Created VSMGWL-65418 for further analysis.

dmenshov, 2023-06-12:
Hi @VunnySodhi ,
I prepared patch for the problem.
You could check if it is work for you
You should add the codtion to https://github.com/intel/media-driver/blob/intel-media-23.1.0/media_driver/agnostic/gen12/hw/vdbox/mhw_vdbox_vdenc_g12_X.h#L1768
like this:
`if (!params->bVdencBRCEnabled && avcPicParams->NumROI && avcPicParams->bNativeROI)`

dmenshov, 2023-06-13:
Hi @VunnySodhi,
Did you have a chance to check if the solution works for you?

VunnySodhi, 2023-06-14:
Hi @dmenshov  I have tried changing https://github.com/intel/media-driver/blob/intel-media-23.1.0/media_driver/agnostic/gen12/hw/vdbox/mhw_vdbox_vdenc_g12_X.h#L1768 to **(!params->bVdencBRCEnabled && avcPicParams->NumROI && avcPicParams->bNativeROI)** 
It worked, pipeline works fine with changes and got proper output.

Sherry-Lin, 2023-06-14:
@dmenshov could you help provide a patch for it?

dmenshov, 2023-06-14:
THe patch is on review now. It will back merged to opensource after finishing testing.

dmenshov, 2023-06-15:
https://github.com/intel/media-driver/commit/6f483dd9fb430c439dc324fc105a1e0312d1c9c0 merged

Sherry-Lin, 2023-06-15:
Thanks @dmenshov. I'm going to close this issue which has been fixed by https://github.com/intel/media-driver/commit/6f483dd9fb430c439dc324fc105a1e0312d1c9c0. Please feel free to re-open it if any concerns.

closed 2023-06-15 by Sherry-Lin

VunnySodhi, 2023-06-15:
Thanks all for fix.