intel/media-driver #1671: [Bug]: libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed

himanshurohilla17, 2023-05-26:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

I installed Ubuntu 18.04. I wrote a program on it, whose job was to receive data from the socket, decode, and render it. Everything is working fine. For this purpose, I installed the following packages : 

**libva2 libva-x11-2 libva-drm2**

Now I have to do this work on Ubuntu 22.04, so I installed a minimal version of Ubuntu 22.04. I also installed above mentioned packages on Ubuntu 22.04. 

After installation of these packages, I ran "vainfo" command on the terminal but its output contains an error i.e. 
**libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed**

I also compiled mentioned program on my development machine. It was successfully compiled. But when I deployed it on the target machine, the program crashed after receiving data from the socket and trying to decode it. I ran the program using GDB then GDB also printed the same error before the crash. 

I have given almost all information in **Debug Information** section
                                                                                                       

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

[vainfo.log](https://github.com/intel/media-driver/files/11574307/vainfo.log)
vainfo.log file is attached. 

### Output of lspci -nn |grep  -Ei 'VGA|DISPLAY'
`00:02.0 VGA compatible controller [0300]: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller [8086:22b1] (rev 35)`

### Output of ls /dev/dri
`by-path  card0  renderD128`

### Do you want to contribute a patch to fix the issue?

None

MicroYY, 2023-06-01:
You may need to build LibVa, Gmm and then build media driver to keep compatibility.
https://github.com/intel/media-driver/wiki/Build-media-driver-from-source-code

Jexu, 2023-07-19:
Please refer to the wiki to update/ install latest libva, gmmlib and media driver to try again while imcompatible vertion will lead media driver init failure.

I will close this issue, and please let me know or re-open it again if you still have any concern or other questions.

closed 2023-07-19 by Jexu