intel/media-driver #1672: [Bug][ADL-S 13th Gen][VPP]: tonemap_vaapi fails on certain input size 1280*(722 vs 720)

almightiest, 2023-05-26:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Use a file with HDR10, like a sample file from [here](https://www.demolandia.net/downloads.html?id=654828652) - click on "<star> Download the UHD HDR10 - Sony Bravia OLED File <star>" link on that page to download/save.

Then run the following 2 commands on i7-13700k or similar iGPU on ubuntu lunar with kernel 6.2 and the latest master media-driver:
1. 1920x802 scale_vaapi - **fails input/output error on 23.2.3 (master) - slow fps but no failures on 23.1.6**
`ffmpeg -analyzeduration 200M -init_hw_device vaapi=va:,driver=iHD,kernel_driver=i915 -init_hw_device qsv=qs@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -autorotate 0 -i file:"/path/to/sample.mkv" -autoscale 0 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=1920:h=802:extra_hw_frames=24,tonemap_vaapi=format=nv12:p=bt709:t=bt709:m=bt709" -f null -`

2. 1920x800 scale_vaapi - **succeeds >200fps on 23.2.3 (master) and 23.1.6**
`ffmpeg -analyzeduration 200M -init_hw_device vaapi=va:,driver=iHD,kernel_driver=i915 -init_hw_device qsv=qs@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -autorotate 0 -i file:"/path/to/sample.mkv" -autoscale 0 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=1920:h=800:extra_hw_frames=24,tonemap_vaapi=format=nv12:p=bt709:t=bt709:m=bt709" -f null -`

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

Any video with HDR* which requires tone mapping to SDR will fail with various croppings, when using VAAPI.

### Debug Information

1. media-driver master (with and without #1628 applied), libva-dev/jammy,now 2.18.0.2-54~u22.04, vainfo/jammy,now 2.17.1.2-1~u22.04, libigdgmm-dev/jammy,now 22.3.5-601~22.04
2. `crw-rw---- 1 root render 226, 128 May 25 09:49 /dev/dri/renderD128`
3. `00:02.0 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
0b:00.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 52)`
4. [vainfo.log](https://github.com/intel/media-driver/files/11574617/vainfo.log)
5. [libva_trace.log.111747.zip](https://github.com/intel/media-driver/files/11574633/libva_trace.log.111747.zip)
6. [dmesg.log](https://github.com/intel/media-driver/files/11574646/dmesg.log)

### Do you want to contribute a patch to fix the issue?

No.

nyanmisaka, 2023-06-06:
@almightiest Can you try this patch and see if it works?

https://github.com/intel/media-driver/pull/1679.patch

closed 2023-06-06 by intel-mediadev

nyanmisaka, 2023-06-06:
@almightiest Nice to hear that. We will also include the fix in our future releases.