intel/media-driver #1681: [Bug]: Failed to map output buffers: 24 (internal encoding error).

TrueWodzu, 2023-06-09:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

Ubuntu 22.10
custom built ffmpeg with vaapi support. Command to execute:
`sudo ./ffmpeg -loglevel debug -vaapi_device /dev/dri/renderD128 -f v4l2 -framerate 30 -video_size 1920x1080 -i /dev/video2 -vf 'format=p010,hwupload' -c:v hevc_vaapi output_2_vaapi.mp4`

Debug output:
```

sicore@sicore-desktop:~/bin$ sudo ./ffmpeg -loglevel debug -vaapi_device /dev/dri/renderD128 -f v4l2 -framerate 30 -video_size 1920x1080 -i /dev/video2 -vf 'format=p010,hwupload' -c:v hevc_vaapi output_2_vaapi.mp4 -y
ffmpeg version N-111056-gb7e79ba395 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Ubuntu 12.2.0-3ubuntu1)
  configuration: --pkg-config-flags=--static --enable-static --disable-shared --prefix=/home/sicore/ffmpeg_build --bindir=/home/sicore/bin --extra-cflags=-I/home/sicore/ffmpeg_build/include --extra-ldflags=-L/home/sicore/ffmpeg_build/lib --extra-cflags=-I/opt/intel/mediasdk/include --extra-ldflags=-L/opt/intel/mediasdk/lib --extra-ldflags=-L/opt/intel/mediasdk/plugins --enable-vaapi --enable-opencl --disable-debug --enable-libdrm --enable-gpl --enable-runtime-cpudetect --enable-libx264 --enable-openssl --enable-pic --extra-libs='-lpthread -lm -lz -ldl' --enable-nonfree
  libavutil      58. 12.100 / 58. 12.100
  libavcodec     60. 17.100 / 60. 17.100
  libavformat    60.  6.100 / 60.  6.100
  libavdevice    60.  2.100 / 60.  2.100
  libavfilter     9.  8.101 /  9.  8.101
  libswscale      7.  3.100 /  7.  3.100
  libswresample   4. 11.100 /  4. 11.100
  libpostproc    57.  2.100 / 57.  2.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-vaapi_device' ... matched as option 'vaapi_device' (set VAAPI hardware device (DirectX adapter index, DRM path or X11 display name)) with argument '/dev/dri/renderD128'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'v4l2'.
Reading option '-framerate' ... matched as AVOption 'framerate' with argument '30'.
Reading option '-video_size' ... matched as AVOption 'video_size' with argument '1920x1080'.
Reading option '-i' ... matched as input url with argument '/dev/video2'.
Reading option '-vf' ... matched as option 'vf' (set video filters) with argument 'format=p010,hwupload'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'hevc_vaapi'.
Reading option 'output_2_vaapi.mp4' ... matched as output url.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Applying option vaapi_device (set VAAPI hardware device (DirectX adapter index, DRM path or X11 display name)) with argument /dev/dri/renderD128.
[AVHWDeviceContext @ 0x565246c84700] libva: VA-API version 1.16.0
[AVHWDeviceContext @ 0x565246c84700] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x565246c84700] libva: Found init function __vaDriverInit_1_16
[AVHWDeviceContext @ 0x565246c84700] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x565246c84700] Initialised VAAPI connection: version 1.16
[AVHWDeviceContext @ 0x565246c84700] Format 0x41524742 -> bgra.
[AVHWDeviceContext @ 0x565246c84700] Format 0x42475241 -> argb.
[AVHWDeviceContext @ 0x565246c84700] Format 0x41424752 -> rgba.
[AVHWDeviceContext @ 0x565246c84700] Format 0x52474241 -> abgr.
[AVHWDeviceContext @ 0x565246c84700] Format 0x58524742 -> bgr0.
[AVHWDeviceContext @ 0x565246c84700] Format 0x42475258 -> 0rgb.
[AVHWDeviceContext @ 0x565246c84700] Format 0x58424752 -> rgb0.
[AVHWDeviceContext @ 0x565246c84700] Format 0x52474258 -> 0bgr.
[AVHWDeviceContext @ 0x565246c84700] Format 0x30335241 -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x30334241 -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x30335258 -> x2rgb10le.
[AVHWDeviceContext @ 0x565246c84700] Format 0x30334258 -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x36314752 -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x56555941 -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x56555958 -> vuyx.
[AVHWDeviceContext @ 0x565246c84700] Format 0x30303859 -> gray.
[AVHWDeviceContext @ 0x565246c84700] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x565246c84700] Format 0x3132564e -> unknown.
[AVHWDeviceContext @ 0x565246c84700] Format 0x32595559 -> yuyv422.
[AVHWDeviceContext @ 0x565246c84700] Format 0x59565955 -> uyvy422.
[AVHWDeviceContext @ 0x565246c84700] Format 0x32315659 -> yuv420p.
[... 196 lines cut ...]
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
```
**5.**
Nothing recorded.

**6.**
GPU did not hang.

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-06-12:
Auto Created VSMGWL-65527 for further analysis.

OttoYang, 2023-07-25:
Hi @TrueWodzu 
Is it also reproduced in the ffmpeg official release version?  
If so could you please attach your input file here for us to investigate?

TrueWodzu, 2023-07-25:
Hi @OttoYang thank you for your interest. Does the official version support vaapi? I thought ffmpeg must be built with `--enable-vaapi` option? Or do you mean that you want me to build ffmpeg from a stable branch?

OttoYang, 2023-07-26:
Yes, I would like to confirm what did you modify in your custom build, if only adding `--enable-vaapi`, could you share your base branch? I'll try to reproduce it later.

tprzedmojski, 2023-09-30:
Hi @TrueWodzu,
I had a very similar issue on Intel Celeron N4500. What helped is installing and loading GuC/HuC firmware. [It is a known limitation of the driver](https://github.com/intel/media-driver#known-issues-and-limitations) that it needs proprietary firmware to handle low power HEVC encoding (`VAEntrypointEncSliceLP`). Following this gist was helpful: [https://gist.github.com/Brainiarc7/aa43570f512906e882ad6cdd835efe57](https://gist.github.com/Brainiarc7/aa43570f512906e882ad6cdd835efe57).

OttoYang, 2023-10-30:
Przedmojski, thank you for your helpful suggestion.
Hi @TrueWodzu ,
May I know if this issue still exists after the solution mentioned by Przedmojski?

TrueWodzu, 2023-10-30:
Hi @tprzedmojski thank you for your suggestion,  I had no chance to test it though as I am no longer working for that company.
Hi @OttoYang thank you for your support, as mentioned above, I am no longer working for that company so I have no hardware to test Tomek's suggestion. Feel free to close the issue if you preffer. Thanks!

OttoYang, 2023-10-30:
Thank you so much for your reply.  
And if you encounter any other encoding issues in other projects, please feel free to report issues.

closed 2023-10-30 by OttoYang