intel/media-driver #1683: [Bug]: GetFormat() returns wrong value for VP9

dwlsalmeida, 2023-06-13:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

In the VP9 specification (section 7.2), one can see that VP9 supports the following subsamplings for profile 1:

```
YUV 4:2:2, YUV 4:4:0 or YUV 4:4:4
```

`media-driver` assumes 4:4:4 subsampling [by default](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/codec/ddi/media_ddi_decode_vp9.cpp#L606).

This then [conflicts](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp#L684) with the [format set in the media surface](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva.cpp#L2640).

This makes `vaEndPicture` return `VA_INVALID_PARAMETER` which tends to abort calling programs.

This can be reproduced with the following GStreamer pipeline:

```
gst-launch-1.0 filesrc location=vp91-2-04-yuv422.webm ! parsebin !  vavp9dec ! fakevideosink
```

Where vp91-2-04-yuv422 can be downloaded from here: https://storage.googleapis.com/downloads.webmproject.org/test_data/libvpx/vp91-2-04-yuv422.webm

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

dwlsalmeida, 2023-07-03:
@XinfengZhang Hi, this breaks quite a lot of libvpx streams. There's a GStreamer reproducer, can we get any thoughts on this issue?

Jexu, 2023-07-19:
Add spec description here for reference.

_VP9 supports the option to output pixels using either 8, 10, or 12 bits per color sample.
In profiles 0 and 1, only 8 bits per color sample is allowed.
In profiles 2 and 3, only greater than 8 bits per color sample is allowed.

VP9 supports 4 profiles:
Profile, Bit depth, SRGB Colorspace support, Chroma subsampling
0 8 No YUV 4:2:0
1 8 Yes YUV 4:2:2, YUV 4:4:0 or YUV 4:4:4
2 10 or 12 No YUV 4:2:0
3 10 or 12 Yes YUV 4:2:2, YUV 4:4:0 or YUV 4:4:4_

intel-mediadev, 2023-07-19:
Auto Created VSMGWL-66617 for further analysis.

Jexu, 2023-07-21:
Hi, do you mean vp9 422/440 format support?
In fact, you can get supported format from feature list, and currently media driver only support 420, 444 format decoding

dwlsalmeida, 2023-07-28:
@Jexu ok then

Jexu, 2023-07-31:
So, let me close this jira if solving your question. Feel free to re-open it again with any other questions.

closed 2023-07-31 by Jexu