intel/media-driver #1685: [Bug]: Corrupted H264 Encode on DG2

VMFortress, 2023-06-16:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

It seems DG2 GPUs create corrupted output when using hardware encode on Chromium-based browsers. This does not appear to be an issue on other Intel platforms, including Skylake.

**Steps to Reproduce:**
_(Prerequisites: webcam or similar device, DG2 GPU)_
1) Use latest libva, gmmlib, and media-driver compiled from source as of June 15th, 2023.
2) Launch a Chromium 114 based browser with the following command: `LIBVA_DRIVER_NAME=iHD chromium --enable-accelerated-video-encode --enable-features=VaapiVideoEncoder`
3) Navigate to https://codec-compare.glitch.me/
4) Leave all settings at default and hit 'Start'
5) Allow webcam permissions in Chromium
6) Observe input compared to single frame of corrupted output
7) Very low usage (typically <1%) of Video and VideoEnhance engines can be seen in `intel_gpu_top`.

The issue appears in all H264 codecs available but does not occur on AV1 or VP9 codecs in same case.

Additional debugging details provided below.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

Any websites or Electron-based applications using H264 encoding will be unable to use acceleration. A mainstream example is Discord screenshare running within Chromium.

### Debug Information

**Logs:**
- DG2 (A380)
  - [Linux kernel version](https://github.com/intel/media-driver/files/11775702/dg2.uname.log)
  - [chromium version](https://github.com/intel/media-driver/files/11775715/dg2.chromium_version.log)
  - [intel-media-driver version](https://github.com/intel/media-driver/files/11775717/dg2.intel-media-driver_version.log)
  - [libva version.](https://github.com/intel/media-driver/files/11775725/dg2.libva_version.log)
  - [libva-utils version](https://github.com/intel/media-driver/files/11775727/dg2.libva-utils_version.log)
  - [intel-gmmlib version](https://github.com/intel/media-driver/files/11775731/dg2.intel-gmmlib_version.log)
  - [ls /dev/dri output](https://github.com/intel/media-driver/files/11775735/dg2.dev_dri.log)
  - [lspci -nn |grep  -Ei 'VGA|DISPLAY' output](https://github.com/intel/media-driver/files/11775737/dg2.hardware_info.log)
  - [vainfo output](https://github.com/intel/media-driver/files/11775752/dg2.vainfo.log)
  - LIBVA_TRACE output
    - [First](https://github.com/intel/media-driver/files/11775756/dg2.log.152814.thd-0x0000a1d0.txt)
    - [Second](https://github.com/intel/media-driver/files/11775757/dg2.log.152817.thd-0x0000a29f.txt)
- Skylake (6700HQ)
  - [Linux kernel version](https://github.com/intel/media-driver/files/11775759/skylake.uname.log)
  - [chromium version](https://github.com/intel/media-driver/files/11775765/skylake.chromium_version.log)
  - [intel-media-driver version](https://github.com/intel/media-driver/files/11775766/skylake.intel-media-driver_version.log)
  - [libva version](https://github.com/intel/media-driver/files/11775768/skylake.libva_version.log)
  - [libva-utils version](https://github.com/intel/media-driver/files/11775769/skylake.libva-utils_version.log)
  - [intel-gmmlib version](https://github.com/intel/media-driver/files/11775771/skylake.intel-gmmlib_version.log)
  - [ls /dev/dri output](https://github.com/intel/media-driver/files/11775778/skylake.dev_dri.log)
  - [lspci -nn |grep  -Ei 'VGA|DISPLAY' output](https://github.com/intel/media-driver/files/11775779/skylake.hardware_info.log)
  - [vainfo output](https://github.com/intel/media-driver/files/11775781/skylake.vainfo.log)
  - LIBVA_TRACE output
    - [First](https://github.com/intel/media-driver/files/11775783/skylake.log.165627.thd-0x00005aac.txt)
    - [Second](https://github.com/intel/media-driver/files/11775785/skylake.log.165636.thd-0x00005b46.txt)

**Additional debug info:**
- Software setups were identical on both with Arch-based systems
- Fresh arch installs on both systems with identically matched steps produced same results
- Chromium logs showed no difference between platforms, even at max verbosity
- No other special provisions for Skylake, Gen9, DG2, Alchemist, or Gen12 could be found in the chromium source that would affect this issue
- No dmesg issues for i915 appear on either system, GuC and HuC authenticate properly
- Behaviour is identical on both A380 and A750
- Behaviour is identical on Skylake on either iHD or i965
- Behaviour is identical on Fedora 38
- Behaviour is identical on a Kaby Lake system to Skylake system
- Behaviour is identical on all chromium versions as far back as v100
- Behaviour is identical on intel-media-driver v23.1.0
- Behaviour is identical on Xwayland and X11
- Behaviour is identical on Gnome and KDE desktop environments
- Behaviour is identical on Skylake whether enabling or disabling `VaapiLowPowerEncoderGen9x` in chromium
- Behaviour is identical on DG2 using various other combinations of chromium flags

I am happy to provide any other details requested or peform any tests that may help.

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2023-06-17:
You don't need a webcam. Using screen recording plugin can trigger the issue on DG2.
https://chrome.google.com/webstore/detail/screen-recorder/hniebljpgcogalllopnjokppmgbhaden

And it produces broken bitstream that cannot be decoded by FFmpeg. 
[screen-capture.7z.zip](https://github.com/intel/media-driver/files/11778458/screen-capture.7z.zip)

```
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
[h264 @ 0000021658e5c940] cbp too large (3199971767) at 3 0
[h264 @ 0000021658e5c940] error while decoding MB 3 0
[h264 @ 0000021658e5c940] concealing 8160 DC, 8160 AC, 8160 MV errors in P frame
Input #0, matroska,webm, from 'R:\screen-capture (1).webm':
  Metadata:
    encoder         : Chrome
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0(eng): Video: h264 (Constrained Baseline), yuv420p(progressive), 1920x1080, SAR 1:1 DAR 16:9, 30.30 fps, 25 tbr, 1k tbn (default)
```

I doubt there could be some incompatible parameters between the [chromium::h264_vaapi_video_encoder_delegate.cc](https://github.com/chromium/chromium/blob/main/media/gpu/vaapi/h264_vaapi_video_encoder_delegate.cc) and the DG2 settings in iHD driver.

Sherry-Lin, 2023-06-19:
@daijh could you help take a look?

daijh, 2023-06-27:
@VMFortress could you report the issue to chromium as well?
Issues - chromium: https://bugs.chromium.org/p/chromium/issues/list

daijh, 2023-06-27:
Could you provide the detailed chromium logs, and perhaps VA logs on DG2?

- chromium logs
add below extra flags into cmdline.
`--vmodule=*/vaapi/*=4,*/media/*=4 --enable-logging=stderr`

- va logs
Chromium Docs - VA-API: https://chromium.googlesource.com/chromium/src/+/refs/heads/main/docs/gpu/vaapi.md

VMFortress, 2023-06-28:
@daijh Chromium issue can be found here: https://bugs.chromium.org/p/chromium/issues/detail?id=1458942

Running `LIBVA_TRACE=./dg2_trace.log chromium --enable-accelerated-video-encode --enable-features=VaapiVideoEncoder --vmodule=*/vaapi/*=4,*/media/*=4 --enable-logging=stderr > ./dg2_chromium.log 2>&1`:
- Chromium `about:gpu` page: [about_gpu.txt](https://github.com/intel/media-driver/files/11897922/about_gpu.txt)
- Chromium verbose logs: [dg2_chromium.log](https://github.com/intel/media-driver/files/11897924/dg2_chromium.log)
- Libva trace logs: [dg2_trace.log.zip](https://github.com/intel/media-driver/files/11897927/dg2_trace.log.zip)

daijh, 2023-06-30:
Sorry, I don't have a DG2 at hand.
The logs showed the iHD driver has error to create the RGBA VASurface.
There are indeed a few fixes were landed on both Chromium and iHD regarding to this specify issue.

[1] https://github.com/intel/media-driver/issues/1210
[2] https://chromium-review.googlesource.com/c/chromium/src/+/4637884

Suggest to include those patches to check this issue.

VMFortress, 2023-07-10:
@daijh I had the chance to test again with the latest intel-media-driver and chromium 117.0.5875.0 which should contain the patches mentioned. Unfortunately, there doesn't not seem to be any improvement. Doing the canonical test again, VP9 encodes fine but H264 results in the corrupted output on DG2 but not on Skylake.

Here are the same logs as before for this test:
- Chromium verbose logs: [dg2_chromium.log](https://github.com/intel/media-driver/files/12008814/dg2_chromium.log)
- Libva trace logs: [dg2_trace.log.zip](https://github.com/intel/media-driver/files/12008813/dg2_trace.log.zip)

DenWolf, 2023-07-13:
Hi @VMFortress ,

Please help to try this patch - if it will resolve this issue, we will prepare the official one.

[potential_fix_for_ChromeOs.patch](https://github.com/intel/media-driver/files/12037616/potential_fix_for_ChromeOs.patch)

Need to change one line - https://github.com/intel/media-driver/blob/03f4cde9fb8a88b9b12a9edd5fd5b43f71226ee5/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp#L1299

from "seqParams->GopPicSize = seq->intra_period;"
to "seqParams->GopPicSize = seq->intra_period **? seq->intra_period : seq->intra_idr_period**;"

If it will not help - we will continue the analysis.

Best regards,
Denis

intel-mediadev, 2023-07-13:
Auto Created VSMGWL-66368 for further analysis.

nyanmisaka, 2023-07-13:
> [potential_fix_for_ChromeOs.patch](https://github.com/intel/media-driver/files/12037616/potential_fix_for_ChromeOs.patch)

It does the trick. No issue anymore in my aforementioned screen recording plugin testing.

VMFortress, 2023-07-13:
@DenWolf I can confirm this resolves all my test cases as well!

DenWolf, 2023-07-13:
Hi @VMFortress , @nyanmisaka ,
Fix has been merged into Media driver repo - https://github.com/intel/media-driver/commit/02a5604273ecf385efca0acf16679aec650c95b7

Please help to verify it (or the latest Media Driver OS code-base - for now it's the latest commit) on your side again, thank you.

Best regards,
Denis

VMFortress, 2023-07-14:
@DenWolf I compiled the latest source and can confirm again it seems to be working in all my cases. 

Thanks for the support! It is very appreciated!

nyanmisaka, 2023-07-14:
@DenWolf So for so good. Thanks for your help!

Sherry-Lin, 2023-07-14:
Fixed in https://github.com/intel/media-driver/commit/02a5604273ecf385efca0acf16679aec650c95b7 so close it. Please feel free to reopen it if the issue is not resolved on your environment.

closed 2023-07-14 by Sherry-Lin

DenWolf, 2023-07-14:
Great! @VMFortress , @nyanmisaka  - thank you very much for your feedback and verification support!