intel/media-driver #1696: [Bug]: Maximum slice size not worked on i5-12500

Warpmh, 2023-07-11:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Hi,

This problem exist in Linux(libva 2.19.0.pre1 , media-driver|oneVPL|onevpl-gpu latest from git 11.07.2023 ).

./sample_encode h264 -i ./hd.yuv -o ./h.h264 -w 1280 -h 720 -b 512 -mss 1400
CONFIGURE LOADER: required implementation: hw
CONFIGURE LOADER: required implementation mfxAccelerationMode: MFX_ACCEL_MODE_VIA_VAAPI
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
Loaded Library configuration:
    Version: 2.9
    ImplName: mfx-gen
    Adapter number : 0
    Adapter type: integrated
    DRMRenderNodeNum: 128
Used implementation number: 0
Loaded modules:
   0: /usr/lib/x86_64-linux-gnu/libmfx-gen.so.1.2.9

Encoding Sample Version 8.4.27.0

Input file format	YUV420
Output video		AVC
Source picture:
	Resolution	1280x720
	Crop X,Y,W,H	0,0,1280,720
Destination picture:
	Resolution	1280x720
	Crop X,Y,W,H	0,0,1280,720
Frame rate	30.00
Bit rate(Kbps)	512
Gop size	0
Ref dist	1
Ref number	0
Idr Interval	0
Target usage	balanced
Memory type	system
Media SDK impl		hw
Media SDK version	2.9

Processing started

[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), SynchronizeFirstTask, SyncOperation fail or timeout at /inst/oneVPL/tools/legacy/sample_encode/src/pipeline_encode.cpp:178

[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), GetFreeTask, m_TaskPool.SynchronizeFirstTask failed at /inst/oneVPL/tools/legacy/sample_encode/src/pipeline_encode.cpp:2230

[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), Run, m_pmfxENC->EncodeFrameAsync failed at /inst/oneVPL/tools/legacy/sample_encode/src/pipeline_encode.cpp:2478

[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), main, pPipeline->Run failed at /inst/oneVPL/tools/legacy/sample_encode/src/sample_encode.cpp:1927
Frame number: 0
Encoding fps: -nan

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

encoding

### Debug Information

libva 2.19.0.pre1 , media-driver|oneVPL|onevpl-gpu latest from git 11.07.2023

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-07-12:
Auto Created VSMGWL-66322 for further analysis.

bai-isaac, 2023-07-13:
Can't reproduce this issue.
Could you please help to dump driver log? Thanks!
Wiki is here: https://github.com/intel/media-driver/wiki/Decode-and-Encode-Debug-Tool-and-surface-dump#how-to-dump-driver-log

Warpmh, 2023-07-13:
[data.zip](https://github.com/intel/media-driver/files/12040160/data.zip)
Hi. In attach.

bai-isaac, 2023-07-14:
hi @Warpmh Thanks for your data.
Looks like your case use a noise YUV, or something going wrong during coding. Could you please help to share the YUV input you use?
Besides, we found you are encoding a all intra stream, from codechal dump. This is different from the parameter you wrote above. Could you please to double check about the parameter? Thanks!

Warpmh, 2023-07-14:
Hi. Yes is noise generated by command cat /dev/urandom > ./hd.yuv . I try with normal yuv source and yes that worked, but totally not comply with target bitrate:
./sample_encode h264 -i ./hd.yuv -o ./d.h264 -w 1280 -h 720 -b 128 -vcm
result size is 658439b at 815fps
./sample_encode h264 -i ./hd.yuv -o ./d.h264 -w 1280 -h 720 -b 128 -vcm -mss 1200
result size is 3154166b at 137fps

On same hd.yuv certainly.

And more tests with hd.yuv from /dev/urandom:

./sample_encode h264 -i ./hd.yuv -o ./d.h264 -w 1280 -h 720 -b 128 -vcm -mss 1200
Its worked result size 213Mb at 79fps

./sample_encode h264 -i ./hd.yuv -o ./d.h264 -w 1280 -h 720 -b 128 -vcm
result size 258Kb at 942fps

./sample_encode h264 -i ./hd.yuv -o ./d.h264 -w 1280 -h 720 -b 512 -vcm -mss 1200
Totally not worked with sts=MFX_ERR_UNDEFINED_BEHAVIOR

Best Regards,
Warp.

Warpmh, 2023-07-14:
After some research I find where problem in my program. If link with libmfx then maximum slice size not work, if link with libmfx-gen then all ok.

Sherry-Lin, 2023-07-17:
It seems issue root caused, could you help close this issue? Please let us know if any other issues found.

Warpmh, 2023-07-17:
Yes is closed. Root cause in libmfx-gen. Thanx.

closed 2023-07-17 by Warpmh