intel/media-driver #1698: [Bug]: vaGetConfigAttributes returns wrong value of cu_qp_delta in VaConfigAttribValEncHevcFeatures with low-power mode

mengker33, 2023-07-20:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

In hevc encoding with low-power mode, vaGetConfigAtttibutes return an incorrect value (i.e. 2)of cu_qp_delta in VaConfigAttribValEncHevcFeatures, which causes artifacts in the encoded clip.
See the wrong one:
[image]

If returned value is 3 in low-power, the result is correct:
See the correct one:
[image]

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

It impacts the whole hevc encoding in low-power mode, simply means it can possibly impact every listed selection above.

### Do you want to contribute a patch to fix the issue?

None

Sherry-Lin, 2023-07-20:
Is it common issue or specific to which platform?

mengker33, 2023-07-20:
> Is it common issue or specific to which platform?

I think it is a common one as I can reproduce on tgl as well as on mtl.

intel-mediadev, 2023-07-20:
Auto Created VSMGWL-66648 for further analysis.

mengker33, 2023-07-26:
Hey, any updates?

bai-isaac, 2023-07-26:
hi mengker33, which va version do you use? and could you please share how to reproduce this issue? Thanks!

mengker33, 2023-07-26:
> hi mengker33, which va version do you use? and could you please share how to reproduce this issue? Thanks!

I use libva 2.19.0. 
The issue can be reproduced using gstreamer-va low-power hevc encoder, i.e. vah265lpenc
<pre>gst-launch-1.0 -vf videotestsrc num-buffers=30 ! video/x-raw,format=NV12 ! vah265lpenc ! h265parse ! filesink location=output.h265 </pre>
You can see the output.h265 has the artifacts as above one.

OttoYang, 2023-07-26:
Hi @mengker33 
We return this param `cu_qp_delta` as an enable flag not as depth.  
The depth only supports 0 or 3 due to HW limitation.
So you may need to change the code in gstreamer as shown below:  
```c
pps->cu_qp_delta_enabled_flag = !!features.bits.cu_qp_delta;
pps->diff_cu_qp_delta_depth = sps->log2_diff_max_min_luma_coding_block_size == 3 ？sps->log2_diff_max_min_luma_coding_block_size: 0;
```

mengker33, 2023-07-26:
Understood！ Appreciate your quick response :)

OttoYang, 2023-07-27:
Hi @mengker33
May I know if the fix works for you?

mengker33, 2023-07-27:
Yes, I will modify codes on GStreamer side
BTW, one more question, why need to set depth to maximum when this flag is enabled?

OttoYang, 2023-07-27:
> Yes, I will modify codes on GStreamer side BTW, one more question, why need to set depth to maximum when this flag is enabled?

Hardware only supports 0 or 3(LCU-64) in low-power mode now. So other depths may lead unexpected behavior.

mengker33, 2023-07-27:
Got it. Thank you, let me close the ticket.

closed 2023-07-27 by mengker33