intel/media-driver #1700: [Bug]: File descriptor leak for  Alder Lake-S GT1

skinkie, 2023-07-23:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

1. Using Qt6 I am using the FFmpeg backend to play video's.
2. I have observed that on Alder Lake-S GT1, /dev/renderR128 is leaking its file descriptor using the iHD VA code.
3. I have been unable to reproduce this issue on for example _CoffeeLake-H GT2_ or older.

I am more than happy to share more details, but at this point in time I am very puzzled.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

Eventually the application crashes because it is out of file descriptors.1

### Debug Information

1. intel-media-driver-23.3.0-r0
2. They exists
3. 00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] [8086:4682] (rev 0c)
4. [vainfo.log](https://github.com/intel/media-driver/files/12138536/vainfo.log)
5. I'll try this later.

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-07-28:
Auto Created VSMGWL-66819 for further analysis.

Jexu, 2023-07-28:
Thanks for reporting, looks that we may need more infomation to know its background, like how to reproduce, where it exists or which descriptor etc.

skinkie, 2023-07-28:
> Thanks for reporting, looks that we may need more infomation to know its background, like how to reproduce, where it exists or which descriptor etc.

For me it is extremely 'difficult' to reproduce in a way we as software developers can actually look at it. I have one system that absolutely fails every time the binary is ran under the condition that it is _not_ ran under valgrind, and is happening _on screen_. When I run this under valgrind I end up with a single file descriptor, obviously _some_ errors.

```
gooitv:/proc/3294/fd$ ls -la | grep renderD128 | wc -l
554
```

Using the same image on an other system older Intel CPU nothing strange is happening. So the background this is an alpine-linux (edge) system (using musl libc), having the latest intel and  Qt6 libraries. I am obviously happy to create a absolutely minimal reproducible case, which can be run using qml6.

[valgrind.txt](https://github.com/intel/media-driver/files/12192317/valgrind.txt)