intel/media-driver #1706: [Bug]: VP9 black screen on DG2

vid-bin, 2023-07-27:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Take an av1_QSV video file and encode vp9_QSV based on that. Results in playback black screen in browser. When playing on VLC seeking will take a very long time but eventually work.

this happens on Linux with recent drivers (~amd64 gentoo) with a DG2 arc a380.

I can supply more information if needed.

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

DenWolf, 2023-07-28:
Hello @vid-bin ,

Based on description, you are using AV1->VP9 transcoding scenario (AV1 decode + VP9 encode).
- Could you please share the command-line which you are using?
- Could you please help to share original AV1 stream?

vid-bin, 2023-07-28:
Yes, that's correct. I want to use VP9 for lower resolution videos and for device compatibility and AV1 on higher resolution videos. Peertube does this by using the last resolution transcoded, which happens to be AV1.

Base AV1:

`/usr/bin/ffmpeg -i /var/www/peertube/storage/videos/private/stuff-1440.mp4 -y -acodec libopus -vcodec av1_qsv -threads 12 -f mp4 -movflags faststart -max_muxing_queue_size 1024 -map_metadata -1 -b:a 128k -strict -2 -vbr on -vf scale=w=-2:h=1080 -pix_fmt yuv420p10le -extbrc 1 -b_strategy 1 -look_ahead 1 -g 600 -preset 1 -profile 1 -tile_cols 1 -tile_rows 1 -minrate 10k -b:v 4000k -maxrate 6000k /var/www/peertube/storage/tmp/transcoded.mp4`

The VP9_QSV command is very similar. I don't have it on hand but will provide it later on over this weekend.

I'll try getting you an example video stream.

vid-bin, 2023-07-28:
Right, so because of copyright reasons I don't really want to provide a clip on github. But you can replicate this on youtube videos if you use yt-dlp and convert them. I'm sure there is a creative commons one floating around.

VP9_QSV:

`/usr/bin/ffmpeg -i /var/www/peertube/versions/peertube-develop/storage/videos/private/uuidhere-1440.mp4 -y -acodec libfdk_aac -vcodec vp9_qsv -threads 10 -f mp4 -movflags faststart -max_muxing_queue_size 1024 -map_metadata -1 -b:a 128k -strict -2 -vbr 2 -cutoff 18000 -af volume=-2dB -vf scale=w=-2:h=480 -pix_fmt yuv420p -b:v 1750k -minrate 0k -maxrate 1750k -g 300 -profile 1 -preset 1 -tile_cols 1 -tile_rows 1`

DenWolf, 2023-08-03:
Hello @vid-bin ,
Thank you for sharing the extended information!
As I right understand from your comments:
- you run your first command-line for AV1_QSV and everything is working good (so, AV1 Encoder is OK)
- then you run your second command-line for VP9_QSV and see the issue on DG2 (potential issue in VP9 Encoder)

Some more questions:
- do you know the original codec for the input MP4 steams - "stuff-1440.mp4" and "uuidhere-1440.mp4" ? Is it AVC or AV1 or ...?

vid-bin, 2023-08-03:
With some of my testing it would go 4k60 vp9 (original) -> 1440p60 qsv av1 -> 720p/480p 60 qsv vp9.

The av1 encode is OK. It’s only the vp9 encode on qsv with the problem.

intel-mediadev, 2023-08-03:
Auto Created VSMGWL-67065 for further analysis.

leyu-yao, 2024-05-21:
Since it is a qsv issue, could you help to confirm whether this commit helps? https://github.com/intel/vpl-gpu-rt/commit/89edcbacb69b94bc4272598837718ffda523f380