intel/media-driver #1715: [Bug]: SIGBUS during vaInitialize after HEVC encode session

kkartaltepe, 2023-08-27:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

1. Start OBS (After v30 beta 3, or built from head)
2. Set recording to QSV HEVC encoder
3. Record for 2s.
4. Set recording to QSV AV1 encoder
5. Start recording
6. Observe SIGBUS during MFXCreateSession's call to `vaInitialize`.

Similar issues (SIGSEG in `vaInitialize`) occur with using H264 encoding and then HEVC encoding. (but H264 then AV1 then HEVC works).

I don't know if it was a regression. Its also unclear if this could be an error in the vpl code calling into this driver.

### What's the usage scenario when you are seeing the problem?

Content Creation, Game Streaming

### What impacted?

Linux users are unable to use streaming/recording software without crashes.

### Debug Information

`onevpl 2023.3.1-1`, `onevpl-intel-gpu 23.3.1-1`, `intel-gmmlib-git 22.3.10.r0.g2864421-1` and `intel-media-driver-git 2023.3.1.r57.gc7cd5a683-1` built from head today.

Arc 750 GPU ` [Arc A750] [8086:56a1] (rev 08)` on a drm-tip kernel `6.5.0-rc6-1-drm-tip-git-g7a825a06c6ee`

The stack trace from HEVC to AV1

```
#0  0x00007ffff3d5dfa0 in  () at /usr/lib/libc.so.6
#1  0x00007fff76fbcb68 in XMHW_STATE_HEAP_INTERFACE::InitializeInterface(MHW_STATE_HEAP_SETTINGS) (this=0x5555586b0d70, StateHeapSettings=...)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:1083
#2  0x00007fff76fbcf1c in Mhw_StateHeapInterface_Create(_MHW_STATE_HEAP_INTERFACE**, MHW_STATE_HEAP_SETTINGS) (ppStateHeapInterface=<optimized out>, StateHeapSettings=...)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:680
#3  0x00007fff767f3fb5 in MhwRenderInterface::AllocateHeaps(MHW_STATE_HEAP_SETTINGS) (this=<optimized out>, stateHeapSettings=...)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/agnostic/common/hw/mhw_render_legacy.c:51
#4  0x00007fff767f98a9 in XRenderHal_Platform_Interface_Legacy::AllocateHeaps(_RENDERHAL_INTERFACE*, MHW_STATE_HEAP_SETTINGS)
    (this=<optimized out>, pRenderHal=<optimized out>, MhwStateHeapSettings=...)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/agnostic/common/renderhal/renderhal_platform_interface_legacy.cpp:845
#5  0x00007fff76f71385 in RenderHal_AllocateStateHeaps(_RENDERHAL_INTERFACE*, _RENDERHAL_STATE_HEAP_SETTINGS*) (pRenderHal=0x555557b14690, pSettings=0x555557b14738)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/renderhal/renderhal.cpp:1444
#6  0x00007fff76f6fd35 in RenderHal_Initialize(_RENDERHAL_INTERFACE*, _RENDERHAL_SETTINGS*) (pRenderHal=0x555557b14690, pSettings=<optimized out>)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/renderhal/renderhal.cpp:5106
#7  0x00007fff767faf40 in RenderCopyState::Initialize() (this=0x555558003a20)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/agnostic/common/shared/mediacopy/media_render_copy.cpp:109
#8  0x00007fff7691cc1a in McpyDeviceXe_Hpm::Initialize(_MOS_INTERFACE*, MhwInterfaces*) (this=0x55555861b810, osInterface=0x5555577ec780, mhwInterfaces=0x7fffd4004a10)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/media_interface/media_interfaces_dg2/media_interfaces_dg2.cpp:380
#9  0x00007fff769a0bfa in McpyDevice::CreateFactory(_MOS_OS_CONTEXT*) (osDriverContext=osDriverContext@entry=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/media_interfaces/media_interfaces.cpp:462
#10 0x00007fff7675d331 in MosMediaCopy::MosMediaCopy(_MOS_OS_CONTEXT*) (this=this@entry=0x55555791b780, mosCtx=mosCtx@entry=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/os/mos_mediacopy.cpp:37
#11 0x00007fff76786e97 in MosUtilities::MosNewUtil<MosMediaCopy, _MOS_OS_CONTEXT*&>(_MOS_OS_CONTEXT*&) ()
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:2998
#12 OsContextSpecificNext::Init(void*) (this=0x55555817cf00, ddiDriverContext=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/linux/common/os/mos_context_specific_next.cpp:278
#13 0x00007fff7695ab13 in DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*, bool&)
    (ctx=ctx@entry=0x55555b1453d0, devicefd=0, major_version=major_version@entry=0x0, minor_version=minor_version@entry=0x0, apoDdiEnabled=@0x7fffffffbb9f: false)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:1801
#14 0x00007fff7695b8b8 in DdiMedia__Initialize(VADriverContext*, int*, int*) (minor_version=0x0, major_version=0x0, ctx=0x55555b1453d0)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:1703
#15 __vaDriverInit(VADriverContext*) (ctx=0x55555b1453d0) at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:7483
#16 __vaDriverInit_1_19(VADriverContextP) (ctx=0x55555b1453d0) at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:7509
#17 0x00007ffff29019f9 in  () at /usr/lib/libva.so.2
#18 0x00007ffff2904c20 in vaInitialize () at /usr/lib/libva.so.2
#19 0x00007fff9e0a1cf9 in  () at /usr/lib/libmfx-gen.so.1.2.9
#20 0x00007fff9e0a29c0 in  () at /usr/lib/libmfx-gen.so.1.2.9
#21 0x00007fff8d58fea5 in  () at /usr/lib/libvpl.so.2
#22 0x00007fff8d598936 in MFXCreateSession () at /usr/lib/libvpl.so.2
```

The second stack with H264 encoding followed by HEVC encoding.
```
SIGSEGV, Segmentation fault.
0x00007fff769a0cfb in McpyDevice::CreateFactory (osDriverContext=osDriverContext@entry=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/media_interfaces/media_interfaces.cpp:465
465	/home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/media_interfaces/media_interfaces.cpp: No such file or directory.
(gdb) bt
#0  0x00007fff769a0cfb in McpyDevice::CreateFactory(_MOS_OS_CONTEXT*) (osDriverContext=osDriverContext@entry=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/media_interfaces/media_interfaces.cpp:465
#1  0x00007fff7675d331 in MosMediaCopy::MosMediaCopy(_MOS_OS_CONTEXT*) (this=this@entry=0x55555634d390, mosCtx=mosCtx@entry=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/os/mos_mediacopy.cpp:37
#2  0x00007fff76786e97 in MosUtilities::MosNewUtil<MosMediaCopy, _MOS_OS_CONTEXT*&>(_MOS_OS_CONTEXT*&) ()
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:2998
#3  OsContextSpecificNext::Init(void*) (this=0x55555774ca30, ddiDriverContext=0x7fffffff4c80)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_softlet/linux/common/os/mos_context_specific_next.cpp:278
#4  0x00007fff7695ab13 in DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*, bool&)
    (ctx=ctx@entry=0x555557464320, devicefd=0, major_version=major_version@entry=0x0, minor_version=minor_version@entry=0x0, apoDdiEnabled=@0x7fffffffbb9f: false) at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:1801
#5  0x00007fff7695b8b8 in DdiMedia__Initialize(VADriverContext*, int*, int*) (minor_version=0x0, major_version=0x0, ctx=0x555557464320)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:1703
#6  __vaDriverInit(VADriverContext*) (ctx=0x555557464320)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:7483
#7  __vaDriverInit_1_19(VADriverContextP) (ctx=0x555557464320)
    at /home/kk/.cache/yay/intel-media-driver-git/src/media-driver/media_driver/linux/common/ddi/media_libva.cpp:7509
#8  0x00007ffff29019f9 in  () at /usr/lib/libva.so.2
#9  0x00007ffff2904c20 in vaInitialize () at /usr/lib/libva.so.2
#10 0x00007fff9e0a1cf9 in  () at /usr/lib/libmfx-gen.so.1.2.9
#11 0x00007fff9e0a29c0 in  () at /usr/lib/libmfx-gen.so.1.2.9
#12 0x00007fff8d58fea5 in  () at /usr/lib/libvpl.so.2
#13 0x00007fff8d598936 in MFXCreateSession () at /usr/lib/libvpl.so.2
```

### Do you want to contribute a patch to fix the issue?

No.

kkartaltepe, 2023-08-29:
Oh sorry it seems the second issue is even simpler and occurs if you just start and stop the h264 encoder once or twice. e.g. start, let record for 2 seconds, stop, start again immediately.

kkartaltepe, 2023-09-21:
At least one issue appears to be https://github.com/intel/media-driver/blob/5d6b96014a3277d28a6ce1fbf4bbb1266ed9eeda/media_driver/linux/common/ddi/media_libva.cpp#L1578
( from the stack `#4  0x00007fff7695ab13 in DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*, bool&)
    (ctx=ctx@entry=0x555557464320, devicefd=0, major_version=major_version@entry=0x0, minor_version=minor_version@entry=0x0, apoDdiEnabled=@0x7fffffffbb9f: false)`)

While technically true that fd 0 can be legal, it is far more likely that libva has passed in an uninitialized fd.

kkartaltepe, 2023-09-22:
After figuring out how to enable the various layers of debug settings in the source, this appears due to incorrect caching of buffer handles in `mos_bufmgf.c`.

I don't see any handling of closing the device (but didnt look very hard), and when starting the encoding session a second time with a new device fd (new VADisplay) old buffers are reused from the cache. 

Disabling buffer caching by marking `reusable = false` for buffers resolves the crashes.

kkartaltepe, 2023-09-22:
h265 appears to have more issues regarding invalid settings as well.

```
bo_create: buf 0:233 (VDEnc Cumulative CU Count Streamout Surface) 4096b cached? 0
[MOS]:  CRITICAL - UpdateMemoryPolicy:92: Invalid setting! Local only memory with cpu visible.

Thread 211 "obs" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 0x7ffef2ffd6c0 (LWP 3169675)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=5, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	     return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=5, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff3c8e8a3 in __pthread_kill_internal (signo=5, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff3c3e668 in __GI_raise (sig=5) at ../sysdeps/posix/raise.c:26
#3  0x00007fff854127d2 in MemoryPolicyManager::UpdateMemoryPolicy(MemoryPolicyParameter*)
    (memPolicyPar=memPolicyPar@entry=0x7ffef2ff9b40)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/os/memory_policy_manager.cpp:92
#4  0x00007fff8543abdd in GraphicsResourceSpecificNext::Allocate(OsContextNext*, GraphicsResourceNext::CreateParams&)
    (this=0x7ffeec058b80, osContextPtr=0x555558fb54a0, params=...)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/linux/common/os/mos_graphicsresource_specific_next.cpp:243
#5  0x00007fff85455124 in MosInterface::AllocateResource(MosStreamState*, _MOS_ALLOC_GFXRES_PARAMS*, _MOS_SPECIFIC_RESOURCE*&, char const*, char const*, int)
    (streamState=0x5555592998e0, params=params@entry=0x7ffef2ffa120, resource=@0x7ffef2ff9eb8: 0x7ffeec058a30, functionName=functionName@entry=0x7fff8636abc4 "AllocateResource", filename=filename@entry=0x7fff863f3718 "/usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/shared/bufferMgr/media_allocator.cpp", line=line@entry=95)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/linux/common/os/mos_interface.cpp:1739
#6  0x00007fff853e3118 in Mos_Specific_AllocateResource(_MOS_INTERFACE*, _MOS_ALLOC_GFXRES_PARAMS*, char const*, char const*, int, _MOS_SPECIFIC_RESOURCE*)
    (pOsInterface=0x5555586e95f0, pParams=0x7ffef2ffa120, functionName=0x7fff8636abc4 "AllocateResource", filename=0x7fff863f3718 "/usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/shared/bufferMgr/media_allocator.cpp", line=95, pOsResource=<optimized out>) at /usr/src/debug/intel-media-driver-git/media-driver/media_driver/linux/common/os/mos_os_specific.c:2265
#7  0x00007fff8627aa3b in Allocator::AllocateResource(_MOS_ALLOC_GFXRES_PARAMS&, bool, MOS_COMPONENT)
    (this=0x5555587d1a30, param=..., zeroOnAllocate=zeroOnAllocate@entry=false, component=component@entry=COMPONENT_Encode)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/shared/bufferMgr/media_allocator.cpp:95
#8  0x00007fff85f456be in encode::EncodeAllocator::AllocateResource(_MOS_ALLOC_GFXRES_PARAMS&, bool, _MOS_HW_RESOURCE_DEF)
    (this=<optimized out>, param=..., zeroOnAllocate=zeroOnAllocate@entry=false, resUsageType=resUsageType@entry=MOS_HW_RESOURCE_DEF_MAX)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/codec/hal/enc/shared/bufferMgr/encode_allocator.cpp:61
#9  0x00007fff8600a257 in encode::HevcVdencPkt::AllocateResources() (this=0x7ffeec051d00)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/codec/hal/enc/hevc/packet/encode_hevc_vdenc_packet.cpp:154
#10 0x00007fff85ffa8c4 in encode::HevcVdencPkt::Init() (this=0x7ffeec051d00)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/codec/hal/enc/hevc/packet/encode_hevc_vdenc_packet.cpp:1591
#11 0x00007fff8626222d in MediaPipeline::GetOrCreate(unsigned int) (this=this@entry=0x555559191560, packetId=packetId@entry=16842754)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/shared/pipeline/media_pipeline.cpp:166
#12 0x00007fff862630f6 in MediaPipeline::ActivatePacket(unsigned int, bool, unsigned short, unsigned char, unsigned char, unsigned char, unsigned char) (this=this@entry=0x555559191560, packetId=<optimized out>, 
    packetId@entry=16842754, immediateSubmit=immediateSubmit@entry=false, pass=pass@entry=0, pipe=pipe@entry=0 '\000', pipeNum=1 '\001', subPass=0 '\000', rowNum=0 '\000')
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/shared/pipeline/media_pipeline.cpp:180
#13 0x00007fff8604b822 in encode::HevcVdencPipeline::ActivateVdencVideoPackets() (this=0x555559191560)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_softlet/agnostic/common/codec/hal/enc/hevc/pipeline/encode_hevc_vdenc_pipeline.cpp:112
#14 0x00007fff85c5de34 in encode::HevcVdencPipelineXe_Xpm_Base::Execute() (this=0x555559191560)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_driver/media_softlet/agnostic/Xe_M/Xe_XPM_base/codec/hal/enc/hevc/pipeline/encode_hevc_vdenc_pipeline_xe_xpm_base.cpp:156
#15 0x00007fff85c5eafb in EncodeHevcVdencPipelineAdapterXe_Xpm_Base::Execute(void*) (this=0x555557c2ca10, params=<optimized out>)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_driver/media_softlet/agnostic/Xe_M/Xe_XPM_base/codec/hal/enc/hevc/pipeline/encode_hevc_vdenc_pipeline_adapter_xe_xpm_base.cpp:56
[... cut ...] ff8578f878 in DdiEncodeHevc::EncodeInCodecHal(unsigned int) (this=0x5555590a0fb0, numSlices=1)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp:477
#17 0x00007fff85768ed6 in DdiEncodeBase::EndPicture(VADriverContext*, unsigned int)
    (this=0x5555590a0fb0, ctx=0x55555882eaf0, context=<optimized out>)
    at /usr/src/debug/intel-media-driver-git/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:77
```

kkartaltepe, 2023-09-22:
invalid configuration appears related to the attempted optimization based on ftrLocalMemory here which is only enabled for discrete devices like my a750, https://github.com/intel/media-driver/blob/master/media_softlet/agnostic/common/codec/hal/enc/hevc/packet/encode_hevc_vdenc_packet.cpp#L145

Disabling that and a similar line at https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/mos_graphicsresource_specific_next.cpp#L195

Avoid these problems. It seems probably the codec should properly mark the memory as NonLockable if this really is device local scratch.

kkartaltepe, 2023-09-22:
Sorry it seems that buffer cache issue is related to leaking vadisplays. I'm not sure why this results in such issues but plumbing through proper closure of the displays resolved the crash. Personally i think it would be nice to fix this as its quite difficult to expect that leftover displays will corrupt the state for future displays.

The HEVC crash is only with debug asserts enabled and on release seems to work fine.

intel-mediadev, 2023-09-22:
Auto Created VSMGWL-68476 for further analysis.

bai-isaac, 2023-10-20:
hi @kkartaltepe  
Does this issue still exist?
From your description, this doesn't seem to be an encoding issue.
Is it because the display was not closure correctly first, causing the new display issue?

OttoYang, 2023-10-30:
Hi @kkartaltepe,
May I know if this issue still exist?

kkartaltepe, 2023-10-31:
> Does this issue still exist?

As described yes.

> From your description, this doesn't seem to be an encoding issue.
> Is it because the display was not closure correctly first, causing the new display issue?

That is a correct interpretation. since it wasn't closed correctly feel free to close this issue.

OttoYang, 2023-10-31:
Thanks for your reply! And feel free to report new issue if you encounter problems in hw encoding.

closed 2023-10-31 by OttoYang