intel/media-driver #1716: [Bug]: Pixelation Issue on Multiple Input Direct Write Operation

TheanSiew, 2023-08-30:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

None

### What happened?

We're trying to perform multiple input direct write to the different position within a same render output. Noticed pixelation issues if this operation is going through EU path. 

[image]

We've utlized libva-utils vppblending test as the baseline and make the following modification to reproduce the abovementioned issue:
1.	Removed blending operations
2.	Make an individual VA call for each input surface
Please refer to attached patch _0001-Quick-change-for-direct-render-output-write.patch_ for the changes.
The configuration file is being attached too: _process_blending.cfg.txt_
[0001-Quick-change-for-direct-render-output-write.patch](https://github.com/intel/media-driver/files/12473362/0001-Quick-change-for-direct-render-output-write.patch)
[process_blending.cfg.txt](https://github.com/intel/media-driver/files/12473366/process_blending.cfg.txt)

libva-utils https://github.com/intel/libva-utils commit: 98505afda65a31c13928e69297ca28acbaf2b897

### What's the usage scenario when you are seeing the problem?

Content Creation

### Do you want to contribute a patch to fix the issue?

None

FurongZhang, 2023-08-31:
I reproduced this issue and am checking it.

intel-mediadev, 2023-08-31:
Auto Created VSMGWL-67892 for further analysis.

peiyigu-intel, 2023-09-13:
Fix Merged : [[VP] Fix FC Corruption When Blending without Colorfill in Legacy Path](https://github.com/intel/media-driver/commit/197841a545b1eaf7f202e2d057a5a6395be46061)

FurongZhang, 2023-11-27:
Since the fix was merged, I will close this issue. Please feel free to contact us if any further follow up needed, thanks in advance!

closed 2023-11-27 by FurongZhang