intel/media-driver #1721: [Bug]: HEVC encode is 2.3x larger then AV1 source

vid-bin, 2023-09-06:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

I am using some YouTube clips to test 8k -> 4k encodes and the overall best encoding settings to use.

When using landscape videos with slow moving content the output format (YouTube 8k60 AV1 10bit or 8bit -> HEVC_QSV 10bit) the end results are usually a smaller encode than YouTube. By a significant amount.

However when you throw a lot of fast moving content into the video (such as raindrops with fast movements) the output can be 2.5x as large as YouTube av1 video.

Example:
8k60 AV1 8bit (1.7GB size) -> 4k60 HEVC_QSV 10bit (4.37GB size)

Encode options on DG2/arc with linux:

"-pix_fmt yuv420p10le", "-global_quality 20", "-extbrc 1", "-b_strategy 1", "-look_ahead 1", "-preset 1", "-tile_cols 1", "-tile_rows 1", "-adaptive_i 1", "-adaptive_b 1", "-tier 256", "-mbbrc 1"

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-09-07:
Auto Created VSMGWL-68043 for further analysis.

vid-bin, 2023-09-15:
Same issue with av1_qsv with the ICQ patch that was just released. The only difference is the file size is around twice as large as the same H265 encode with a global_quality of 21.

~900MB AV1 8k source is around 1.6GB H265_QSV or 2.6GB AV1_QSV when converted to 5k resolution on a slow moving landscape video.

For av1_qsv with a global_quality of 30 it's around 600MB.

OttoYang, 2023-09-18:
Hi @vid-bin, if you want to reduce file size in ICQ mode, you can increase the value of `global_quality`.  
Because high motion contents require a higher bit rate to achieve the quality set by user.

vid-bin, 2023-09-18:
> Hi @vid-bin, if you want to reduce file size in ICQ mode, you can increase the value of `global_quality`.
> Because high motion contents require a higher bit rate to achieve the quality set by user.

Yeah but it shouldn't be larger then the source file should it? Seems odd. It's not like I'm doing an ICQ of 5-7 or anything like that.

OttoYang, 2023-09-18:
> Yeah but it shouldn't be larger then the source file should it? Seems odd. It's not like I'm doing an ICQ of 5-7 or anything like that.

For transcoding, it is divided into two parts, decoding and encoding, the encoding process references the decoded raw data rather than the source file. If you want the new encoded file to be smaller than the source file, maybe you need to set the target bitrate or qp value smaller than the source file.

BlakeB415, 2023-09-18:
> Yeah but it shouldn't be larger then the source file should it? Seems odd. It's not like I'm doing an ICQ of 5-7 or anything like that.

If the source is from YouTube, YouTube often does CPU encoding using libaom-av1 which can hit lower bitrate targets. Consumer fixed-function encoders don't usually have the same efficiency.

OttoYang, 2023-10-23:
Hi @vid-bin , I'm not sure if these answers will help you?  
And if you have any other concerns or questions, please feel free to post them here.

OttoYang, 2023-10-30:
Hi @vid-bin ,  I'd like to know if you have any other questions about this?

fredrik-eriksson, 2023-11-05:
I have a related (or maybe same) observation regarding AV1 encoding. My understanding of ICQ is that it is an algorithm to ensure consistent subjective quality and my assumption has been that this subjective quality parameter is independent on the codec used. Meaning that the quality will be the same for any codec with an ICQ of 16 (my test sample), but depending on the efficiency of the codec the resulting file size will differ. Either this assumption is not true or the intel AV1 encoding needs some work:

h264 (8 bit, as 10 bit encoding is not supported):
```
-c:v h264_qsv -profile:v high -global_quality:v 16 -look_ahead 1 -scenario:v archive -preset:v veryslow -low_power 1
```
Filesize: ~1.9G

hevc:
```
-vf vpp_qsv=format=p010 -c:v hevc_qsv -profile:v main10 -global_quality:v 16 -look_ahead 1 -scenario:v archive -preset:v veryslow -low_power 1
```
Filesize: ~1.3G

av1:
```
-vf vpp_qsv=format=p010 -c:v av1_qsv -profile:v main -global_quality:v 16 -look_ahead 1 -scenario:v archive -preset:v veryslow -low_power 1
```
Filesize: ~5.9G

My eyes are not good enough to compare the quality of the encodes (they all look great), but I found this very unexpected. Is my assumption about ICQ values wrong and how does they vary between codecs in that case? Or is this a bug in the AV1 encoder?

leyu-yao, 2023-12-06:
> I have a related (or maybe same) observation regarding AV1 encoding. My understanding of ICQ is that it is an algorithm to ensure consistent subjective quality and my assumption has been that this [...]

Hi @fredrik-eriksson 
Since AV1 and HEVC is using different ICQ algorisms, we are expecting some certain file size difference. We run quality tests upon our internal clips. We do observe some file size difference between AV1 and HEVC on some clips, but not so large as what you observed, like 5.9G over 1.3G. The max file size difference we found is AV1 is 65% larger than HEVC.
To further investigate into this issue, we would like to debug into the specific clip if you are fine with sharing out the test clip, that would be of great help for us to narrow this issue.

fredrik-eriksson, 2024-01-04:
Since I've heard no reply, I assume my e-mail was never received. I've shortened the test file to 1 minute and ~260MB and made it available [here](https://cloud.feffe.ax/s/kDefwrs8opz6YLE). I still get roughly the same compress-ratios as posted in previous comment with this file. Please let me know when the clip is no longer needed.

leyu-yao, 2024-01-05:
> Since I've heard no reply, I assume my e-mail was never received. I've shortened the test file to 1 minute and ~260MB and made it available [here](https://cloud.feffe.ax/s/kDefwrs8opz6YLE). I still [...]

Thanks @fredrik-eriksson 
We had downloaded your clip. We are analyzing some of our clips on AV1/HEVC ICQ, and we will work on the clip you provided.