intel/media-driver #1726: error when exporting SSLKEYLOGFILE. When I run google-chrome or chromium, it creates a key successfully but when I transfer it to wireshark it does not decrypt the text, there seems to be an error or something misconfigured, any help, thanks in advance, this is my first post. I would be grateful, it's been weeks. running into this error

luisv92, 2023-10-06:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

export SSLKEYLOGFILE="/home/pirate/./ssl-key.log"

┌──(pirate㉿kali)-[~]
└─$ chromium &
[1] 9296

┌──(pirate㉿kali)-[~]
└─$ [9296:9339:1006/163658.895946:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.DBus.StartServiceByName: object_path= /org/freedesktop/DBus: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[9296:9296:1006/163658.971306:ERROR:chrome_browser_cloud_management_controller.cc(163)] Cloud management controller initialization aborted as CBCM is not enabled.
libva error: vaGetDriverNames() failed with unknown libva error
[9296:9296:1006/163704.042745:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.UnknownMethod: No such interface “org.freedesktop.portal.Settings” on object at path /org/freedesktop/portal/desktop
[9359:9359:1006/163716.686494:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted
[9359:9359:1006/163716.690925:ERROR:context_group.cc(154)] ContextResult::kFatalFailure: WebGL2 blocklisted
[9359:9359:1006/163716.699280:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

*o tengo mucho conocimiento cualquier ayuda es agradecida ,creo el error  esta en este punto algo mas pero ni idea he probado muchas cosas y ninguna solucion,vainfo >vainfo.log
libva info: VA-API version 1.20.0
libva error: vaGetDriverNames() failed with unknown libva error
vaInitialize failed with error code -1 (unknown libva error),exit

### Debug Information

2.- ls /dev/dri                                                                                                                                            
by-path  card0  renderD128
3.-lspci -nn |grep  -Ei 'VGA|DISPLAY'                                                                                                                     
00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]
4.-vainfo >vainfo.log
libva info: VA-API version 1.20.0
libva error: vaGetDriverNames() failed with unknown libva error
vaInitialize failed with error code -1 (unknown libva error),exit

### Do you want to contribute a patch to fix the issue?

No.

XinfengZhang, 2023-10-07:
15ad:0405 , it looks not a intel graphic card?

Jexu, 2024-04-24:
Let me close this issue since there is no response for long time, feel free to repoen it again if having any other questions.

closed 2024-04-24 by Jexu