intel/media-driver #1736: [Bug]: unable to use ICQ rate control mode in FFMpeg

iori2333, 2023-11-05:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

When trying to encode an h265 video using ICQ rate control method, I received an error:
```
# ffmpeg -loglevel debug -i source.mp4 -c:v hevc_qsv -global_quality 21 output.mp4
...
[hevc_qsv @ 0x55be34da1040] Encoder: input is system memory surface
[hevc_qsv @ 0x55be34da1040] Use Intel(R) oneVPL to create MFX session, the required implementation version is 1.1
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55be34d84e00] Trying to use DRM render node for device 0.
[AVHWDeviceContext @ 0x55be34d84e00] libva: VA-API version 1.19.0
[AVHWDeviceContext @ 0x55be34d84e00] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x55be34d84e00] libva: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55be34d84e00] libva: Found init function __vaDriverInit_1_19
[AVHWDeviceContext @ 0x55be34d84e00] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55be34d84e00] Initialised VAAPI connection: version 1.19
...
[hevc_qsv @ 0x55be34da1040] Initialized an internal MFX session using hardware accelerated implementation
[hevc_qsv @ 0x55be34da1040] Using the intelligent constant quality (ICQ) ratecontrol method
[hevc_qsv @ 0x55be34da1040] Selected ratecontrol mode is unsupported
[hevc_qsv @ 0x55be34da1040] Low power mode is unsupported
[hevc_qsv @ 0x55be34da1040] some encoding parameters are not supported by the QSV runtime. Please double check the input parameters.
[vost#0:0/hevc_qsv @ 0x55be34e14f00] Error initializing output stream: Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
```
It works fine when I switch rate control method to CQP.
```
# ffmpeg -loglevel debug -i source.mp4 -c:v hevc_qsv -q:v 21 output.mp4
...
[hevc_qsv @ 0x562f0c137fc0] Initialized an internal MFX session using hardware accelerated implementation
[hevc_qsv @ 0x562f0c137fc0] Using the constant quantization parameter (CQP) ratecontrol method
[hevc_qsv @ 0x562f0c137fc0] profile: hevc main; level: 296
[hevc_qsv @ 0x562f0c137fc0] GopPicSize: 65535; GopRefDist: 1; GopOptFlag:; IdrInterval: 1
[hevc_qsv @ 0x562f0c137fc0] TargetUsage: 4; RateControlMethod: CQP
[hevc_qsv @ 0x562f0c137fc0] QPI: 16; QPP: 21; QPB: 27
[hevc_qsv @ 0x562f0c137fc0] NumSlice: 1; NumRefFrame: 2
...
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

Env: Intel Celeron N5105 with JasperLake Graphics, I have enabled GuC/HuC following [Arch Wiki](https://wiki.archlinux.org/title/intel_graphics#Enable_GuC_/_HuC_firmware_loading)

```
# ls /dev/dri
card0  renderD128

# lspci -nn |grep  -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation JasperLake [UHD Graphics] [8086:4e61] (rev 01)

# vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.2.4 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
```

### Do you want to contribute a patch to fix the issue?

None

XinfengZhang, 2024-02-20:
@OttoYang , suppose JSL does not support HEVC ICQ encoding?

OttoYang, 2024-02-20:
> @OttoYang , suppose JSL does not support HEVC ICQ encoding?

Yes!

XinfengZhang, 2024-02-20:
@iori2333 , JSL does not support HEVC ICQ encoding,  you could use vainfo --all to check all of the BRC mode. so it is expected behavior.

closed 2024-02-22 by iori2333