intel/media-driver #1737: [Bug]: DRI3 implementation lacks vaPutSurface()

terr72, 2023-11-07:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

1. I'm using Linux Mint 21.2 Xfce, which is based on Ubuntu Jammy.
2. It currently comes with libva2 2.14.0-1 and iHD 22.3.1.
3. Run `xine -V vaapi --verbose "xvdr+tcp://127.0.0.1:37890#nocache"` or `xine -V vaapi --verbose randomrecording.ts`

4. This results in a black screen with only audio playing. The verbose log quickly fills with repeated `video_out_vaapi Error : vaPutSurface(): unknown libva error` entries.

The problem seems to be a lack of vaPutSurface() support in your DRI3 implementation.
See also [this libva issue report](https://github.com/intel/libva/issues/677), which affects hardware decoding in Chromium as well.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

xine, vdr-sxfe, vdr-plugin-xineliboutput
[Apparently also chromium.](https://github.com/intel/libva/issues/677)

### Debug Information

1. libva2 2.14.0-1 and intel-media-driver 22.3.1
2. `by-path  card0  renderD128`
3. `00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:46d1]`
4. [vainfo.log](https://github.com/intel/media-driver/files/13277837/vainfo.log)

### Do you want to contribute a patch to fix the issue?

None

atic-atac, 2023-11-24:
While there is a workaround of using...

export LIBVA_DRI3_DISABLE=1 

to your environment, it's not a great long term solution.

Is Intel working on this ?

terr72, 2023-11-26:
> Is Intel working on this ?

@atic-atac From my understanding this is an issue where both involved parties point fingers at each other.
The Intel devs say "_hey, you could fix this with a simple patch in your code_" while the libva team states "_you need to implement this in your code_". I can understand both sides, but being an end user caught between the stools, this is a pretty unsatisfying situation.