intel/media-driver #1740: [Bug]: QSV decoding stops if the stream contains an insert with a video format PAL

marselos88, 2023-11-22:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

I'm trying to transcode a udp stream on an intel video card using QSV. I just can’t solve the problem with stopping the ffmpeg process. When an insert in PAL format appears in a stream, while the main stream is in Component format.

Error:
[h264_qsv @ 0x55986c469140] A decode call did not consume any data: expect more data at input (-10)

That is, the problem is specifically in decoding. If I do transcoding hybrid CPU - QSV, VAAPI - QSV, or just VAAPI-VAAPI, this problem does not arise, but it takes a lot of server resources. It feels like only the QSV decoder is sensitive to format changes. Please tell me what I can try to do, maybe someone has encountered a similar problem.

command:
-init_hw_device qsv=hw -threads 1 -i udp://"225.30.30.30:1234?overrun_nonfatal=1&fifo_size=1000000" -c:v h264_qsv -b:v 2M -maxrate 2M -profile:v main -preset:v veryfast -g:v 50 -bf 2 -framerate 25 -r:v 25 -filter:v:0 "format=yuv420p,hwupload=extra_hw_frames=100,scale_qsv=720:576,vpp_qsv=deinterlace=2" -f null -

Linux Debian 12
ffmpeg version 5.1.3-1

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Frame parameters mismatch context 720,576,114 != 720,576,23

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Clipping frame in rate conversion by 0.199989

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], [NULL @ 0x55986c454900] ct_type:0 pic_struct:2

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], [h264_qsv @ 0x55986c469140] A decode call did not consume any data: expect more data at input (-10)

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Frame parameters mismatch context 720,576,114 != 720,576,23

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Clipping frame in rate conversion by 0.199989

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], [NULL @ 0x55986c454900] ct_type:0 pic_struct:1

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], [h264_qsv @ 0x55986c469140] A decode call did not consume any data: expect more data at input (-10)

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Frame parameters mismatch context 720,576,114 != 720,576,23

2023-10-27 16:24:04 INFO [НТВ:ntv_ufa], Clipping frame in rate conversion by 0.199989

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [NULL @ 0x55986c454900] ct_type:0 pic_struct:2

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [h264_qsv @ 0x55986c469140] A decode call did not consume any data: expect more data at input (-10)

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [h264_qsv @ 0x55986c469140] Format qsv chosen by get_format().

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x41524742 -> bgra.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x42475241 -> argb.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x41424752 -> rgba.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x52474241 -> abgr.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x58524742 -> bgr0.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x42475258 -> 0rgb.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x58424752 -> rgb0.

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVHWDeviceContext @ 0x55986c962c80] Format 0x52474258 -> 0bgr.
[... 145 lines cut ...]
2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [format @ 0x55986c745e80] Setting 'pix_fmts' to value 'qsv'

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [auto_scale_0 @ 0x55986d28ab80] w:iw h:ih flags:'' interl:0

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [format @ 0x55986c745e80] auto-inserting filter 'auto_scale_0' between the filter 'scaler_out_0_0' and the filter 'format'

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], Impossible to convert between the formats supported by the filter 'scaler_out_0_0' and the filter 'auto_scale_0'

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], Error reinitializing filters!

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], Failed to inject frame into filter network: Function not implemented

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], Error while processing the decoded data for stream #0:0

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [aac @ 0x55986c4687c0] Qavg: 600.674

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [aac @ 0x55986c4687c0] 2 frames left in the queue on closing

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], [AVIOContext @ 0x55986c43d240] Statistics: 273835912 bytes read, 0 seeks

2023-10-27 16:24:05 INFO [НТВ:ntv_ufa], Conversion failed!

### Do you want to contribute a patch to fix the issue?

None

marselos88, 2023-11-22:
intel-media-driver version 23.1.1+ds1-1

Jexu, 2023-12-26:
I think here is not right place to trace this issue. It is more like vpl rt behavior that could not get more bitstream data from your video. That is not media driver issue.

Jexu, 2024-04-25:
I will close this issue since it has no response for long time, but you could re-open it again if still having any questions.
As i mentioned above, this is probably vpl rt issue, you could report it in https://github.com/intel/vpl-gpu-rt again.

closed 2024-04-25 by Jexu