intel/media-driver #1743: [Bug]: wayland/gnome crash, after fullscreen or zoom video, then quit, mpv, 6.1.0-13-amd64, both intel free and non-free drivers

zenaan, 2023-11-25:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

1) run mpv, on any gif, mp4 or webm

1.b) "l"oop so mpv does not quit too soon

2) zoom ("Ctrl +"), or "f"ullscreen (this step is necessary)

3) "q"uit

and then my Debian 12 bookworm (Jan 2023/ stable) Gnome desktop, Wayland, quits/logs out.

Linux zen-L7 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29)

I can reliably reproduce this bug with either of the intel-media-va-driver-non-free and intel-media-va-driver drivers. Possibly easier to reproduce with the non-free driver, but can consistently and quickly reproduce on either.

Tested and reproduced the bug with .webm , .mp4 and .gif file.

This is on a laptop, no gpu.

grep model /proc/cpuinfo :
Intel(R) Core(TM) i7-10810U CPU

Able to test.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

desktop crashes/ logs out.

### Debug Information

tested both intel-media-va-driver-non-free and intel-media-va-driver, bug shows in both

$ ls -l /dev/dri
------->> 20231126@10:00:28 <<-------
total 0
drwxr-xr-x  2 root root         80 Nov 26 09:13 by-path
crw-rw----+ 1 root video  226,   0 Nov 26 09:13 card0
crw-rw----+ 1 root render 226, 128 Nov 26 09:13 renderD128

$ lspci -nn |grep  -Ei 'VGA|DISPLAY'
------->> 20231126@10:01:05 <<-------
00:02.0 VGA compatible controller [0300]: Intel Corporation Comet Lake UHD Graphics [8086:9bca] (rev 04)

[vainfo.log](https://github.com/intel/media-driver/files/13465422/vainfo.log)

steps 5 and 6 coming - will "submit" this in case desktop hangs, which is likely

### Do you want to contribute a patch to fix the issue?

None

zenaan, 2023-11-25:
step 5: ran export LIBVA_TRACE=/tmp/libva_trace.log then mpv test.mp4, reproduced the desktop hang/logout, but no /tmp/libva_trace.log file can be found...

step 6: see dmesg.log attached in case useful:
[dmesg.log](https://github.com/intel/media-driver/files/13465441/dmesg.log)

XinfengZhang, 2023-11-26:
> step 5: ran export LIBVA_TRACE=/tmp/libva_trace.log then mpv test.mp4, reproduced the desktop hang/logout, but no /tmp/libva_trace.log file can be found...
> 
no libva log found, it looks no driver was called?   please pay attention , the file name will be changed by adding process and thread ID.

zenaan, 2023-11-26:
On 11/26/23, Zhang, Xinfeng ***@***.***> wrote:
>> step 5: ran export LIBVA_TRACE=/tmp/libva_trace.log then mpv test.mp4,
>> reproduced the desktop hang/logout, but no /tmp/libva_trace.log file can
>> be found...
>
> no libva log found, it looks no driver was called?   please pay attention ,
> the file name will be changed by adding process and thread ID.

no libva log at all. I ran this as root:

find /tmp/ | grep libva

and nothing. This is what I run as user:

export LIBVA_TRACE=/tmp/libva_trace.log
mpv t.mp4

Tested again on free and non-free drivers. A bit harder to trigger
with just zoom on the free driver, zoom then full-screen, or vice
versa, followed by quit, seems to trigger this. Please let me know how
I can test...

XinfengZhang, 2023-11-27:
is there libva log when you run vainfo? want to confirm that the libva log works

zenaan, 2023-11-29:
On Monday, November 27, 2023, Zenaan Harkness ***@***.***>
wrote:
>> is there libva log when you run vainfo? want to confirm
>> that the libva log works
>
> yes:
> export LIBVA_TRACE=/tmp/libva_trace.log
> vainfo
> find /tmp/|grep libva
>
> gives:
> /tmp/libva_trace.log.035629.thd-0x00005168

BTW, this is basically a fresh Debian stable install on my laptop. If
there's something else I can test, or something else I should file a bug
against, please let me know.

mpv perhaps?

Jexu, 2024-04-25:
May i know if you still experience such crash?
Libva trace log is helpful if you could capture it, then we can find some clues about any error happen in media driver.

eero-t, 2024-05-10:
> desktop crashes/ logs out.

Are there any kernel warnings in `dmesg` about issues with GPU?

If not, I do not see how it could be media-driver bug, as desktop itself should not be using media-driver.

This sounds more like bug between `mpv` and Gnome desktop...  Does `mpv` trigger the same issue _without_ intel media driver?

zenaan, 2024-05-10:
Yes it still happens.

> This sounds more like bug between mpv and Gnome desktop... Does mpv
trigger the same issue *without* intel media driver?

Not sure how to test that.

https://github.com/mpv-player/mpv/issues/13062
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1057119
https://github.com/mpv-player/mpv/issues/11342Message ID:
***@***.***>

>

eero-t, 2024-05-13:
>> This sounds more like bug between mpv and Gnome desktop... Does mpv trigger the same issue *without* intel media driver?
> Not sure how to test that.

I've never used MPV, but cannot you tell it to use only CPU?

If not, you could e.g. try just removing Intel media driver...

>> Are there any kernel warnings in dmesg about issues with GPU?

Please answer this too.

If there are no errors in kernel log (which would indicate it being Intel kernel or user-space driver issue), then you need to check your desktop logs.  Typically you find those from your Desktop session manager logs, e.g. under `/var/log/gdm/` if you're using Gdm.

Some warnings you could see also with `systemctl status <session manager name>` and `journalctl -u <session manager name> -b 0`  (session manager logs since last boot).