intel/media-driver #1746: VAInfo on OpenSUSE returns error

js333031, 2023-12-04:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

Followed the SLES instructions from https://dgpu-docs.intel.com/driver/installation.html#sles-install-steps

OS: openSUSE Leap 15.5
HW: Xeon 4th Gen, Flex 170

I am able to use the Flex 170 for openvino inferencing. However, vainfo is failing:

```
(base) ats@localhost:~> vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva error: /usr/lib64/dri/iHD_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
(base) ats@localhost:~>
```

vainfo w/strace (last few lines):
```
brk(0x2062000)                          = 0x2062000
munmap(0x7fdd3077b000, 73441)           = 0
write(2, "libva error: /usr/lib64/dri/iHD_"..., 80libva error: /usr/lib64/dri/iHD_drv_video.so has no function __vaDriverInit_1_0
) = 80
write(2, "libva info: va_openDriver() retu"..., 39libva info: va_openDriver() returns -1
) = 39
write(2, "vaInitialize failed with error c"..., 66vaInitialize failed with error code -1 (unknown libva error),exit
) = 66
poll([{fd=3, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=3, revents=POLLOUT}])
writev(3, [{iov_base="<\0\2\0\0\0`\0+\3\1\0", iov_len=12}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 12
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\1\n\0\0\0\0\0\1\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 32
shutdown(3, SHUT_RDWR)                  = 0
close(3)                                = 0
exit_group(3)                           = ?
+++ exited with 3 +++
(base) ats@localhost:~>
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Others

### What impacted?

I compiled ffmpeg w/onevpl enabled (--enable-libvpl). It was not able to use hwaccel qsv option despite QSV being shown in decoders list. Then, I checked that vainfo is erroring out.
```
(base) ats@localhost:~/ffmpeg_work/ffmpeg> ffmpeg -decoders | grep qsv
ffmpeg version N-112898-g8c117b75af Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 7 (SUSE Linux)
  configuration: --enable-gpl --enable-version3 --enable-shared --enable-nonfree --enable-postproc --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libxvid --enable-libvpl --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib
  libavutil      58. 32.100 / 58. 32.100
  libavcodec     60. 35.100 / 60. 35.100
  libavformat    60. 18.100 / 60. 18.100
  libavdevice    60.  4.100 / 60.  4.100
  libavfilter     9. 14.100 /  9. 14.100
  libswscale      7.  6.100 /  7.  6.100
  libswresample   4. 13.100 /  4. 13.100
  libpostproc    57.  4.100 / 57.  4.100
 V....D av1_qsv              AV1 video (Intel Quick Sync Video acceleration) (codec av1)
 V....D h264_qsv             H264 video (Intel Quick Sync Video acceleration) (codec h264)
 V....D hevc_qsv             HEVC video (Intel Quick Sync Video acceleration) (codec hevc)
 V....D mjpeg_qsv            MJPEG video (Intel Quick Sync Video acceleration) (codec mjpeg)
 V....D mpeg2_qsv            MPEG2VIDEO video (Intel Quick Sync Video acceleration) (codec mpeg2video)
 V....D vc1_qsv              VC1 video (Intel Quick Sync Video acceleration) (codec vc1)
 V....D vp8_qsv              VP8 video (Intel Quick Sync Video acceleration) (codec vp8)
 V....D vp9_qsv              VP9 video (Intel Quick Sync Video acceleration) (codec vp9)
(base) ats@localhost:~/ffmpeg_work/ffmpeg>
```

### Debug Information

List of packages:

```
sudo zypper search -si intel
S  | Name                    | Type    | Version                 | Arch   | Repository
---+-------------------------+---------+-------------------------+--------+-------------------------------------------------------------
i+ | intel-fw-gpu            | package | 2023.39.2-255           | x86_64 | Intel graphics 15sp5 flex
i  | intel-gsc               | package | 0.8.9-65                | x86_64 | Intel graphics 15sp4 flex
i  | intel-gsc               | package | 0.8.9-65                | x86_64 | Intel graphics 15sp5 flex
i+ | intel-i915-dkms         | package | 1.23.6.37.230425.49-109 | x86_64 | Intel graphics 15sp5 flex
i+ | intel-igc-cm            | package | 1.0.202-i682            | x86_64 | Intel graphics 15sp5 flex
i  | intel-level-zero-gpu    | package | 1.3.26516.32-682        | x86_64 | Intel graphics 15sp4 flex
i+ | intel-media-driver      | package | 23.2.4-i682             | x86_64 | Intel graphics 15sp5 flex
i  | intel-metrics-discovery | package | 1.12.166-682            | x86_64 | Intel graphics 15sp4 flex
i  | intel-metrics-library   | package | 1.0.133-682             | x86_64 | Intel graphics 15sp4 flex
i+ | intel-ocloc             | package | 23.22.26516.32-682      | x86_64 | Intel graphics 15sp5 flex
i+ | intel-opencl            | package | 23.22.26516.32-682      | x86_64 | Intel graphics 15sp5 flex
i  | kernel-firmware-intel   | package | 20230724-150500.3.6.1   | noarch | Update repository with updates from SUSE Linux Enterprise 15
i  | libdrm_intel1           | package | 2.4.114-150500.3.2      | x86_64 | Main Repository
i  | libvulkan_intel         | package | 22.3.5-150500.75.2      | x86_64 | Main Repository
i  | ucode-intel             | package | 20231114-150200.35.1    | x86_64 | Update repository with updates from SUSE Linux Enterprise 15
(base) ats@localhost:~>
```

### Do you want to contribute a patch to fix the issue?

None

symo521, 2023-12-14:
I don't use OpenSuse but I am experiencing the same in linux mint 21.2 out of the blue

Jexu, 2023-12-26:
Please try to install latest version of libva firstly

js333031, 2024-01-12:
I tried to build but media-driver after building libva, but get following error:
```
[ 18%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva.cpp.o
/home/ats/media-driver/media_driver/linux/common/ddi/media_libva.cpp: In function ‘VAStatus DdiMedia_CreateSurfaces2(VADriverContextP, uint32_t, uint32_t, uint32_t, VASurfaceID*, uint32_t, VASurfaceAttrib*, uint32_t)’:
/home/ats/media-driver/media_driver/linux/common/ddi/media_libva.cpp:2747:5: error: ‘VADRMPRIME3SurfaceDescriptor’ was not declared in this scope
     VADRMPRIME3SurfaceDescriptor drmPrime3SurfaceDescriptor;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ats/media-driver/media_driver/linux/common/ddi/media_libva.cpp:2747:5: note: suggested alternative: ‘VADRMPRIMESurfaceDescriptor’
     VADRMPRIME3SurfaceDescriptor drmPrime3SurfaceDescriptor;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VADRMPRIMESurfaceDescriptor
In file included from /home/ats/media-driver/media_common/agnostic/common/os/mos_os.h:31:0,
                 from /home/ats/media-driver/media_softlet/agnostic/common/codec/hal/codechal_common.h:29,
```

Sherry-Lin, 2024-01-15:
What's the libva version you installed in your environment? Could you try to re-build again latest Libva version?

js333031, 2024-01-17:
I installed fresh openSUSE 15.5 Leap (not Tumbleweed). 

vainfo still gave errors so I updated to following versions by downloading src code and building locally:
```
libva-2.20.0
intel-gmmlib-22.3.15
intel-media-23.4.3
```

vainfo gives following error with updated SW:
```
ats@localhost:~/media-driver-work/media-driver> vainfo
Trying display: wayland
Trying display: x11
libva info: Open new log file /tmp/libva_trace.log.220404.thd-0x0000b5cb for the thread 0x0000b5cb
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.220404.thd-0x0000b5cb
libva info: VA-API version 1.20.0
Segmentation fault (core dumped)
ats@localhost:~/media-driver-work/media-driver>
```

Attaching strace output of running vainfo. 

clinfo:
```
ats@localhost:~/media-driver-work/media-driver> clinfo | head -n 5
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0
  Platform Profile                                FULL_PROFILE
ats@localhost:~/media-driver-work/media-driver>
```

kernel info:
```
ats@localhost:~/media-driver-work/media-driver> uname -a
Linux localhost.localdomain 5.14.21-150500.55.39-default #1 SMP PREEMPT_DYNAMIC Tue Dec 5 10:06:35 UTC 2023 (2e4092e) x86_64 x86_64 x86_64 GNU/Linux
ats@localhost:~/media-driver-work/media-driver>
```

dmesg log:
```
[69762.276812] show_signal_msg: 45 callbacks suppressed
[69762.276823] vainfo[21714]: segfault at 0 ip 0000000000000000 sp 00007ffeb62ec3a8 error 14 in vainfo[400000+6000]
[69762.276864] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[70172.542884] vainfo[24700]: segfault at 0 ip 0000000000000000 sp 00007ffc6df0ae38 error 14 in vainfo[400000+6000]
[70172.542924] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[72310.624800] vainfo[25243]: segfault at 0 ip 0000000000000000 sp 00007ffd85396168 error 14 in vainfo[400000+6000]
[72310.624832] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[72832.979817] vainfo[35212]: segfault at 0 ip 0000000000000000 sp 00007fff1aed3648 error 14 in vainfo[400000+6000]
[72832.979860] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[72989.818334] vainfo[35258]: segfault at 0 ip 0000000000000000 sp 00007ffec66e53a8 error 14 in vainfo[400000+6000]
[72989.818361] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[73022.231658] vainfo[35264]: segfault at 0 ip 0000000000000000 sp 00007ffcdff83e78 error 14 in vainfo[400000+6000]
[73022.231686] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[73057.598953] vainfo[35272]: segfault at 0 ip 0000000000000000 sp 00007fffd3c94be8 error 14 in vainfo[400000+6000]
[73057.598981] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[87075.334540] BTRFS info (device nvme0n1p2): qgroup scan completed (inconsistency flag cleared)
[103312.128786] vainfo[46483]: segfault at 0 ip 0000000000000000 sp 00007ffe83129108 error 14 in vainfo[400000+6000]
[103312.128817] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[103643.439025] vainfo[46539]: segfault at 0 ip 0000000000000000 sp 00007ffc5723c088 error 14 in vainfo[400000+6000]
[103643.439052] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
ats@localhost:~/media-driver-work/media-driver>
[vainfo_error.txt](https://github.com/intel/media-driver/files/13969242/vainfo_error.txt)

```

devices:
```
ats@localhost:~/media-driver-work/media-driver> ls -l /dev/dri/
total 0
drwxr-xr-x  2 root root        100 Jan 16 17:16 by-path
crw-rw----+ 1 root video  226,   0 Jan 16 17:16 card0
crw-rw----+ 1 root video  226,   1 Jan 16 17:16 card1
crw-rw-rw-  1 root render 226, 128 Jan 16 17:16 renderD128
ats@localhost:~/media-driver-work/media-driver>
```

Sherry-Lin, 2024-01-23:
Good to know building pass after libva upgrade. 
Your device is `Xeon 4th Gen, Flex 170` and the official support kernel is [intel-gpu/intel-gpu-i915-backports](https://github.com/intel-gpu/intel-gpu-i915-backports). You could get the pre-built package from Agama(refer to [Install from Agama](https://github.com/intel/media-driver/wiki/Install-from-Agama)).
If you still want to try upstream kernel, I'd suggest to upgrade to >6.2 at least, however there are some known patches yet upstream, so some failures may be expected.

js333031, 2024-01-23:
It did not pass unfortunately. Please have a look at my earlier response. vainfo now core-dumps (dmesg). clinfo has always worked.

Sherry-Lin, 2024-02-23:
5.14 kernel is too old, could you try >6.2 kernel? 

Your device is Xeon 4th Gen, Flex 170 and the official support kernel is [intel-gpu/intel-gpu-i915-backports](https://github.com/intel-gpu/intel-gpu-i915-backports). You could get the pre-built package from Agama(refer to [Install from Agama](https://github.com/intel/media-driver/wiki/Install-from-Agama)).
If you still want to try upstream kernel, I'd suggest to upgrade to >6.2 at least, however there are some known patches yet upstream, so some failures may be expected.

Sherry-Lin, 2024-03-26:
As no further feedback in recent 1 month, I'm going to close this issue. Please feel free to re-open it if any concerns. Thanks.

closed 2024-03-26 by Sherry-Lin