intel/media-driver #1747: [Bug]: multiple rules generate ...media_common/agnostic/common/vp/kernel/vpkrnheader.h with DBUILD_KERNELS and DENABLE_KERNELS

frantisekz, 2023-12-04:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

I gave a shot to try free kernels building after some time (if by any chance https://github.com/intel/media-driver/issues/1356 got resolved).

The command line used was:

`/usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON -DENABLE_NONFREE_KERNELS:BOOL=OFF -DENABLE_KERNELS:BOOL=ON -DBUILD_KERNELS:BOOL=ON -DMEDIA_BUILD_FATAL_WARNINGS=OFF -DVC1_Decode_Supported=no -DAVC_Encode_VME_Supported=no -DAVC_Encode_VDEnc_Supported=no -DHEVC_Encode_VME_Supported=no -DHEVC_Encode_VDEnc_Supported=no -DAVC_Decode_Supported=no -DHEVC_Decode_Supported=no -DAVC_Decode_Supported=no -G Ninja`

And the build ended with:

Run Build Command(s): /usr/bin/ninja-build -v -j 12
`ninja: error: build.ninja:45204: multiple rules generate /builddir/build/BUILD/media-driver-intel-media-23.4.2/media_common/agnostic/common/vp/kernel/vpkrnheader.h`

I am testing the 23.4.2 tag, with kernels removed as per https://github.com/intel/media-driver/wiki/Media-Driver-Shaders-%28EU-Kernels%29#disable-shaders-at-compile-time . Build with DBUILD_KERNELS and DENABLE_KERNELS both set to OFF works fine.

Full buildlog:

[build.log](https://github.com/intel/media-driver/files/13552290/build.log)

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

frantisekz, 2024-02-18:
Happens only with ninja, title amended.

intel-mediadev, 2024-03-04:
Auto Created VSMGWL-72139 for further analysis.

frantisekz, 2024-07-12:
> hi @frantisekz , sorry for inconvenience , several issues reported ,#1356 #1775 all related with the "BUILD_KERNEL"
> 
> 1. CMC version related issue, older compiler (llvm8) could compile , newer could not (I guess even new compiler could work,  the final binary maybe is different :)

Yeah, the older compiler didn't work either because of #1356 . Newer compiler fails even earlier because of #1775 .

> 2. new media driver regression? same compiler , but newer media driver release could work?  it is related with current issue we are discussing [[Bug]: With ninja generator, multiple rules generate ...media_common/agnostic/common/vp/kernel/vpkrnheader.h with DBUILD_KERNELS and DENABLE_KERNELS #1747](https://github.com/intel/media-driver/issues/1747)？

This is just an inconvenience and I can use the default cmake generator just fine, so just a minor issue, not really important for the Fedora package, and this one can be closed if the ninja support isn't planned in the media-driver.