intel/media-driver #1750: [Bug]: [drm] *ERROR* CPU pipe A FIFO underrun port,transcoder Intel ARC

nzrf, 2023-12-16:
### Is it regression? Good in old configuration?

None

### What happened?

System hang and becomes unresponsive. No video displayed anymore. 
The last thing that appears in the log every time is. 
`i915 0000:03:00.0: [drm] *ERROR* CPU pipe A FIFO underrun: port,transcoder,`

This will randomly happen, but not very often (every week to 30 days) Unsure what puts it in quite the state. 

System is fully patched Fedora 38 Patched  

Though this has happened with a number of kernels and kernel options still things seem to becoming unstable. Currently running options "i915.force_probe=56a1 i915.enable_psr=0" which can be turned off at any time, but this has happened with and without. So may as well remove them. Thought it had resolve it, but was wrong. 

System installed has the following intel packages installed
```
intel-media-driver-23.1.6-1.fc38.x86_64
intel-gmmlib-22.3.12-1.fc38.x86_64
intel-gpu-firmware-20231111-1.fc38.noarch
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

Server hosting plex media server for for local transcoding services and becomes unresponsive and will require a hard boot.

### Debug Information

ls /dev/dri
by-path  card1  renderD128

lspci -nn |grep  -Ei 'VGA|DISPLAY'
03:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A750] [8086:56a1] (rev 08)

[vainfo.log](https://github.com/intel/media-driver/files/13694179/vainfo.log)
[dmesg.log](https://github.com/intel/media-driver/files/13694186/dmesg.log)

### Do you want to contribute a patch to fix the issue?

None

nzrf, 2023-12-18:
Just as note changed out kernel options from. 
`i915.force_probe=56a1 i915.enable_psr=0` 
and put in place hoping to get catch something a bit more.
`i915.force_probe=56a1 i915.verbose_state_checks=1 drm.debug=0xe"`

filipmnowak, 2024-01-26:
got same error, and also `gnome-shell` segfault:

```
...
i915 0000:00:02.0: [drm] *ERROR* Failed to read DPCD register 0x92
i915 0000:00:02.0: [drm] *ERROR* Failed to read DPCD register 0x92
gnome-shell[6747]: segfault at 8 ip 00007f7f3eb54b4c sp 00007ffee7b06df0 error 4 in libmutter-12.so.0.0.0[7f7f3ea4e000+157000] likely on CPU 2 (core 4, socket 0)
i915 0000:00:02.0: [drm] *ERROR* CPU pipe A FIFO underrun: port,transcoder,
...
```

[dmesg_2024-01-26.txt](https://github.com/intel/media-driver/files/14063503/dmesg_2024-01-26.txt)

nzrf, 2024-01-31:
Just an update here at least on my own issue. Turned off the cstates on the motherboards and resolved the issue. 

I also have not had issue the last week in new ASUS motherboard W680 with a new CPU with the ARC GPU and no issues. 

Original system was i8400 with  Supermicro C7Z370-CG-L Gaming after disable cstates and no more issues.

filipmnowak, 2024-01-31:
> Just an update here at least on my own issue. Turned off the cstates on the motherboards and resolved the issue.
> (...)

thanks for a hint, will try it out!

MartinX3, 2024-03-09:
It suddenly also appears on my ThinkPad T460P with skylake iGPU.
i7-6820HQ with Intel HD Graphics 530.
I'll try the cstate setting, too.

filipmnowak, 2024-03-09:
> Just an update here at least on my own issue. Turned off the cstates on the motherboards and resolved the issue.
>
> I also have not had issue the last week in new ASUS motherboard W680 with a new CPU [...]

With those settings issues are pretty much gone (thanks again for the hints):

`i915.enable_guc=3 i915.enable_fbc=1 i915.modeset=1 i915.error_capture intel_idle.max_cstate=1`

i am not using a laptop, and i really don't care about most of CPU power saving modes, but still - those problems are kind of a bummer.

sr-tamim, 2024-04-07:
I am facing screen blinking issue for past week. I couldn't trace anything. I was preparing to re-installing my system. Today, while blinking screen I checked `journalctl` and found out `fifo underrun` error. I will try `max_cstate` fix. But is there any side-effects of this settings?

Device: Thinkpad P52
GPU: Nvidia Quadro P1000 (deactivated), Intel UHD Graphics (active)
Processor: Intel i7-8850H
OS: Pop OS 22.01

filipmnowak, 2024-05-01:
> I will try `max_cstate` fix. But is there any side-effects of this settings?

take a look [here](https://www.kernel.org/doc/html/latest/admin-guide/pm/cpuidle.html?highlight=max_cstate). your laptop might consume more energy when idle.

anszom, 2024-06-03:
I've been experiencing the same freeze with some regularity. I've recently added `intel_idle.max_cstate=1` to cmdline, but the bug occurred again. There was one new message though, something about changing power states.

```
[1140850.781238] i915 0000:19:00.0: [drm] *ERROR* CPU pipe A FIFO underrun: port,transcoder,
[1140892.092450] pcieport 0000:18:04.0: Unable to change power state from D3hot to D0, device inaccessible
```

relevant lspci output:
```
17:00.0 PCI bridge: Intel Corporation Device 4fa0 (rev 01)
18:01.0 PCI bridge: Intel Corporation Device 4fa4
18:04.0 PCI bridge: Intel Corporation Device 4fa4
19:00.0 VGA compatible controller: Intel Corporation Device 56a1 (rev 08)
1a:00.0 Audio device: Intel Corporation Device 4f90
```

```
 +-[0000:16]-+-00.0-[17-1a]----00.0-[18-1a]--+-01.0-[19]----00.0
 |           |                               \-04.0-[1a]----00.0
```

anszom, 2024-06-21:
I got the freeze yet again. I have the intel_idle.max_cstate=1 AND disabled C-states in a bios setting, so I'm pretty sure that disabling C-states is not an universal workaround.

filipmnowak, 2024-10-03:
> I got the freeze yet again. I have the intel_idle.max_cstate=1 AND disabled C-states in a bios setting, so I'm pretty sure that disabling C-states is not an universal workaround.

can confirm, still happens to me, although less frequently.

anszom, 2025-05-07:
It's me again. This time I got an oops instead of a silent freeze. Here's the log:
```
[1030969.502025] i915 0000:19:00.0: [drm] *ERROR* CPU pipe A FIFO underrun: port,transcoder,
[1030970.715548] rcs0: context submitted with incorrect RING_START [ffffffff], expected fedd0000
[1030972.028052] rcs0: context submitted with incorrect RING_CTL [ffffffff], expected 00000001
[1030973.340568] rcs0: context submitted with STOP_RING [ffffffff] in RING_MI_MODE
[1030975.965633] pcieport 0000:18:04.0: Unable to change power state from D3hot to D0, device inaccessible
[1030975.975190] kernel BUG at drivers/gpu/drm/i915/gt/intel_execlists_submission.c:702!
[1030975.983158] Oops: invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
[1030975.989178] CPU: 3 PID: 38 Comm: ksoftirqd/3 Tainted: G     U  W          6.10.0-drmtip+ #16
[1030975.997933] Hardware name: Micro-Star International Co., Ltd. MS-7A93/X299 SLI PLUS (MS-7A93), BIOS 1.I0 09/19/2022
[1030976.008633] RIP: 0010:execlists_submission_tasklet+0x15f8/0x1730 [i915]
[1030976.015704] Code: 0d e9 3a fb ff ff 0f 0b 0f 0b be 02 00 00 00 4c 89 ff e8 fb fc db da e9 15 fb ff ff 48 8b 7b 10 e8 4d c7 fc ff e9 f8 fb ff ff <0f> 0b 0f 0b be 03 00 00 00 e8 da fc db da e9 a2 ed ff ff 0f 0b 48
[1030976.034963] RSP: 0018:ffffb4878135fd38 EFLAGS: 00010082
[1030976.040451] RAX: ffffb48782221000 RBX: 00000080fedb051d RCX: ffff977dee305fc0
[1030976.047879] RDX: 0000000000000400 RSI: 0000000000000b98 RDI: 0000000000000000
[1030976.055308] RBP: ffffb4878135fdf0 R08: ffff977dec906000 R09: 0000000000000000
[1030976.062649] R10: ffff977df0c50000 R11: 0000000000000000 R12: ffffb4878135fda0
[1030976.070153] R13: ffff977e06983440 R14: 0000000000000000 R15: ffff977dec906000
[1030976.077581] FS:  0000000000000000(0000) GS:ffff978d1f780000(0000) knlGS:0000000000000000
[1030976.085981] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1030976.091996] CR2: 0000025b43a40000 CR3: 0000000d024a0001 CR4: 00000000003726f0
[1030976.099424] Call Trace:
[1030976.102086]  <TASK>
[1030976.104396]  ? show_regs+0x71/0x90
[1030976.108030]  ? die+0x38/0xa0
[1030976.111046]  ? do_trap+0xdb/0x100
[1030976.114666]  ? do_error_trap+0x75/0xb0
[1030976.118651]  ? execlists_submission_tasklet+0x15f8/0x1730 [i915]
[1030976.125087]  ? exc_invalid_op+0x53/0x80
[1030976.129165]  ? execlists_submission_tasklet+0x15f8/0x1730 [i915]
[1030976.135575]  ? asm_exc_invalid_op+0x1b/0x20
[1030976.139918]  ? execlists_submission_tasklet+0x15f8/0x1730 [i915]
[1030976.146305]  ? execlists_submission_tasklet+0x1277/0x1730 [i915]
[1030976.152601]  ? raw_spin_rq_unlock+0x10/0x40
[1030976.157104]  tasklet_action_common.constprop.0+0xd7/0x1e0
[1030976.162680]  tasklet_hi_action+0x29/0x40
[1030976.166917]  handle_softirqs+0xdc/0x310
[1030976.170990]  ? __pfx_smpboot_thread_fn+0x10/0x10
[1030976.175860]  run_ksoftirqd+0x3a/0x60
[1030976.179581]  smpboot_thread_fn+0xe1/0x1e0
[1030976.183819]  kthread+0xe1/0x120
[1030976.187263]  ? __pfx_kthread+0x10/0x10
[1030976.191249]  ret_from_fork+0x43/0x70
[1030976.195059]  ? __pfx_kthread+0x10/0x10
[1030976.198956]  ret_from_fork_asm+0x1a/0x30
[1030976.203195]  </TASK>
[1030976.205590] Modules linked in: uhid tls macvtap macvlan vhost_net vhost vhost_iotlb tap rfcomm xt_set ip_set xt_addrtype xfrm_user xfrm_algo xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink bridge stp llc overlay cmac algif_hash algif_skcipher af_alg bnep binfmt_misc nls_iso8859_1 intel_rapl_msr intel_rapl_common intel_uncore_frequency intel_uncore_frequency_common mei_hdcp mei_pxp isst_if_common snd_soc_avs snd_hda_codec_realtek snd_soc_hda_codec snd_hda_codec_hdmi snd_hda_ext_core snd_hda_codec_generic snd_hda_scodec_component snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine nfit snd_hda_intel x86_pkg_temp_thermal snd_intel_dspcfg intel_powerclamp snd_intel_sdw_acpi snd_hda_codec snd_usb_audio coretemp snd_hda_core snd_usbmidi_lib snd_hwdep snd_pcm kvm_intel snd_seq_midi snd_seq_midi_event kvm uvcvideo videobuf2_vmalloc snd_rawmidi uvc videobuf2_memops btusb snd_seq videobuf2_v4l2 btrtl
[1030976.205636]  snd_seq_device btintel videodev snd_timer rapl btbcm btmtk videobuf2_common intel_cstate intel_wmi_thunderbolt snd input_leds mei_gsc mc bluetooth soundcore mei_me mei ioatdma mac_hid acpi_tad sch_fq_codel msr parport_pc ppdev lp parport efi_pstore ip_tables x_tables autofs4 btrfs blake2b_generic xor raid6_pq libcrc32c dm_mirror dm_region_hash dm_log hid_generic usbhid hid uas usb_storage i915 nouveau drm_gpuvm drm_exec prime_numbers gpu_sched drm_buddy video drm_ttm_helper ttm crct10dif_pclmul crc32_pclmul drm_display_helper polyval_clmulni cec polyval_generic ghash_clmulni_intel rc_core sha512_ssse3 drm_kms_helper sha256_ssse3 nvme i2c_i801 igb mxm_wmi sha1_ssse3 ahci i2c_mux dca xhci_pci e1000e nvme_core drm libahci i2c_smbus i2c_algo_bit xhci_pci_renesas wmi aesni_intel crypto_simd cryptd
[1030976.369567] ---[ end trace 0000000000000000 ]---
[1030976.546831] RIP: 0010:execlists_submission_tasklet+0x15f8/0x1730 [i915]
[1030976.553848] Code: 0d e9 3a fb ff ff 0f 0b 0f 0b be 02 00 00 00 4c 89 ff e8 fb fc db da e9 15 fb ff ff 48 8b 7b 10 e8 4d c7 fc ff e9 f8 fb ff ff <0f> 0b 0f 0b be 03 00 00 00 e8 da fc db da e9 a2 ed ff ff 0f 0b 48
[1030976.573021] RSP: 0018:ffffb4878135fd38 EFLAGS: 00010082
[1030976.578580] RAX: ffffb48782221000 RBX: 00000080fedb051d RCX: ffff977dee305fc0
[... cut ...]
[1030976.615698] FS:  0000000000000000(0000) GS:ffff978d1f780000(0000) knlGS:0000000000000000
[1030976.624098] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1030976.630112] CR2: 0000025b43a40000 CR3: 0000000d024a0001 CR4: 00000000003726f0
[1030976.637541] Kernel panic - not syncing: Fatal exception in interrupt
[1030977.667534] Shutting down cpus with NMI
[1030977.681847] Kernel Offset: 0x1a400000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[1030977.868432] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

```

MarcelloCarlo, 2026-09-05:
Happens to my machine too, but the display issue happens when I connect my laptop to my external monitor and use anything beyond 60Hz refresh rate (The monitor supports 240Hz).

CPU: Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz
GPU: VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
Kernel: Linux 7.1.13-200.fc44.x86_64
My GRUB: GRUB_CMDLINE_LINUX="rhgb quiet preempt=full i915.enable_psr=0 i915.enable_fbc=0 i915.enable_rc6=0"

```
[  +8.855010] i915 0000:00:02.0: [drm] *ERROR* CPU pipe B FIFO underrun
[  +0.000038] i915 0000:00:02.0: [drm] *ERROR* uncleared pch fifo underrun on pch transcoder B
[  +0.000004] i915 0000:00:02.0: [drm] *ERROR* PCH transcoder B FIFO underrun
```

welshi32, 2026-10-22:
Tryna run Android x86, same here as well, quite an ancient machine, but still. Won't even reach boot animation
CPU: Pentium M 733(ULV Dothan) @ 1.1GHz
GPU: Intel 82855 GMA