intel/media-driver #1758: [Bug]: HDR Decode Issues - Kodi/FFMpeg and Arc A380

ilikenwf, 2024-01-02:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

In Linux, using Kodi, who uses VAAPI and ffmpeg for hardware video decoding, I attempt to play an HDR youtube video - any of them will do. This issue looks possibly similar to #629.

Interestingly, if I use yt-dlp or youtube-dl to download the HDR webm, it then plays properly. I have attempted to get help from the kodi devs and their inputstreamdirect people, however they have not been able to assist me in determining this issue.

The end result is:

```
2024-01-02 17:18:46.713 T:930     debug <general>: ffmpeg[0x55a833b0f730]: [SWR] Using fltp internally between filters
2024-01-02 17:18:46.714 T:1030    debug <general>: ffmpeg[0x55a83538fa50]: [vp9] Decode context initialised: 0x44/0x10000000.
2024-01-02 17:18:46.714 T:1030    debug <general>: ffmpeg[0x55a83538fa50]: [vp9] Param buffer (type 0, 92 bytes) is 0.
2024-01-02 17:18:46.714 T:1030    debug <general>: ffmpeg[0x55a83538fa50]: [vp9] Slice 0 param buffer (316 bytes) is 0x1.
2024-01-02 17:18:46.716 T:1030    debug <general>: ffmpeg[0x55a83538fa50]: [vp9] Slice 0 data buffer (39504 bytes) is 0x2.
2024-01-02 17:18:46.716 T:1030    debug <general>: ffmpeg[0x55a83538fa50]: [vp9] Decode to surface 0.
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Failed to end picture decode issue: 18 (invalid parameter).
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Not all references are available
2024-01-02 17:18:46.716 T:1030     info <general>: Skipped 1 duplicate messages..
2024-01-02 17:18:46.716 T:1030    error <general>: GetPicture - avcodec_receive_frame returned failure
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo - video decoder returned error
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Not all references are available
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Not all references are available
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Not all references are available
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
2024-01-02 17:18:46.716 T:1030    error <general>: ffmpeg[0x55a83538fa50]: [vp9] Not all references are available
2024-01-02 17:18:46.716 T:1030    debug <general>: CVideoPlayerVideo::CalcDropRequirement - hurry: 0
```

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

I am using the -git head of libva, gmmlib, and media-driver.

The card is a sparkle A380 - ls /dev/dri renders 
`by-path  card0	renderD128`

`03:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A380] [8086:56a5] (rev 05)`

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-01-15:
ffmpeg log shows some inputs wrong sending to media driver. According to your description, it always decodes well with native bitstream, but only fails with online inputstream, so please check if application sends incompleted bitstream data to media driver.

**Capture the libva trace log** for both native bitstream and online bitstream firstly, then we can have some initial analysis:
mkdir -p /home/trace
export LIBVA_TRACE=/home/trace/file
export LIBVA_TRACE_BUFDATA=1

If possible, use debug version media driver and **capture some driver logs** as well.
mkdir -p /data/driver_dump
put bellow config in /etc/igfx_user_feature_next.txt

[config]
Codec Message Tags=7
DDI Message Tags=7
Message HLT Enabled=1
Message HLT Output Directory=/data/driver_dump
Message Print Enabled=1
Mhw Message Tags=4
Mos Message Tags=4
[report]

intel-mediadev, 2024-03-04:
Auto Created VSMGWL-72137 for further analysis.

ilikenwf, 2024-03-05:
Sorry, I haven't had a chance to test this.

ilikenwf, 2024-10-04:
@Jexu another user who is using integrated Xe graphics has now confirmed this issue as well:

https://github.com/xbmc/xbmc/issues/23957#issuecomment-2392438443