intel/media-driver #1761: [Bug]: init fail with Xe KMD (although build support for that was enabled)

eero-t, 2024-01-09:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

 `vainfo` fails at driver init:
```
$ strace -f -k vainfo --display drm --device /dev/dri/renderD128
...
ioctl(3, DRM_IOCTL_VERSION, 0x7ffc7b535f10) = 0
 > /usr/lib/x86_64-linux-gnu/libc.so.6(ioctl+0x3f) [0x11a75f]
 > /usr/local/lib/libdrm.so.2.4.0(drmIoctl+0x30) [0x8160]
 > /usr/local/lib/dri/iHD_drv_video.so(map_gtt(mos_linux_bo*)+0xb6f) [0x17868f]
 > /usr/local/lib/dri/iHD_drv_video.so(DdiMedia_MapBuffer2+0x2944) [0x350964]
 > /usr/local/lib/dri/iHD_drv_video.so(__vaDriverInit_1_20+0x53) [0x351333]
 > /usr/local/lib/libva.so.2.2000.0(vaInitialize+0x5cd) [0x497d]
 > /usr/bin/vainfo() [0x28e0]
write(2, "libva error: /usr/local/lib/dri/"..., 61libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
) = 61
 > /usr/lib/x86_64-linux-gnu/libc.so.6(write+0x17) [0x114697]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(_IO_file_write+0x2d) [0x8aeed]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(_IO_file_xsputn+0x1d0) [0x8b7d0]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(psiginfo+0x15292) [0x776d2]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__fprintf_chk+0xa3) [0x134b23]
 > /usr/local/lib/libva.so.2.2000.0(vaSetInfoCallback+0x141) [0x3d81]
 > /usr/local/lib/libva.so.2.2000.0(vaInitialize+0x3ad) [0x475d]
 > /usr/bin/vainfo() [0x28e0]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_init_first+0x90) [0x29d90]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x29e40]
 > /usr/bin/vainfo() [0x2e75]
...
write(2, "libva info: va_openDriver() retu"..., 39libva info: va_openDriver() returns 18
) = 39
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Playback, Web browser, Cloud Gaming, Video Analytics, Video Conference, Immersive Media, Content Creation, Game Streaming, Others

### What impacted?

Media stack not working on upstream Xe KMD although support for that is compiled to the media-driver.

### Debug Information

Media stack is built from following versions:
- libva:    2.20.0
- GMMlib:   intel-gmmlib-22.3.16
- Media:    intel-media-24.1.0
- MediaSDK: intel-mediasdk-23.2.2
- oneVPL:   v2023.4.0
- VPL-GPU:  intel-onevpl-24.1.0

Using media-driver options that are supposed to enable Xe KMD support:
```
-DENABLE_PRODUCTION_KMD=FALSE -D=ENABLE_NEW_KMD=TRUE -DGEN8=OFF
```

Kernel is built from Xe repo default "drm-xe-next" branch (yesterday HEAD commit): https://gitlab.freedesktop.org/drm/xe/kernel

With Xe driver enabled:
```
# grep _XE[^A-Z] /boot/drm_xe.config 
CONFIG_DRM_XE=m
CONFIG_DRM_XE_FORCE_PROBE=""
CONFIG_DRM_XE_JOB_TIMEOUT_MAX=10000
CONFIG_DRM_XE_JOB_TIMEOUT_MIN=1
CONFIG_DRM_XE_TIMESLICE_MAX=10000000
CONFIG_DRM_XE_TIMESLICE_MIN=1
CONFIG_DRM_XE_PREEMPT_TIMEOUT=640000
CONFIG_DRM_XE_PREEMPT_TIMEOUT_MAX=10000000
CONFIG_DRM_XE_PREEMPT_TIMEOUT_MIN=1
CONFIG_DRM_XE_ENABLE_SCHEDTIMEOUT_LIMIT=y
```

And according to dmesg it works fine on TGL where I'm testing this:
```
# dmesg | grep xe[^a-z]
[    0.000000] Command line: BOOT_IMAGE=/boot/drm_xe rootwait fsck.repair=yes i915.force_probe=!9a60 xe.force_probe=9a60 ro
[    0.038111] Kernel command line: BOOT_IMAGE=/boot/drm_xe rootwait fsck.repair=yes i915.force_probe=!9a60 xe.force_probe=9a60 ro
[    3.068875] xe 0000:00:02.0: vgaarb: deactivate vga console
[    3.198711] xe 0000:00:02.0: [drm] Using GuC firmware from i915/tgl_guc_70.bin version 70.13.1
[    3.202558] xe 0000:00:02.0: [drm] Using HuC firmware from i915/tgl_huc.bin version 7.9.3
[    3.204943] xe REG[0x2340-0x235f]: allow read access
[    3.204946] xe REG[0x7010-0x7017]: allow rw access
[    3.204947] xe REG[0x7018-0x701f]: allow rw access
[    3.204974] xe REG[0x223a8-0x223af]: allow read access
[    3.204993] xe REG[0x1c03a8-0x1c03af]: allow read access
[    3.205011] xe REG[0x1d03a8-0x1d03af]: allow read access
[    3.205030] xe REG[0x1c83a8-0x1c83af]: allow read access
[    3.212040] [drm] Initialized xe 1.1.0 20201103 for 0000:00:02.0 on minor 0
[    4.462524] xe 0000:00:02.0: [drm] GT0: suspended
```

At least enough for Mesa device info tool to work with it:
```
$ /usr/local/bin/intel_dev_info
/dev/dri/renderD128:
devinfo struct size = 680
   name: Intel(R) UHD Graphics (TGL GT1)
   gen: 12
   PCI device id: 0x9a60
...
```

(Xe KMD did not provide any displays, so Mesa itself did not work.)

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2024-01-10:
please cherry-pick PR https://github.com/intel/media-driver/pull/1753 as well, there is no xe drm init entry currectly

eero-t, 2024-01-10:
> please cherry-pick PR #1753 as well, there is no xe drm init entry currectly

Thanks, I'll also update media-driver (.0 => intel-media-24.1.1), as new release tag included Xe KMD uAPI update.

eero-t, 2024-01-10:
FYI: by adding following to kernel config:
```
CONFIG_EXPERT=y
CONFIG_DRM_XE_DISPLAY=y
```

Xe KMD provided also displays, and X server & Mesa started to work.  In couple of the 3D benchmarks I tried, perf was still significantly worse than with i915, but benchmarks were GPU accelerated and their validation passed, so Xe KMD itself seems OK.

(Xe providing displays did not help media-driver though.)

Jexu, 2024-01-11:
Please check your kernel version, and you can get the uapi version that media driver in use  in https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/xe/include/README.md

And any critical log by runing vainfo directly?

Jexu, 2024-01-11:
Please check your kernel version, and you can find the uapi version used by media driver in https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/xe/include/README.md.

BTW, any critical log by running vainfo directly?

Jexu, 2024-01-11:
Have a update in https://github.com/intel/media-driver/pull/1753;

So please cherry-pick both https://github.com/intel/media-driver/pull/1753 and https://github.com/intel/media-driver/pull/1762 for try again.
It passed on my local.

eero-t, 2024-01-11:
> Have a update in #1753;
> 
> So please cherry-pick both #1753 and #1762 for trying again. It passed on my local.

Tried updated #1753, but #1762 does not apply to latest `intel-media-24.1.1` release.

(There's no `pat_index` in whole `mos_bufmgr_xe.c` file: https://github.com/intel/media-driver/commit/e2848ef13b7d41d2ac55e016feb9f763022d9de5.patch)

With just #1753, init advances a bit more, but still fails [1].

> BTW, we have not given a release version to fully enable xe kmd support in open source media, and I will give a try later and take back the feaback.

I tried also master `branch` with patches from both the PRs, but driver init still failed with the same strace backtrace [1].

`drm-xe-next` is the stuff going to upstream, so I would not expect there to be that much actual user-space ABI changes, although also those are allowed until KMD gets enabled by default for some HW...

----
[1] `strace` backtrace:
```
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0x64, 0x41, 0x30), 0x7fff5e016a60) = -1 EINVAL (Invalid argument)
 > /usr/lib/x86_64-linux-gnu/libc.so.6(ioctl+0x3f) [0x11a75f]
 > /usr/local/lib/libdrm.so.2.4.0(drmIoctl+0x30) [0x8160]
 > /usr/local/lib/dri/iHD_drv_video.so(mos_bo_alloc_xe(mos_bufmgr*, mos_drm_bo_alloc*)+0x1a2) [0x181e32]
 > /usr/local/lib/dri/iHD_drv_video.so(std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> > >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x2615) [0x18e9a5]
 > /usr/local/lib/dri/iHD_drv_video.so(std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> > >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x3731) [0x18fac1]
 > /usr/local/lib/dri/iHD_drv_video.so(MOS_SetUltFlag+0x1357) [0x15bd87]
 > /usr/local/lib/dri/iHD_drv_video.so(std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned char> > >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x1365) [0x18d6f5]
 > /usr/local/lib/dri/iHD_drv_video.so(DdiMedia_MapBuffer2+0x23d1) [0x359f61]
 > /usr/local/lib/dri/iHD_drv_video.so(__vaDriverInit_1_20+0x53) [0x35aea3]
 > /usr/local/lib/libva.so.2.2000.0(vaInitialize+0x5cd) [0x497d]
 > /usr/bin/vainfo() [0x28e0]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_init_first+0x90) [0x29d90]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x29e40]
 > /usr/bin/vainfo() [0x2e75]
openat(AT_FDCWD, "/etc/igfx_user_feature_next.txt", O_RDONLY) = -1 ENOENT (No such file or directory)
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__open64+0x5b) [0x11434b]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(_IO_file_open+0x26) [0x8c0b6]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(_IO_file_fopen+0x292) [0x8c412]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(fopen+0x7e) [0x7f6ae]
 > /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30(std::__basic_file<char>::open(char const*, std::_Ios_Openmode, int)+0x34) [0xd0bb4]
 > /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30(std::basic_filebuf<char, std::char_traits<char> >::open(char const*, std::_Ios_Openmode)+0x3e) [0x118ffe]
 > /usr/local/lib/dri/iHD_drv_video.so(std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > >, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::cha [... cut ...]
 > /usr/local/lib/dri/iHD_drv_video.so(DdiMedia_MapBuffer2+0x2eba) [0x35aa4a]
 > /usr/local/lib/dri/iHD_drv_video.so(__vaDriverInit_1_20+0x53) [0x35aea3]
 > /usr/local/lib/libva.so.2.2000.0(vaInitialize+0x5cd) [0x497d]
 > /usr/bin/vainfo() [0x28e0]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_init_first+0x90) [0x29d90]
 > /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x29e40]
 > /usr/bin/vainfo() [0x2e75]
write(2, "libva error: /usr/local/lib/dri/"..., 61libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
) = 61
```

eero-t, 2024-01-11:
> Please check your kernel version, and you can find the uapi version used by media driver in https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/xe/include/README.md.

That 3 week old commit mentioned there seems somewhat poorly chosen.

After `0f1d88f278` commit, there are a lot of uAPI changes _from that same day_, followed by a tag.

From newest to oldest:
```
$ git log --decorate=short --format=fuller 0f1d88f278...drm-xe-next ./drivers/gpu/drm/xe/ | awk '
/tag:/ {print}
/^commit/ {commit=$0}
/^CommitDate/ {date=$0}
/[uU][aA][pP][iI]/ {printf("%s\n%s\n%s\n", commit, date, $0)}
'
commit 8ebd9cd71f8acf675921283e0ba44d438932ad30
CommitDate: Thu Dec 21 16:31:30 2023 -0500
    compliance with DRM uAPI rules on compute platforms.
commit b6e1b708176846248c87318786d22465ac96dd2c (tag: drm-xe-next-2023-12-21-pr1-1)
commit 77a0d4d1cea2140ef56929ab1cfa5e525772c90e
CommitDate: Thu Dec 21 11:47:37 2023 -0500
    drm/xe/uapi: Remove reset uevent for now
commit 535881a8c50b79085327e7dbe26a4c55f3e1591b
CommitDate: Thu Dec 21 11:47:01 2023 -0500
    drm/xe/uapi: Document the memory_region bitmask
commit 535881a8c50b79085327e7dbe26a4c55f3e1591b
CommitDate: Thu Dec 21 11:47:01 2023 -0500
    The uAPI should stay generic in regarding to the bitmask. It is
commit d3d767396a02fa225eab7f919b727cff4e3304bc
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    drm/xe/uapi: Remove sync binds
commit d3d767396a02fa225eab7f919b727cff4e3304bc
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    The following bits have dropped from the uAPI:
commit 7e9337c29fb9251e27d7af092108f05857e733c1
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    drm/xe/uapi: Ensure every uapi struct has drm_xe prefix
commit 90a8b23f9b85a05ac3147498c42b32348bfcc274
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    drm/xe/pmu: Remove PMU from Xe till uapi is finalized
commit 90a8b23f9b85a05ac3147498c42b32348bfcc274
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    PMU uapi is likely to change in the future. Till the uapi is finalized,
commit 90a8b23f9b85a05ac3147498c42b32348bfcc274
CommitDate: Thu Dec 21 11:46:59 2023 -0500
    remove PMU from Xe. PMU can be re-added after uapi is finalized.
commit e670f0b4ef2419a7a51d1726044c8715ff4d4cda
CommitDate: Thu Dec 21 11:46:20 2023 -0500
    drm/xe/uapi: Return correct error code for xe_wait_user_fence_ioctl
commit 9212da07187f86db8bd124b1ce551a18b8a710d6
CommitDate: Thu Dec 21 11:46:20 2023 -0500
    drm/xe/uapi: add exec_queue_id member to drm_xe_wait_user_fence structure
```

Newest `8ebd9cd71f8` commit in above list _adds_ PVC IDs i.e. there are no uAPI _changes_ after the `drm-xe-next-2023-12-21-pr1-1` tag (at least ones where commit description would document it as uAPI change).

=> I would suggest media driver Xe KMD support to be updated to Xe KMD `drm-xe-next-2023-12-21-pr1-1` tag (commit `commitb6e1b708176846248c87318786d22465ac96dd2c`).

Jexu, 2024-01-12:
> That 3 week old commit mentioned there seems somewhat poorly chosen.
>
> After `0f1d88f278` commit, there are a lot of uAPI changes _from that same day_, followed by a tag.
>
> From newest to oldest:
>
> ``` [...]

It is the coming task that is already in plan and you will get notification in above README when it is done.
But at this moment, we don't have stable release tag for xe kmd support and I still recommend you use latest master branch of media, plus recommended uapi version.

eero-t, 2024-01-12:
**Media driver works**

With the suggested `0f1d88f278` Xe commit, _media-driver does work_, after applying _both_ of the PRs (tested `vainfo`, `vpl-inspect` & `ffmpeg`).

PR #1753 is already merged to `master`, but what about the other PR?  PR #1762 is marked as "do not merge", but driver does not work without that too...

**Xe KMD commits / building it**

> But at this moment, we don't have stable release tag for xe kmd support and I still recommend you use latest master branch of media, plus recommended uapi version.

FYI: the suggested `0f1d88f278` commit does not build as-is, at least on Ubuntu 22.04 LTS.

Building it requires another, later commit from the same day: https://gitlab.freedesktop.org/drm/xe/kernel/-/commit/e157f0f76258f11920fd5859a8ac1473a8ce5340.patch

Besides the uABI changes, that is one more reason for updating to newer commit. Because there have been no (documented) uAPI changes after that, I would actually recommend updating to a much recent commit, as `0f1d88f278` is `-rc3` i.e. likely to be buggy, and not really something you would want to recommend to users.

Especially as those older commits need kernel to be configured with `EXPERT` config option, or one does not have a display.  Display config issue will be fixed with:
https://lore.kernel.org/r/20240111104716.3548744-1-jani.nikula@intel.com

Jexu, 2024-01-26:
uapi version is already update after media commit '40d8ebfc7e31ffe9aa50e84d040ff8536d0c01d2'

https://github.com/intel/media-driver/pull/1762 is a test code, formal fix will come later.

eero-t, 2024-02-01:
> uapi version is already updated after media commit [40d8ebf](https://github.com/intel/media-driver/commit/40d8ebfc7e31ffe9aa50e84d040ff8536d0c01d2)

I'll test Xe commit listed in the updated README, but note that it's not in any branch:
```
xe$ git branch --contains 8ea438d53bbde68d8b0aedd1690f4b1644a1eb5e
```

Corresponding uAPI update one can be found:
```
xe$ git log --decorate=short --format=fuller drm-xe-next-2023-12-21-pr1-1...drm-xe-next ./drivers/gpu/drm/xe/ include/uapi/drm/xe_drm.h | awk '
/tag:/ {print}
/^commit/ {commit=$0}
/^CommitDate/ {date=$0}
/[uU][aA][pP][iI]/ {printf("%s\n%s\n%s\n", commit, date, $0)}
'
commit facd388708f06ab0c5cf492323c130e924c87aed
CommitDate: Wed Jan 24 11:08:18 2024 -0800
    xe_device, it will be simpler to implement the uapi version query if we
commit bf3ff145df184698a8a80b33265064638572366f (tag: drm-xe-next-fixes-2024-01-16, origin/drm-xe-next-fixes)
commit f031c3a7af8ea06790dd0a71872c4f0175084baa
CommitDate: Tue Dec 26 10:41:04 2023 -0800
    drm/xe/uapi: Remove DRM_XE_VM_BIND_FLAG_ASYNC comment left over
commit f031c3a7af8ea06790dd0a71872c4f0175084baa
CommitDate: Tue Dec 26 10:41:04 2023 -0800
    ("drm/xe/uapi: Remove sync binds").
commit f031c3a7af8ea06790dd0a71872c4f0175084baa
CommitDate: Tue Dec 26 10:41:04 2023 -0800
    Fixes: d3d767396a02 ("drm/xe/uapi: Remove sync binds")
commit 315acff5196f4e2f84a2a2d093000e0c6b0b4d1c (tag: drm-xe-next-fixes-2023-12-26)
```

from the `drm-xe-next` branch though:
```
xe$ git branch --contains f031c3a7af8ea06790dd0a71872c4f0175084baa
* drm-xe-next
```

I would think README Xe commit recommendation could be updated to `drm-xe-next-fixes-2024-01-16`  though, because there are no further uAPI changes between that and current commit (as you can see from above), and that contains the (trivial) kernel display config fix I mentioned earlier:
```
xe$ git show  drm-xe-next-fixes-2024-01-16
tag drm-xe-next-fixes-2024-01-16
Tagger: Thomas Hellström <thomas.hellstrom@linux.intel.com>
Date:   Tue Jan 16 10:34:54 2024 +0100

Driver Changes:
- Fix for definition of wakeref_t
- Fix for an error code aliasing
- Fix for VM_UNBIND_ALL in the case there are no bound VMAs
- Fixes for a number of __iomem address space mismatches reported by sparse
- Fixes for the assignment of exec_queue priority
- A Fix for skip_guc_pc not taking effect
- Workaround for a build problem on GCC 11
- A couple of fixes for error paths
- Fix a Flat CCS compression metadata copy issue
- Fix a misplace array bounds checking
- Don't have display support depend on EXPERT (as discussed on IRC)
```

(I'm going to test `drm-xe-next-fixes-2024-01-16` tag before the commit listed in media-driver README.)

eero-t, 2024-02-01:
> https://github.com/intel/media-driver/pull/1762 is test code, formal fix will come later.

That seems to be now merged into media-driver `master`,  and `master` seems to work fine with Xe repo `drm-xe-next-fixes-2024-01-16` tag.  Maybe you could update Xe KMD to that, verify that it works fine also in your own tests, and update the README recommendation for next media-driver release?

I'll close this once that media-driver release is out (i.e. release supporting Xe KMD commit belonging to some Xe repo branch, which does not need patching for kernel to build, and is preferably tagged).

eero-t, 2024-02-13:
Xe uAPI update ([40d8ebf](https://github.com/intel/media-driver/commit/40d8ebfc7e31ffe9aa50e84d040ff8536d0c01d2)) is now included to the latest `intel-media-24.1.3` tag.

However, PAT index workaround ([061a0c9](https://github.com/intel/media-driver/commit/061a0c94b94b80dbacdefc46ba6a40ca75c72546)) is still only in `master` branch, and [README](https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/xe/include/README.md) still points to non-branch Xe commit (instead of e.g. to `drm-xe-next-fixes-2024-01-16` tag).

closed 2024-03-04 by eero-t