intel/media-driver #1764: [Bug]: [Regression] catch libva-utils test_va_api CreateSurfaces/VAAPICreateSurfaces 1000+ failed cases

wenqingx, 2024-01-26:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

catch new Regression on libva-utils test_va_api Regression by https://github.com/intel/media-driver/commit/32b7f5558f94a558684fad8d2b88e65b26352211 

[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/3925, where GetParam() = (32:VAProfileAV1Profile0, 1:VAEntrypointVLD, 8192x8192)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/3926, where GetParam() = (32:VAProfileAV1Profile0, 1:VAEntrypointVLD, 10240x10240)

1326 FAILED TESTS

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Regression issues should be prioritized for repair.thanks.

### Debug Information

my env:
kernel:6.5.0-17-generic
OS:22.04.3 LTS (Jammy Jellyfish)
CPU:11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz, but it;s should be a common platform issue.

libva:3457aa8703bd
gmmlib:f5943626a108
libva-utils:b3ee1c3a0221

test cmdline:
`libva-utils/test/test_va_api --gtest_output=xml:libva-utils.test.test_va_api/All.xml --gtest_color=no `

log:
`
...
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/3925, where GetParam() = (32:VAProfileAV1Profile0, 1:VAEntrypointVLD, 8192x8192)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/3926, where GetParam() = (32:VAProfileAV1Profile0, 1:VAEntrypointVLD, 10240x10240)
1326 FAILED TESTS
`

### Do you want to contribute a patch to fix the issue?

None

Sherry-Lin, 2024-01-26:
@lixin5x to double confirm

lixin5x, 2024-01-29:
Need add prime3 test in libva-utils, https://github.com/intel/libva-utils/pull/357 is the PR

wenqingx, 2024-01-29:
> Need add prime3 test in libva-utils, [intel/libva-utils#357](https://github.com/intel/libva-utils/pull/357) is the PR

jsut tested on my local TGL device ,the issue fixed. thanks
`
[----------] Global test environment tear-down
[==========] 19698 tests from 14 test suites ran. (98063 ms total)
[  PASSED  ] 19698 tests.
`

wenqingx, 2024-02-02:
fixed patch merged and check latest report is pass , closed the issue ,thanks.

closed 2024-02-02 by wenqingx

lixin5x, 2024-03-07:
after https://github.com/intel/libva/pull/800 merged libvautils test will fail, please update libva