intel/media-driver #1771: [Bug]: dg2 A770 av1_qsv decode not working

zlice, 2024-02-04:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

I recently updated intel related libraries but I can't not remember the last time av1_qsv decode worked.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Playback, Web browser

### What impacted?

Using shotcut I am able to export av1_qsv and intel_gpu_top shows the GPU being used, the CPU is not pegged and it is also clearly faster than using CPU to encode av1. However ffmpeg, mpv, firefox do/can not use A770 GPU to decode av1 codecs.

### Debug Information

<details>
<summary>kernel and firmware</summary>

```
Linux 6.7.2 #1 SMP PREEMPT_DYNAMIC Mon Jan 29 13:02:56 EST 2024 x86_64 GNU/Linux

[    0.892814] Setting dangerous option enable_guc - tainting kernel        <<< enable_guc=2/3
[    0.911376] i915 0000:0b:00.0: [drm] Finished loading DMC firmware i915/dg2_dmc_ver2_08.bin (v2.8)
[    0.942020] i915 0000:0b:00.0: [drm] GT0: GuC firmware i915/dg2_guc_70.bin version 70.13.1
[    0.942022] i915 0000:0b:00.0: [drm] GT0: HuC firmware i915/dg2_huc_gsc.bin version 7.10.7
[    0.960566] i915 0000:0b:00.0: [drm] GT0: GUC: submission enabled
[    0.960566] i915 0000:0b:00.0: [drm] GT0: GUC: SLPC enabled
[    0.961566] i915 0000:0b:00.0: [drm] GT0: GUC: RC enabled
[    6.194712] i915 0000:0b:00.0: [drm] GT0: HuC: timed out waiting for MEI PXP

0b:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A770] (rev 08) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation Device 1020

```
</details>

<details>
<summary>packages and libraries</summary>

```
ffmpeg-6.0.1
MesaLib-devel-24.0.0_5
MesaLib-devel-32bit-24.0.0_5
glu-9.0.2_2
glu-32bit-9.0.2_2
glu-devel-9.0.2_2
glu-devel-32bit-9.0.2_2
intel-gmmlib-22.3.17_1
intel-gmmlib-devel-22.3.17_1
intel-gpu-tools-1.27.1_1
intel-media-driver-24.1.3_1
libOSMesa-24.0.0_5
libOSMesa-32bit-24.0.0_5
libgbm-24.0.0_5
libgbm-32bit-24.0.0_5
libgbm-devel-24.0.0_5
libgbm-devel-32bit-24.0.0_5
libxatracker-24.0.0_5
libxatracker-32bit-24.0.0_5
libva-2.20.0_1
libva-32bit-2.20.0_1
libva-devel-2.20.0_1
libva-utils-2.20.1_1
mesa-24.0.0_5
mesa-32bit-24.0.0_5
mesa-demos-8.4.0_3
mesa-dri-24.0.0_5
mesa-dri-32bit-24.0.0_5
mesa-opencl-24.0.0_5
mesa-opencl-32bit-24.0.0_5
mesa-vaapi-24.0.0_5
mesa-vdpau-24.0.0_5
mesa-vulkan-intel-24.0.0_5
mesa-vulkan-intel-32bit-24.0.0_5
mesa-vulkan-radeon-24.0.0_5
mesa-vulkan-radeon-32bit-24.0.0_5
ocl-icd-2.3.2_2
ocl-icd-32bit-2.3.2_2
oneVPL-2023.4.0_1
oneVPL-examples-2023.4.0_1
oneVPL-intel-gpu-24.1.3_1
opencl2-headers-2023.12.14_1
```

</details>

<details>
<summary>vainfo</summary>

```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.3 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

</details>

[ffmpeg-libva-trace.234359.thd-0x00000edb.txt](https://github.com/intel/media-driver/files/14154568/ffmpeg-libva-trace.234359.thd-0x00000edb.txt)
[ffmpeg-libva-trace.234359.thd-0x00000edc.txt](https://github.com/intel/media-driver/files/14154569/ffmpeg-libva-trace.234359.thd-0x00000edc.txt)
[ffmpeg-out.txt](https://github.com/intel/media-driver/files/14154570/ffmpeg-out.txt)

[mpv-libva-trace.235045.thd-0x000011fe.txt](https://github.com/intel/media-driver/files/14154571/mpv-libva-trace.235045.thd-0x000011fe.txt)
[mpv-libva-trace.235045.thd-0x00001208.txt](https://github.com/intel/media-driver/files/14154572/mpv-libva-trace.235045.thd-0x00001208.txt)
[mpv-out.txt](https://github.com/intel/media-driver/files/14154573/mpv-out.txt)

I briefly tried the newer onevpl library but since the version went 'down' I didn't want to deal with it right now. If anything looks bad on my end or I can troubleshoot something let me know. Thanks.

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-02-05:
It looks that decoding error happens on first frame. Do you see any gpu hang from dmesg log?

You said it is caused by updating av1 qsv lib, do you check if it works by ffmpeg-vaaapi? If so, please help capture ffmpeg-vaapi trace log as well and report the issue to https://github.com/oneapi-src/oneVPL-intel-gpu

zlice, 2024-02-05:
Nothing in dmesg. intel_gpu_top has activity for a second and then goes to 0%.

This wasn't caused by updating, updating was an attempted fix because it stopped working. I have updated kernel, mesa and firmware and am not sure what may have caused it.

I'm not sure what you mean by "ffmepg-aapi". h264 qsv/vaapi works fine but afaict has no vaapi for av1.

```
 V..... libdav1d             dav1d AV1 decoder by VideoLAN (codec av1)
 V....D libaom-av1           libaom AV1 (codec av1)
 V....D av1                  Alliance for Open Media AV1
 V..... av1_cuvid            Nvidia CUVID AV1 decoder (codec av1)
 V....D av1_qsv              AV1 video (Intel Quick Sync Video acceleration) (codec av1)
 A....D wmav1                Windows Media Audio 1
```

frantisekz, 2024-02-09:
I see the same output and issue with (only) av1 vaapi hw decode in mpv test with Fedora build of the media-driver. oneVPL-intel-gpu isn't on the system at all, just media-driver-24.1.3, and its dependencies, on top of the default package set of Fedora. 

The tested GPU was some integrated Iris Xe (I'll be able to access the system next week to get the full model).

zlice, 2024-02-15:
My distro updated LLVM and I recompiled everything but still had the same issue. I compiled intel-media-driver with 'nonfree' `-DENABLE_NONFREE_KERNELS=ON` and av1_vaapi shows up and works. I assume that nonfree is supposed to be enabled, sorry for the noise. Feel free to close pending frantisek's issue.

frantisekz, 2024-02-15:
> My distro updated LLVM and I recompiled everything but still had the same issue. I compiled intel-media-driver with 'nonfree' `-DENABLE_NONFREE_KERNELS=ON` and av1_vaapi shows up and works. I assume [...]

av1 should work even with -DENABLE_NONFREE_KERNELS=OFF I'd say (this is the way I build it for Fedora).

Jexu, 2024-02-18:
AV1 has feature named filmgrain that needs to load filmgrain binary kernel on dg2, so such build option is necessary.

So far, issue should be clear.

Jexu, 2024-03-04:
Let me close this issue and please help re-open it again if you have any concern or other question.

closed 2024-03-04 by Jexu