intel/media-driver #1772: [Bug]: DG2: Using `-extbrc 1` causes: Error initializing the encoder: invalid video parameters (-15)

JohnVillalovos, 2024-02-14:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

I have an Intel Arc A380 GPU and an i7-6700k (Skylake) (Intel HD Graphics 530).

 * A380 = renderD129
 * i7-6700k = renderD128

When I run the following command:
```
ffmpeg -hide_banner -init_hw_device vaapi=intel:/dev/dri/renderD129 -init_hw_device qsv=hw@intel -filter_hw_device hw \
 -an  -i input.mkv -frames:v 500 -c:v hevc_qsv -preset medium -profile:v main -async_depth 1 \
 -b:v 5000k -maxrate 10000k -bufsize 20000k -rc_init_occupancy 10000k -look_ahead_depth 40 \
 -extbrc 1 -b_strategy 1 -adaptive_i 1 -adaptive_b 1 -bf 7 -g 256 -strict -1 -fps_mode passthrough -y output.mkv
```

I get the  error:
```
<snip>
[hevc_qsv @ 0x5646a02e8680] Error initializing the encoder: invalid video parameters (-15)
[vost#0:0/hevc_qsv @ 0x5646a02846c0] Error initializing output stream: Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
```

If I change the above command by changing `renderD129` to `renderD128` then it works.

```
ffmpeg -hide_banner -init_hw_device vaapi=intel:/dev/dri/renderD128 -init_hw_device qsv=hw@intel -filter_hw_device hw \
 -an  -i input.mkv -frames:v 500 -c:v hevc_qsv -preset medium -profile:v main -async_depth 1 \
 -b:v 5000k -maxrate 10000k -bufsize 20000k -rc_init_occupancy 10000k -look_ahead_depth 40 \
 -extbrc 1 -b_strategy 1 -adaptive_i 1 -adaptive_b 1 -bf 7 -g 256 -strict -1 -fps_mode passthrough -y output.mkv
```

If I use `renderD129` and remove `-extbrc 1` then it also works.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

1. intel-media-driver-23.4.3-1.fc39.x86_64

3. Have both `renderD128` and `renderD129` in `/dev/dri/`

4.
```
lspci -nn |grep  -Ei 'VGA|DISPLAY'
00:02.0 Display controller [0380]: Intel Corporation HD Graphics 530 [8086:1912] (rev 06)
06:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A380] [8086:56a5] (rev 05)
```

5.
```
[8372.608432][ctx       none]==========va_TraceInitialize
[8372.608437][ctx       none]========== VA-API vendor string: Intel iHD driver for Intel(R) Gen Graphics - 23.4.3 ()
[8372.608438][ctx       none]=========vaInitialize ret = VA_STATUS_SUCCESS, success (no error)
[8372.608440][ctx       none]==========va_TraceGetDisplayAttributes
[8372.608440][ctx       none]   num_attributes = 1
[8372.608441][ctx       none]   attr_list[0] =
[8372.608441][ctx       none]     type = 0x00000013
[8372.608441][ctx       none]     min_value = -2147483648
[8372.608442][ctx       none]     max_value = -2147483648
[8372.608442][ctx       none]     value = -2147483648
[8372.608442][ctx       none]     flags = 0
[8372.564423][ctx       none]=========vaGetDisplayAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.611117][ctx       none]==========va_TraceGetDisplayAttributes
[8372.611118][ctx       none]   num_attributes = 1
[8372.611119][ctx       none]   attr_list[0] =
[8372.611119][ctx       none]     type = 0x00000015
[8372.611119][ctx       none]     min_value = -2138679643
[8372.611120][ctx       none]     max_value = -2138679643
[8372.611120][ctx       none]     value = -2138679643
[8372.611120][ctx       none]     flags = 1
[8372.611130][ctx       none]=========vaGetDisplayAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.611132][ctx       none]==========va_TraceMaxNumDisplayAttributes
[8372.611132][ctx       none]   max_display_attributes = 2
[8372.611137][ctx       none]==========va_TraceQueryDisplayAttributes
[8372.611137][ctx       none]   num_attributes = 2
[8372.611137][ctx       none]   attr_list[0] =
[8372.611137][ctx       none]     type = 0x00000014
[8372.611138][ctx       none]     min_value = 7
[8372.611138][ctx       none]     max_value = 7
[8372.611138][ctx       none]     value = 7
[8372.611138][ctx       none]     flags = 1
[8372.611139][ctx       none]   attr_list[1] =
[8372.611139][ctx       none]     type = 0x00000015
[8372.611139][ctx       none]     min_value = -2138679643
[8372.611139][ctx       none]     max_value = -2138679643
[8372.611139][ctx       none]     value = -2138679643
[8372.611140][ctx       none]     flags = 1
[8372.611143][ctx       none]=========vaQueryDisplayAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.762284][ctx       none]==========va_TraceGetDisplayAttributes
[8372.762287][ctx       none]   num_attributes = 1
[8372.762287][ctx       none]   attr_list[0] =
[8372.762288][ctx       none]     type = 0x00000015
[8372.762288][ctx       none]     min_value = -2138679643
[8372.762289][ctx       none]     max_value = -2138679643
[8372.762289][ctx       none]     value = -2138679643
[8372.762290][ctx       none]     flags = 1
[8372.762373][ctx       none]=========vaGetDisplayAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.762376][ctx       none]==========va_TraceMaxNumDisplayAttributes
[8372.762376][ctx       none]   max_display_attributes = 2
[8372.762383][ctx       none]==========va_TraceQueryDisplayAttributes
[8372.762384][ctx       none]   num_attributes = 2
[8372.762384][ctx       none]   attr_list[0] =
[8372.762385][ctx       none]     type = 0x00000014
[8372.762385][ctx       none]     min_value = 7
[8372.762386][ctx       none]     max_value = 7
[8372.762386][ctx       none]     value = 7
[8372.762387][ctx       none]     flags = 1
[8372.762387][ctx       none]   attr_list[1] =
[8372.762387][ctx       none]     type = 0x00000015
[8372.762388][ctx       none]     min_value = -2138679643
[8372.762388][ctx       none]     max_value = -2138679643
[8372.762389][ctx       none]     value = -2138679643
[8372.762389][ctx       none]     flags = 1
[8372.762394][ctx       none]=========vaQueryDisplayAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.762814][ctx       none]=========vaGetConfigAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.763395][ctx       none]=========vaGetConfigAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.763807][ctx       none]=========vaGetConfigAttributes ret = VA_STATUS_SUCCESS, success (no error)
[8372.764502][ctx       none]=========vaGetConfigAttributes ret = VA_STATUS_SUCCESS, success (no error)
```

### Do you want to contribute a patch to fix the issue?

None

Sherry-Lin, 2024-02-19:
`extbrc` is majorly used for DG1 media delivery. May I know what's the reason you want to use it on DG2/ACM? Is it ok for you to remove it?

JohnVillalovos, 2024-02-19:
I saw in https://github.com/intel/media-delivery/blob/c14a6d1e27a287e22b934e5c980282fa81d47d86/doc/quality.rst#L315-L337 that I should use it in order to use EncTools. And it seemed to recommend using EncTools. Is that not the case for DG2?