intel/media-driver #1775: [Bug]: Building kernels with cm_compiler from cmc_monorepo_110 fails

frantisekz, 2024-02-19:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. Build media driver with -DENABLE_NONFREE_KERNELS:BOOL=OFF -DENABLE_KERNELS:BOOL=ON -DBUILD_KERNELS:BOOL=ON
2. Use cmc_compiler from cmc_monorepo_110 branch (llvm 11)

```
gmake[2]: Entering directory '/builddir/build/BUILD/media-driver-intel-media-24.1.3/redhat-linux-build'
[  1%] Creating codec cmfc kernels output directory
cd /builddir/build/BUILD/media-driver-intel-media-24.1.3/redhat-linux-build/media_driver && /usr/bin/cmake -E make_directory /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel
cd /builddir/build/BUILD/media-driver-intel-media-24.1.3/redhat-linux-build/media_driver && /usr/bin/cmake -E make_directory /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel/kernel
[  1%] Compiling /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/hme_kernel_genx.cpp... to /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel/hme_kernel_genx_0.dat                Checking if renaming is needed...
cd /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel && /usr/bin/cmc -c -Qxcm -Qxcm_jit_target=icllp -I /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/ -Qxcm_jit_option="-nocompaction\ -output\ -binary" -mCM_emit_common_isa -mCM_no_input_reorder -mCM_printregusage -mCM_old_asm_name /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/hme_kernel_genx.cpp
[  1%] Compiling /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/downscale_convert_kernel_genx.cpp... to /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel/downscale_convert_kernel_genx_0.dat                Checking if renaming is needed...
cd /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/cache_kernel && /usr/bin/cmc -c -Qxcm -Qxcm_jit_target=icllp -I /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/ -Qxcm_jit_option="-nocompaction\ -output\ -binary" -mCM_emit_common_isa -mCM_no_input_reorder -mCM_printregusage -mCM_old_asm_name /builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/codec/kernel_free/Source/downscale_convert_kernel_genx.cpp
[  1%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_mos.dir/agnostic/common/os/mos_cmdbufmgr.cpp.o
/usr/bin/gmake  -f media_driver/CMakeFiles/iHD_drv_video_VP.dir/build.make media_driver/CMakeFiles/iHD_drv_video_VP.dir/depend
cd /builddir/build/BUILD/media-driver-intel-media-24.1.3/redhat-linux-build/media_driver && /usr/bin/g++ -DBUILD_KERNELS -DCLASS_TRACE=0 -DENABLE_KERNELS -DIGFX_ARL_SUPPORTED -DIGFX_DG2_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_MTL_SUPPORTED -DIGFX_XEHP_SDV_CMFCPATCH_SUPPORTED -DIGFX_XEHP_SDV_SUPPORTED -DIGFX_XE_HPG_CMFCPATCH_SUPPORTED -DIGFX_XE_HPG_SUPPORTED -DMEDIA_VERSION=\"24.1.3\" -DMEDIA_VERSION_DETAILS=\"\" -DVEBOX_AUTO_DENOISE_SUPPORTED=0 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_FULL_OPEN_SOURCE -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MANUAL_SOFTLET_ -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_RELEASE -D_VP8_DECODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/linux/common/os/i915/include -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/linux/common/os/i915/include/uapi -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_common/agnostic/common/os/user_setting -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_common/agnostic/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_common/linux/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/agnostic/common/os/share -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/agnostic/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/linux/common/os/osservice -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_softlet/linux/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/linux/common/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/cmrtlib/linux/hardware -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_common/linux/common/cp/os -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/cm -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/hw -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/media_interfaces -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/shared/mediacopy -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/common/shared -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8/cm -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8/hw -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8_bdw/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen8_bdw/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9/cm -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9/hw -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9_bxt/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9_skl/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9_glk/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen9_kbl/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/cm -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/hw -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen11_jsl_ehl/renderhal -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen12/cm -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen12/hw/vdbox -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/gen12/hw -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/g12/g12_base/hw/render -I/builddir/build/BUILD/media-driver-intel-media-24.1.3/media_driver/agnostic/Xe_M/Xe_XPM/hw/vdbox -I/builddir/build/BUILD [... cut ...]
            write_plane(EncSurfIndex, GENX_SURFACE_Y_PLANE, final_h_pos, final_v_pos + 24, YPix.select<8, 1, 32, 1>(24,0));
            ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
gmake[2]: Entering directory '/builddir/build/BUILD/media-driver-intel-media-24.1.3/redhat-linux-build'
LLVM ERROR: Frontend detected a fatal error!
```

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

XinfengZhang, 2024-02-20:
suppose it is related with CMC compiler version, how about to use a older version?

frantisekz, 2024-02-20:
> suppose it is related with CMC compiler version, how about to use a older version?

Yeah, it probably is. This is for Fedora packaging of the media-driver, and we no longer have llvm 8.0 in the distribution repositories, so older cm is unfortunately no-go.

I can imagine updating/fixing the media-driver for the newer cm-compiler isn't a priority, but I'd guess it'll be done someday?

XinfengZhang, 2024-02-21:
:) why not "DBUILD_KERNELS:BOOL=OFF", the final iHD_video_drv.so should be same, I am expecting no one will change media kernel code,  "open source media kernel" just was used to prove that the binary could be compiled.

frantisekz, 2024-02-21:
> :) why not "DBUILD_KERNELS:BOOL=OFF", the final iHD_video_drv.so should be same, I am expecting no one will change media kernel code, "open source media kernel" just was used to prove that the binary could be compiled.

That is unfortunately prohibited in Fedora packages, it has to be built from the source, or removed, so we're currently shipping the media-driver without kernels :(

intel-mediadev, 2024-03-04:
Auto Created VSMGWL-72132 for further analysis.

XinfengZhang, 2024-07-05:
hi @frantisekz , 
actually, we are expecting all user will use default build options, the "non free" kernel is actually free to use, it is under MIT license, they are existing in media driver as arrays.  the source file is close source just because these reasons : 1. maintenance effort; 2. partial code are assembly code, partial are c for media , because of history; 3. some algorithm should be reserved to protect IP; 4. some algorithm will be placed by HW in future.   such as: most of color format conversion / scaling will be handled by HW directly. no need to maintain an shader path. 
so, our main focus is delivering the full feature media driver.  the trending is shader will be replaced by HW. 
maintaining shader path (actually there are two shader path, full feature and free kernel, considering there are also an build_kernel option) is complex. and suppose no customer will use this free kernel path. because the feature is restricted. 
is it possible we only provide one iHD driver version in future?

DemiMarie, 2024-09-15:
> > suppose it is related with CMC compiler version, how about to use a older version?
> 
> Yeah, it probably is. This is for Fedora packaging of the media-driver, and we no longer have llvm 8.0 in the distribution repositories, so older cm is unfortunately no-go.

Could the media driver be built with a bundled compiler?