intel/media-driver #1776: [Bug]: 12th Gen Intel(R) Core(TM) i7-1255U encode cost too much cpu

okfujian, 2024-02-20:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

when we use 12th Gen Intel(R) Core(TM) i7-1255U  oneVPL to encode,we meet one issue.   
 1. when use  oneVPL GPU Runtime 2023Q2 Release - 23.2.4       https://github.com/oneapi-src/oneVPL-intel-gpu/releases/download/intel-onevpl-23.2.4/MediaStack.tar.gz    ./install_media.sh
    use this command:   "/opt/intel/media/share/vpl/samples/_bin/sample_encode  h265  -gpucopy::on -i /tmp/yuv/4208bit.yuv -o test1.h265 -nv12 -w 1920 -h 1080   -dstw 1920 -dsth 1080 -b 10000  -fps 60  -f 60   -hw -async 2 -lowpower:on -u speed    -timeout 600  -g 65535 -x 4 -r 8"
 
sample_encode cost about 43% cpu. why cost too much cpu?   

2023Q3 release and 2023Q4 release  also have this issue.

2. when we use  oneVPL GPU Runtime 2023Q1 Release - 23.1.5      https://github.com/oneapi-src/oneVPL-intel-gpu/releases/download/intel-onevpl-23.1.5/MediaStack.tar.gz  use less cpu.
use same command     "/opt/intel/media/share/vpl/samples/_bin/sample_encode  h265  -gpucopy::on -i /tmp/yuv/4208bit.yuv -o test1.h265 -nv12 -w 1920 -h 1080   -dstw 1920 -dsth 1080 -b 10000  -fps 60  -f 60   -hw -async 2 -lowpower:on -u speed    -timeout 600  -g 65535 -x 4 -r 8" 

sample_encode cost about 23% cpu
 

This is strange.should be 2023Q1 => 2023Q2  have some change cause this issue, Please help us look at this issue

OS version:  cat /proc/versionLinux version 5.15.0-56-generic (root@tpack) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP Thu Dec 8 13:31:18 UTC 2022

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-03-04:
Auto Created VSMGWL-72131 for further analysis.

eero-t, 2024-05-10:
Comparison of the perf output:
```
$ sudo perf record -a
<wait few mins>
^C
$ sudo perf report
```

For both driver versions could be interesting (i.e. where CPU goes in both cases).

WilsonWang-789, 2024-12-24:
Any updates here?

WilsonWang-789, 2024-12-24:
Any updates here? Thanks.